ipxe/ipxe #1200: HTTP Boot not getting DNS info (Dell R640)

champtar, 2024-04-22:
Since 9bbe77669c6e2b71826449d854f5aa0e2cee7767 we should get DNS when doing HTTP Boot.
HTTP booting ipxe.efi or snponly.efi, I get ip/mask/gw/cwuri/cwduri, but I do not get the dns setting.

How can I check if the info is present but not parsed correctly, or just not there ?

Testing with a Dell R640 (fully up to date), haven't tried other servers yet (I have access to some other Dell models and some HP Gen10 also)

mcb30, 2024-04-22:
> Since [9bbe776](https://github.com/ipxe/ipxe/commit/9bbe77669c6e2b71826449d854f5aa0e2cee7767) we should get DNS when doing HTTP Boot. HTTP booting ipxe.efi or snponly.efi, I get ip/mask/gw/cwuri/cwduri, but I do not get the dns setting.
> 
> How can I check if the info is present but not parsed correctly, or just not there ?

Try building with `DEBUG=efi_path`: the debug messages there should give enough information.

champtar, 2024-04-22:
I only have IPv4() and Uri(), no DNS
In IPv4() I also see DHCP even if this is a static IP ...
I'll try to test on more recent HW, and see if Dell is interested in a feature request

closed 2024-04-22 by champtar