ipxe/ipxe #1202: Make vlan available as a variable

dupondje, 2024-04-26:
When booting iPXE via UEFI, it automatically configures the UEFI PXE VLAN that is used, which is great.
But now it would be useful if the vlan was also available as a variable.

For example if you want to kickstart a CentOS install, the VM also needs its IP configured.
So you need to pass the vlan tag to the CentOS arguments, but the vlan is not available as a variable.

mcb30, 2024-04-26:
Yes, this would be useful, and should be exposed via the netdevice settings mechanism in `netdev_settings.c`.

The only real question is what to call the variable.  `${vlan}` is potentially too generic a name and so likely to have been used in existing scripts.  There is also the question of whether we need to expose the 12-bit VLAN tag number (as returned by `vlan_tag()`), the whole 16-bit VLAN tag control information number (as returned by `vlan_tci()`), or both.  We could maybe have `${vlantag}` and `${vlantci}`?

dupondje, 2024-04-26:
@mcb30 : Thanks for the very fast response :)

What about `${vlan-id}` or `${vlan-tag}` ? And I think only the tag is useful, don't see what additional info the tci will give.
But of course, it does not hurt to have `${vlan-tci}` ... So if both is same effort, I would just do both :)

aleks-mariusz, 2026-03-11:
> The only real question is what to call the variable

is this REALLY blocked for two years now because we can't come up with what to call the variable? :-) i don't even know what tci is, i just need the vlan number so i don't have to hard code these values into my boot.ipxe's when generating the kernel command line params to pass installers