ipxe/ipxe #1215: Intel 10GBit X540-AT2 100MBit/s not working correctly

PeterSurda, 2024-05-14:
I have a server, Dell PowerEdge C6220 II, with a dual port 10-Gigabit X540-AT2 network card. PCI IDs 8086:1528, in ipxe this is catered for by the intelx driver. The switch is set to 100MBit/s. Unfortunately, ipxe doesn't handle this situation correctly, it says there is no media connected. At least this is my conclusion based on experimentation. On Ubuntu the network card autodetects 100MBit/s and works fine. This is a rented dedicated server and I can't replace the network card with a 1Gbit quad port version, or change the switch port speed.

I tried several different builds of ipxe, the official one, my own, and netboot.xyz, they all behave the same way.

Let me know what other tests I can run.

NiKiZe, 2024-05-14:
Output of `ifstat` is always good, but in this case, have you tried snp.efi? How do you start ipxe?

NiKiZe, 2024-06-26:
Closing due to no feedback.

closed 2024-09-24 by zube[bot]