ipxe/ipxe #1219: Building with gcc 14.1.1 fails

taoky, 2024-05-19:
With Arch Linux current gcc version:

```console
> gcc --version
gcc (GCC) 14.1.1 20240507
Copyright (C) 2024 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

And current master branch (dc118c53696af6a0b1a8ee78fc9a4d28a217fb21) ipxe `make` fails with a `maybe-uninitialized`:

```
In file included from drivers/net/etherfabric.c:36:
In function ‘falcon_xaui_link_ok’,
    inlined from ‘falcon_init_xmac’ at drivers/net/etherfabric.c:2355:14,
    inlined from ‘falcon_init_xmac’ at drivers/net/etherfabric.c:2314:1:
drivers/net/etherfabric.h:524:28: error: ‘reg.u32[0]’ may be used uninitialized [-Werror=maybe-uninitialized]
  522 |         (dword).u32[0] = ( ( (dword).u32[0]                                   \
      |                          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  523 |                              & ~EFAB_INPLACE_MASK32 (  0,  31, field ) )      \
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  524 |                            | EFAB_INSERT_FIELD32 (  0,  31, field, value ) ); \
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/net/etherfabric.c:2233:9: note: in expansion of macro ‘EFAB_SET_DWORD_FIELD’
 2233 |         EFAB_SET_DWORD_FIELD ( reg, FCN_XX_DISPERR, FCN_XX_DISPERR_RESET);
      |         ^~~~~~~~~~~~~~~~~~~~
drivers/net/etherfabric.c: In function ‘falcon_init_xmac’:
drivers/net/etherfabric.c:2213:22: note: ‘reg.u32[0]’ was declared here
 2213 |         efab_dword_t reg;
      |                      ^~~
cc1: all warnings being treated as errors
make: *** [Makefile.housekeeping:964: bin/etherfabric.o] Error 1
```

Zepmann, 2024-06-01:
Workaround:

Edit `Makefile.housekeeping`. Find the first instance of variable WORKAROUND_CFLAGS. Edit it as follows:

```
WORKAROUND_CFLAGS := -Wno-error=maybe-uninitialized
```

Once applied compilation will succeed with a warning instead of an error.

kcinickgx, 2024-06-20:
Fix:

Edit `src/drivers/net/etherfabric.c`. Goto line 2219, add this line:
```
    int has_phyxs;
    int link_ok = 1;

    reg.u32[0] = 0; # <--- this one

    /* Read Falcon XAUI side */
```
a more advanced programmer may find another initialization, a malloc maybe?, but i think as u32 is an int, 0 should be fine, anyway, no error, no warning, cheers!

mcb30, 2024-09-02:
Fixed in https://github.com/ipxe/ipxe/commit/7f75d320f

closed 2024-09-02 by mcb30