ipxe/ipxe #1221: E810

aguaiyoung, 2024-05-21:


aguaiyoung, 2024-05-21:
Run into qemu preboot with E810 NIC pcie passthrough. 
This is qemu configuration:
................................
 <hostdev mode='subsystem' type='pci' managed='yes'>
      <driver name='vfio'/>
      <source>
        <address domain='0x0000' bus='0x4b' slot='0x00' function='0x2'/>
      </source>
      <alias name='hostdev0'/>
      <rom bar='on' file='/home/ipxe/src/bin/80861593.rom'/>
      <address type='pci' domain='0x0000' bus='0x11' slot='0x00' function='0x0'/>
    </hostdev>
................................
And this console from start up VM with debug:
make -C src bin/80861593.rom DEBUG=dhcp,pci,ice:3,netdevice,intelxl -j128
ooting from ROM...fined from test_ipxe.xml
iPXE (PCI 11:00.0) starting execution...ok
iPXE initialising devices...roy test_ipxe
0000:00:00.0 (8086:29c0 class 060000) has no driver
0000:00:01.0 (1013:00b8 class 030000) has no driver
0000:00:02.0 (1b36:000c class 060400) has no drivernsole test_ipxe
0000:00:02.1 (1b36:000c class 060400) has no driver
0000:00:02.2 (1b36:000c class 060400) has no driver
0000:00:02.3 (1b36:000c class 060400) has no driver
0000:00:02.4 (1b36:000c class 060400) has no driver
0000:00:02.5 (1b36:000c class 060400) has no driver
0000:00:02.6 (1b36:000c class 060400) has no driver
0000:00:02.7 (1b36:000c class 060400) has no driver
0000:00:03.0 (1b36:000c class 060400) has no driver
0000:00:03.1 (1b36:000c class 060400) has no driver
0000:00:03.2 (1b36:000c class 060400) has no driver
0000:00:03.3 (1b36:000c class 060400) has no driver
0000:00:03.4 (1b36:000c class 060400) has no driver
0000:00:03.5 (1b36:000c class 060400) has no driver
0000:00:03.6 (1b36:000c class 060400) has no driver
0000:00:03.7 (1b36:000c class 060400) has no driver
0000:00:1f.0 (8086:2918 class 060100) has no driver
0000:00:1f.2 (8086:2922 class 010601) has no driver
0000:00:1f.3 (8086:2930 class 0c0500) has no driver
0000:01:00.0 (1b36:000e class 060400) has no driver
0000:02:01.0 (8086:7020 class 0c0300) has no driver
0000:05:00.0 (1af4:1043 class 078000) has no driver
0000:07:00.0 (1af4:1045 class 00ff00) has no driver
0000:11:00.0 (8086:1593) has driver "e810c-sfp"
0000:11:00.0 has mem f4000000 io 0 irq 11
0000:11:00.0 device not enabled by BIOS! Updating PCI command 0142->0147
0000:11:00.0 latency timer is unreasonably low at 0. Setting to 32.
ICE 0x27944 PF 2 using port 2
INTELXL 0x27944 ARQ is at [7ef38c00,7ef38c80) buf [7ef34c00,7ef38c00)
INTELXL 0x27944 ATQ is at [7ef3cd00,7ef3cd80) buf [7ef38d00,7ef3cd00)
ICE 0x27944 firmware v1/7.5.4 API v0/1.7.11
INTELXL 0x27944 already in non-PXE mode
ICE 0x27944 SEID 0x0002:
00000000 : 02 00 02 00 00 00                               : ......          
ICE 0x27944 SEID 0x800a:
00000000 : 0a 80 02 00 02 00                               : ......          
ICE 0x27944 VSI 0x000a uplink 0x0002 func 0x0002
ICE 0x27944 SEID 0x0000:
00000000 : 00 00 00 00 00 00                               : ......          
ICE 0x27944 TEID 0x00000147 type 3
ICE 0x27944 has MAC address 68:05:ca:c1:be:a2
NETDEV net0 registered (phys 0000:11:00.0 hwaddr 68:05:ca:c1:be:a2)
ICE 0x27944 speed 0x00 status 0x40 cause ice_probe

iPXE 1.21.1+ (gd7e58c) -- Open Source Network Boot Firmware -- https://ipxe.org
Features: DNS HTTP iSCSI TFTP AoE ELF MBOOT PXE bzImage Menu PXEXT

NETDEV net0 opening                      
INTELXL 0x27944 ring 290000 is at [7ef3ce00,7ef3d600)
INTELXL 0x27944 ring 2c0000 is at [7ef3d600,7ef3da00)
[... 142 lines cut ...]
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 63 82 53 63 35 01 01 39 02 05 
c0 5d 02 00 00 5e 03 01 02 01 3c 20 50 58 45 43 
6c 69 65 6e 74 3a 41 72 63 68 3a 30 30 30 30 30 
3a 55 4e 44 49 3a 30 30 32 30 30 31 4d 04 69 50 
58 45 37 18 01 03 06 07 0c 0f 11 1a 2a 2b 3c 42 
43 77 80 81 82 83 84 85 86 87 af cb af 2d b1 05 
01 80 86 15 93 eb 03 01 15 01 17 01 01 22 01 01 
13 01 01 11 01 01 27 01 01 19 01 01 10 01 02 21 
01 01 15 01 01 18 01 01 12 01 01 3d 07 01 68 05 
ca c1 be a2 61 11 00 33 f5 6e 51 46 e8 cc 40 9d 
94 2f c3 51 ea 79 cb ff 
/ipxe.org/4c106035) configuration via dhcp failed: Connection timed out (https:/
 No configuration methods succeeded (https://ipxe.org/040ee119)
No more network devices

I think the promisc switch rule wasn't set up into firmware. And the opcode 0x0601 was duplicated when ice_admin_link was invoked.

Please help to address the root cause. Thanks a lot.

NiKiZe, 2024-05-21:
With no description I can only guess, but maybe one of these can help https://github.com/search?q=repo%3Aipxe%2Fipxe+e810&type=issues  , if not feel free to open a discussion (or just add a comment here)

But we can't work with an empty description