ipxe/ipxe #1222: The Mellanox nic  does not work in the latest branch and works fine on 1.21.1-tag

IAseven, 2024-05-22:
Mellanox nic:
[image]

The network adapter is started and the ip address is configured properly, but the gateway cannot be pinged from the master-branch，

What data do I need to check to solve this bug,

Thank you very much for your answers

NiKiZe, 2024-05-22:
This is probably a duplicate of existing issues.
But to verify which binary are you using, and what is the output of ifstat in ipxe?

IAseven, 2024-05-23:
@NiKiZe 
I'm very sorry, this is my problem, I used the latest version wrong, I re-downloaded the latest code and can communicate normally after compilation.
I am very sorry that I did not inquire relevant questions in the issues in advance, which has delayed your precious time.
The next time I have a problem, I look at the issues first
Finally, thank you very much for your selfless dedication to ipxe over the past ten years.

closed 2024-05-23 by IAseven