ipxe/ipxe #1227: The Option ROM driver is getting the initial link status as "down."

PaVaSjAiN, 2024-05-28:
Upon chainloading the iPXE.efi from DHCP server the initial link status seen is "DOWN".

Whenever we boot in this fashion, it "Boots from a network device." Then it "Opens the device and displays the device status." At this time, it calls "ifstat()", which checks the link status and gets the status as "down."

[image]

At the time of boot, whenever the driver is attached to the device, i.e., when "nii_start()" is executed, it fetches the initialization information using "nii_get_init_info()".

This ""nii_get_init_info" issues a callback of our UNDI driver ""UNDI_GetInitInfo" which sets the status flag "stat".

The link status depends on a "media" flag:

nii->media = ( stat & PXE_STATFLAGS_GET_STATUS_NO_MEDIA_SUPPORTED );

Here, the media flag gets the valid value of "0x0002."

Now, as per the condition in "nii_start()":

	/* Set initial link state (if media detection is not supported) */
		if ( ! nii->media )
		    netdev_link_up ( netdev );

The comment says the initial "link state" will be set if "media detection is not supported."

In our case, the condition is as per the comment, so the link should be set. But since "nii->media" has a non-zero value, the "NOT(!)" operator fails the if condition, and the link is not being set up.

This is why we are getting the initial link status as "down."

So here the comment and the "if" condition seems to be contradicting each other.
So is this a desired behaviour? Can anyone comment on this?

NiKiZe, 2024-05-28:
If you enter ipxe shell, or is using an embedded script and check ifstat, then all interfaces in iPXE is expected to be down, use ifopen, dhcp, or autoboot to bring the interface up, does that work?

PaVaSjAiN, 2024-05-29:
When DHCP is configured from the iPXE shell the link status is seen "UP".
 
But my querry is why the link status is "DOWN" in the initial banner?

NiKiZe, 2024-05-29:
All interfaces in iPXE is down until they are opened in one way or another.

So what are you using to run ifstat? Embedded script?

PaVaSjAiN, 2024-05-29:
I am not using any Embedded script, just using chainloading method to load ipxe.efi from DHCP server
Then after the initial banner appears, it configures the network.
Then i am entering to the iPXE shell using "Esc + b" and checking the "ifstat", here the link is showing up.
[image]

mcb30, 2024-05-29:
> So here the comment and the "if" condition seems to be contradicting each other. So is this a desired behaviour? Can anyone comment on this?

This issue is a perfect example of why flag constants should **never** be named negatively (with "no" or "not" in the name).

The actual media detection is performed by the `PXE_OPCODE_GET_STATUS` call.  The UEFI specification describes the returned status flags as including:

> This flag is set if there is no media present:
>
> `PXE_STATFLAGS_GET_STATUS_NO_MEDIA`

So, the relevant status flag is the "NO_MEDIA" flag, used to indicate that media is absent.  A driver that supports media detection will set the "NO_MEDIA" flag if it detects that media is absent, or leave the flag clear if media is present.  A driver that does not support media detection will will (implicitly) never touch this flag and so it will be left as clear.

Now, let's look at the returned status flags from the `PXE_OPCODE_GET_INIT_INFO` call.  The UEFI specification is not at all explicit here, stating only:

> To determine if cable detection is supported by this UNDI/NIC, use these macros with the CDB.StatFlags field:
>
> `PXE_STATFLAGS_CABLE_DETECT_MASK`
> `PXE_STATFLAGS_CABLE_DETECT_NOT_SUPPORTED`
> `PXE_STATFLAGS_CABLE_DETECT_SUPPORTED`
> `PXE_STATFLAGS_GET_STATUS_NO_MEDIA_MASK`
> `PXE_STATFLAGS_GET_STATUS_NO_MEDIA_NOT_SUPPORTED`
> `PXE_STATFLAGS_GET_STATUS_NO_MEDIA_SUPPORTED`

(As an aside: the name `PXE_STATFLAGS_GET_STATUS_NO_MEDIA_NOT_SUPPORTED` is an abomination that should have caused the author to stop and reconsider their thought process, but it's in the specification so we're stuck with it.)

The question that arises is how to interpret `PXE_STATFLAGS_GET_STATUS_NO_MEDIA_SUPPORTED`.  When read in isolation, it could legitimately be interpreted as meaning "no media detection is supported", i.e. that "media detection is **not** supported".

However, when taken in context of the surrounding definitions and the description of the relevant `PXE_STATFLAGS_GET_STATUS_NO_MEDIA` flag, it can only be interpreted to mean "the NO_MEDIA flag is supported", i.e. that "media detection **is** supported".

The logic in iPXE is therefore correct.  If your driver does not support media detection, then it needs to leave the `PXE_STATFLAGS_GET_STATUS_NO_MEDIA_SUPPORTED` flag clear in the response to `PXE_OPCODE_GET_INIT_INFO`.

closed 2024-05-29 by mcb30

PaVaSjAiN, 2024-05-29:
Hi Michael,

Thank you for such a detailed clafication, i will update my driver accordingly.

Warm Regards,
Pavas Jain