ipxe/ipxe #1239: Booting Linux with long kernel arguments fails

mikroskeem, 2024-06-18:
# Issue

Loading aarch64 Linux over HTTP or HTTPS and having kernel arguments over 283 characters hangs booting.
Similar issue appears to exist on x86_64, but I haven't tried narrowing it down to exact lengths yet.

# Environment

Commit e965f179e1654103eca33feed7a9cc4c51d91be6, `make bin-arm64-efi/ipxe.efi`

Built using Ubuntu 24.04 (let me know if you need more details about toolchains)

Running on QEMU emulator version 8.2.4, from nixpkgs `24.05.20240422.1e1dc66` on macOS

<details>
<summary>config.h</summary>

```patch
diff --git a/src/config/general.h b/src/config/general.h
index e883e072..5e7c99c7 100644
--- a/src/config/general.h
+++ b/src/config/general.h
@@ -56,7 +56,7 @@ FILE_LICENCE ( GPL2_OR_LATER_OR_UBDL );

 #define	DOWNLOAD_PROTO_TFTP	/* Trivial File Transfer Protocol */
 #define	DOWNLOAD_PROTO_HTTP	/* Hypertext Transfer Protocol */
-#undef	DOWNLOAD_PROTO_HTTPS	/* Secure Hypertext Transfer Protocol */
+#define	DOWNLOAD_PROTO_HTTPS	/* Secure Hypertext Transfer Protocol */
 #undef	DOWNLOAD_PROTO_FTP	/* File Transfer Protocol */
 #undef	DOWNLOAD_PROTO_SLAM	/* Scalable Local Area Multicast */
 #undef	DOWNLOAD_PROTO_NFS	/* Network File System Protocol */
@@ -71,7 +71,7 @@ FILE_LICENCE ( GPL2_OR_LATER_OR_UBDL );
 //#undef	SANBOOT_PROTO_AOE	/* AoE protocol */
 //#undef	SANBOOT_PROTO_IB_SRP	/* Infiniband SCSI RDMA protocol */
 //#undef	SANBOOT_PROTO_FCP	/* Fibre Channel protocol */
-//#undef	SANBOOT_PROTO_HTTP	/* HTTP SAN protocol */
+#define	SANBOOT_PROTO_HTTP	/* HTTP SAN protocol */

 /*
  * HTTP extensions
```

</details>

<details>
<summary>autoexec.ipxe</summary>

```
#!ipxe

set STREAM stable
set RELEASE 40.20240519.3.0
set INSTALLDEV /dev/sda
set CONFIGURL https://example.com/example.ign
set SERIAL_CONSOLE ttyAMA0,115200

set BASEURL https://builds.coreos.fedoraproject.org/prod/streams/${STREAM}/builds/${RELEASE}/aarch64
#set BASEURL http://10.0.2.2:8888/fedora/fedora/fedora/fedora/fedora/fedora/fedora/fedora/fedora/fedora
set KERNELURL ${BASEURL}/fedora-coreos-${RELEASE}-live-kernel-aarch64
set INITRDURL ${BASEURL}/fedora-coreos-${RELEASE}-live-initramfs.aarch64.img
set ROOTFSURL ${BASEURL}/fedora-coreos-${RELEASE}-live-rootfs.aarch64.img

dhcp
route
show dns

kernel ${KERNELURL} initrd=main console=tty0 console=${SERIAL_CONSOLE} coreos.live.rootfs_url=${ROOTFSURL} coreos.inst.install_dev=${INSTALLDEV} coreos.inst.ignition_url=${CONFIGURL} ignition.platform.id=metal
initrd --name main ${INITRDURL}

imgstat

boot
```

</details>

# Reproduction

Set up iPXE with following script and try to boot. You'll see it hang after `EFI Stub` messages like this:

```
iPXE 1.21.1+ (ge965) -- Open Source Network Boot Firmware -- https://ipxe.org
Features: DNS HTTP HTTPS iSCSI TFTP VLAN SRP AoE EFI Menu
Configuring (net0 52:54:00:12:34:56)...... ok
net0: 10.0.2.15/255.255.255.0 gw 10.0.2.2
net0: fec0::5054:ff:fe12:3456/64 gw fe80::2
net0: fe80::5054:ff:fe12:3456/64
net0.dhcp/dns:ipv4 = 10.0.2.3
https://builds.coreos.fedoraproject.org/prod/streams/stable/builds/40.20240519.3.0/aarch64/fedora-coreos-40.20240519.3.0-live-kernel-aarch64..... ok
https://builds.coreos.fedoraproject.org/prod/streams/stable/builds/40.20240519.3.0/aarch64/fedora-coreos-40.20240519.3.0-live-initramfs.aarch64.img... ok
fedora-coreos-40.20240519.3.0-live-kernel-aarch64 : 17842688 bytes [EFI] [SELECTED] "initrd=main console=tty0 console=ttyAMA0,115200 coreos.live.rootfs_url=https://builds.coreos.fedoraproject.org/prod/streams/stable/builds/40.20240519.3.0/aarch64/fedora-coreos-40.20240519.3.0-live-rootfs.aarch64.img coreos.inst.install_dev=/dev/sda coreos.inst.ignition_url=https://example.com/example.ign ignition.platform.id=metal"
main : 77729772 bytes
EFI stub: Decompressing Linux Kernel...
EFI stub: Loaded initrd from LINUX_EFI_INITRD_MEDIA_GUID device path
EFI stub: Generating empty DTB
EFI stub: Exiting boot services...
```

Alternatively, turns out this can be done using 3rd party https://github.com/netbootxyz/netboot.xyz/releases/download/2.0.79/netboot.xyz-multiarch.iso image as well, it has Fedora CoreOS in the menu for convenient reproduction.

NiKiZe, 2024-06-18:
Since you have "EFI stub:" messages, which does not come from iPXE, did you verify that this only happens in iPXE? My guess is that the issue is with the kernel, not iPXE.

mikroskeem, 2024-06-18:
Good catch, this does not appear to be iPXE problem. Tried to boot via UEFI shell and GRUB, and it hangs in a similar way.

For reference, both `edk2-stable202302-for-qemu` from nixpkgs and `edk2-stable202402-prebuilt.qemu.org` from upstream qemu repo have the same issue, if anyone stumbles upon this.

Closing for now.

closed 2024-06-18 by mikroskeem