ipxe/ipxe #1247: Mellanox nic use commit 4bd064d ipxe.efi will happen DHCPDISCOVER failed,but use the latest master ipxe.efi will work ok

xcc-123, 2024-06-29:
Mellanox nic use commit 4bd064d ipxe.efi will happen DHCPDISCOVER failed,

but use the latest master ipxe.efi will work ok

if i let nic paramter KEEP_ETH_LINK_UP_P1=TRUE,it also can use commit 4bd064d ipxe.efi,it will work ok
but set KEEP_ETH_LINK_UP_P1=FALSE, use the latest master ipxe.efi can work.

why？ which possible commit can solve it?

xcc-123, 2024-06-29:
DHCPDISCOVER failed,:
[image]
the latest master ipxe.efi will work ok:
[image]

NiKiZe, 2024-06-29:
If this is fixed in current master, why are you opening an issue? Just use master!
There is discussions that can be used for these kind of questions. Also always use the search function.

Is there any reason why you would be using a commit from 2020 at all?

Use git bisect if you want to do the research.