ipxe/ipxe #1259: Menus with Items Display Off over COM ports

ericspidledw, 2024-07-16:
[image]
[image]

When using iPXE menu's over a COM port the lines with items on them seem to be off some pixels and there are also some extra background color pixels above and below the menu itself. Not entirely sure what causes this and I know the ipxe commands are fine because on a real display these menus show up as expected

ericspidledw, 2024-07-16:
The bottom one is my own custom menu and the top one I took from the NIC config choice example [here ](https://gist.github.com/robinsmidsrod/3871687)just to make sure it wasn't my specific menu configuration

stappersg, 2024-08-05:
_`acknowledge`_ on the observation. Please express what are the expections from the `iPXE`-project. (In other words: Please explain why this issue should remain open.)

NiKiZe, 2024-09-10:
When using serial, there is only text. (Pixels becomes a confusing term)
This might as well be a bug in your serial terminal.
Which terminal are you using, and what size does the terminal window have, is it exactly 80x25 or something else? (If not exactly the same as native, it is quite common with a number of rendering issues)
Also is this thru iPXE native serial support, or thru you bios/firmware serial console?

Problem description as I see it, menu items lacks the background color in the last column, there is also other places where background color is missing over serial.

mcb30, 2024-09-11:
Your serial terminal emulator is failing to apply the background colour when clearing the screen, so that you end up with only the explicitly drawn characters having a background.

What serial terminal emulator are you using?

NiKiZe, 2024-09-11:
Thanks for the info, closing as it's not an issue with iPXE

closed 2024-12-11 by zube[bot]