ipxe/ipxe #1261: Support OCSP Stapling

Cornelicorn, 2024-07-17:
Supporting OCSP stapling would allow using https on systems that don't have an internet connection without setting up an OCSP proxy by just using what common webserver provide out of the box.

NiKiZe, 2024-07-17:
I'll just link to existing plausibly relevant OCSP topics, (and hide this comment as of topic to have it focused on stapling)
#127
#623

NiKiZe, 2026-08-19:
OCSP support is suggested to be dropped , one source https://developers.google.com/public-key-infrastructure/updates/april2025-ocsp-notice.
Another source which seems of on the date: https://youtu.be/-9KiLFr8_hI?t=1781
If that is the case then iPXE should maybe follow and this feature request would become irrelevant?