ipxe/ipxe #1274: dhcp + multi interfaces problem

maxatome, 2024-08-08:
Hi,

We have a motherboard with several network cards, as `ifstat` shows in logs below. Only `net2` interface can reach a DHCP server.

Using the following embedded `embed.ipxe` file:
```
#!ipxe
echo DHCPing
dhcp
ifstat
echo net0 IP address: ${net0/ip}
echo net2 IP address: ${net2/ip}
chain http://${next-server}/ipxe/main.ipxe
```

iPXE is compiled as:
```
make -j DEBUG=dhcp EMBED=embed.ipxe bin-x86_64-efi/ipxe.efi
```

And resulting `src/bin-x86_64-efi/ipxe.efi` is served using TFTP.

With master (59e2b03e), `dhcp` succeeds for `net0` interface, while there is no DHCP server on this network. For a reason I ignore, `net2` interface has an IP address, the same DHCP server would provide for a working iPXE image:
```
>>Checking Media Presence......
>>Media Present......
 Downloading NBP file...

  NBP file downloaded successfully.
iPXE initialising devices...
autoexec.ipxe... Not found (https://ipxe.org/2d12618e)

iPXE 1.21.1+ (g59e2) -- Open Source Network Boot Firmware -- https://ipxe.org
Features: DNS HTTP iSCSI TFTP VLAN SRP AoE EFI Menu
DHCPing
DHCP 0x96dfd168 entering discovery state
Configuring (net0 94:6d:ae:***c)...DHCP 0x96dfd168 DHCPDISCOVER
.DHCP 0x96dfd168 DHCPDISCOVER
..DHCP 0x96dfd168 DHCPDISCOVER
....DHCP 0x96dfd168 DHCPDISCOVER
........ ok
net0: 94:6d:ae:***c using ConnectX-7 on 0000:01:00.0 (Ethernet) [open]
  [Link:up, TX:6 TXE:0 RX:3 RXE:0]
net1: 94:6d:ae:***d using ConnectX-7 on 0000:01:00.1 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net2: 58:a2:e1:***2 using ConnectX-6Lx on 0000:61:00.0 (Ethernet) [closed]
  [Link:down, TX:0 TXE:1 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
  [TXE: 1 x "Network unreachable (https://ipxe.org/28086090)"]
net3: 58:a2:e1:***3 using ConnectX-6Lx on 0000:61:00.1 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net4: *** using 14e4-1760 on 0000:81:00.0 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net5: *** using NII on NII-0000:65:00.4 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net0 IP address: 
net2 IP address: 10.1.2.3
http://***/ipxe/main.ipxe... Network unreachable (https://ipxe.org/280a6090)
```

Commit 170bbfd4 is the last commit for which `dhcp` works correctly: `net0` and `net1` fail to find DHCP server, but `net2` succeeds and the boot is OK. So it starts to fail for commit 9bbe7766.
```
>>Checking Media Presence......
>>Media Present......
 Downloading NBP file...

  NBP file downloaded successfully.
iPXE initialising devices...ok

iPXE 1.21.1+ (g170bb) -- Open Source Network Boot Firmware -- https://ipxe.org
Features: DNS HTTP iSCSI TFTP VLAN SRP AoE EFI Menu
DHCPing
DHCP 0x96b7d028 entering discovery state
Configuring (net0 94:6d:ae:***c)...DHCP 0x96b7d028 DHCPDISCOVER
.DHCP 0x96b7d028 DHCPDISCOVER
..DHCP 0x96b7d028 DHCPDISCOVER
....DHCP 0x96b7d028 DHCPDISCOVER
........ No configuration methods succeeded (https://ipxe.org/040ee186)
DHCP 0x96b7d028 entering discovery state
Configuring (net1 94:6d:ae:***d)...DHCP 0x96b7d028 DHCPDISCOVER
DHCP 0x96b7d028 DHCPDISCOVER
...DHCP 0x96b7d028 DHCPDISCOVER
....DHCP 0x96b7d028 DHCPDISCOVER
........ No configuration methods succeeded (https://ipxe.org/040ee186)
DHCP 0x96b7d028 entering discovery state
Configuring (net2 58:a2:e1:***2)...DHCP 0x96b7d028 DHCPDISCOVER
DHCP 0x96b7d028 DHCPOFFER from 10.1.2.10:67 (***1/***1) for 10.1.2.3
DHCP 0x96b7d028 DHCPOFFER from 10.1.2.11:67 (***1/***1) for 10.1.2.3
DHCP 0x96b7d028 DHCPOFFER from 10.1.2.11:67 (***1/***1) for 10.1.2.3
DHCP 0x96b7d028 DHCPOFFER from 10.1.2.10:67 (***1/***1) for 10.1.2.3
.DHCP 0x96b7d028 DHCPDISCOVER
DHCP 0x96b7d028 DHCPOFFER from 10.1.2.11:67 (***2/***2) for 10.1.2.3
DHCP 0x96b7d028 DHCPOFFER from 10.1.2.10:67 (***2/***2) for 10.1.2.3
DHCP 0x96b7d028 DHCPOFFER from 10.1.2.11:67 (***1/***1) for 10.1.2.3
DHCP 0x96b7d028 DHCPOFFER from 10.1.2.10:67 (***1/***1) for 10.1.2.3
..DHCP 0x96b7d028 entering request state
DHCP 0x96b7d028 DHCPREQUEST to ***1:67 for 10.1.2.3
DHCP 0x96b7d028 DHCPACK from 10.1.2.10:67 (***2/***2) for 10.1.2.3
DHCP 0x96b7d028 DHCPACK from 10.1.2.11:67 (***2/***2) for 10.1.2.3
DHCP 0x96b7d028 DHCPACK from 10.1.2.11:67 (***1/***1) for 10.1.2.3
 ok
net0: 94:6d:ae:***c using ConnectX-7 on 0000:01:00.0 (Ethernet) [closed]
  [Link:down, TX:4 TXE:0 RX:1 RXE:64]
  [Link status: Down (https://ipxe.org/38086193)]
  [RXE: 64 x "Operation canceled (https://ipxe.org/0b136095)"]
net1: 94:6d:ae:***d using ConnectX-7 on 0000:01:00.1 (Ethernet) [closed]
  [Link:down, TX:4 TXE:0 RX:1 RXE:64]
  [Link status: Down (https://ipxe.org/38086193)]
  [RXE: 64 x "Operation canceled (https://ipxe.org/0b136095)"]
net2: 58:a2:e1:***2 using ConnectX-6Lx on 0000:61:00.0 (Ethernet) [open]
  [Link:up, TX:6 TXE:1 RX:143 RXE:16]
  [TXE: 1 x "Network unreachable (https://ipxe.org/28086090)"]
  [RXE: 5 x "Operation not supported (https://ipxe.org/3c086083)"]
  [RXE: 7 x "The socket is not connected (https://ipxe.org/380a6093)"]
  [RXE: 4 x "The socket is not connected (https://ipxe.org/380f6093)"]
net3: 58:a2:e1:***3 using ConnectX-6Lx on 0000:61:00.1 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net4: *** using 14e4-1760 on 0000:81:00.0 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net5: *** using NII on NII-0000:65:00.4 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net0 IP address: 
net2 IP address: 10.1.2.3
http://***/ipxe/main.ipxe... ok
```

Does anyone has an idea of what happens?

Thanks.

stappersg, 2024-08-21:
This is not an iPXE issue.  ( This _"github issue"_ should **be closed** to prevent further draining of human energy from the iPXE community. )

maxatome, 2024-08-22:
Hello @stappersg, perhaps could you please elaborate a little bit?
As it works for a given commit, and not for the next one, it is not obvious to me it is not an iPXE issue...

NiKiZe, 2024-08-22:
Looking at your output again and trying to make sense of it, I can see that you have multiple DHCP offers from different servers. That is not a valid setup, or at least such a behaviour is undefined, and thus unsupported.

maxatome, 2024-08-22:
@NiKiZe you are right, we use 2 DHCP servers, but when the problem occurs, these 2 DHCP servers are not involved at all: net0 cannot receive their response as they are not reachable on net0 network.

Both given examples use the same hardware setup, only the iPXE version changes:
- the first example works (even with 2 DHCP servers, but it is not the point): it matches the commit 170bbfd4875b2a3479101b5f5892cdedcf857fd0
- the second example does not work (without receiving any DHCP response from anywhere): it matches the commit 9bbe77669c6e2b71826449d854f5aa0e2cee7767

The problem here is that starting 9bbe77669c6e2b71826449d854f5aa0e2cee7767 net0 receives no response to its DHCPDISCOVER from any DHCP server, but iPXE still says "ok". While before this commit, it skips this interface and try net1, and so on.

I understand that multiple DHCP offers are not supported, but if you look at the 2nd example (the one for which the problem occurs), these multiple offers are never received. So how this problem can be related to multiple DHCP offers?

Thanks for your help.

maxatome, 2024-08-22:
Hello @mcb30, sorry to bother you but as you are the author of 9bbe776 (that seems to introduce the problem), perhaps could you have an explanation.

The steps follow:
- the host boots, BIOS is configured to reach a DHCP server on **net2**, and it succeeds to obtain an IP : 10.1.2.3;
- BIOS successfully downloads iPXE binary over TFTP using this IP;
- iPXE starts, then DHCP discovers on **net0**, without any response (which normal, as there is no DHCP server on this network);
- when the DHCP discovery times out on **net0**, instead of failing and switching to next interface, it terminates loop and says "ok".

However, as seen in logs of `ifstat`, **net0** has no IP and is open while **net2** has an IP and is closed.

I think that perhaps the **net2** IP, the BIOS gets in a first place, is taken by iPXE as a _"I have an IP all is OK"_, while this IP is not on the right interface. So iPXE does not loop on next interface while it should.

NiKiZe, 2024-08-22:
What exactly does the DHCP pack contains that gets cached before iPXE boots? Or how was iPXE loaded in the first place? (Static settings in firmware?)

stappersg, 2024-08-22:
> Commit 170bbfd is the last commit for which dhcp works correctly: net0 and net1 fail to find DHCP server, but net2 succeeds and the boot is OK. So it starts to fail for commit 9bbe776.

https://ipxe.org/howto/bisect

```text
stappers@paddy:~/src/githubforks/ipxe
$ git bisect start
status: waiting for both good and bad commits
stappers@paddy:~/src/githubforks/ipxe
$ git bisect good 170bbfd 
status: waiting for bad commit, 1 good commit known
stappers@paddy:~/src/githubforks/ipxe
$ git bisect bad 9bbe776
9bbe77669c6e2b71826449d854f5aa0e2cee7767 is the first bad commit
commit 9bbe77669c6e2b71826449d854f5aa0e2cee7767
Author: Michael Brown <mcb30@ipxe.org>
Date:   Tue Mar 26 15:17:23 2024 +0000

    [efi] Extract basic network settings from loaded image device path
    
    The UEFI HTTP boot mechanism is extraordinarily badly designed, even
    by the standards of the UEFI specification in general.  It has the
    symptoms of a feature that has been designed entirely in terms of user
    stories, without any consideration at all being given to the
    underlying technical architecture.  It does work, provided that you
    are doing precisely and only what was envisioned by the product owner.
    If you want to try anything outside the bounds of the product owner's
    extremely limited imagination, then you are almost certainly about to
    enter a world of pain.
    
    As one very minor example of this: the cached DHCP packet is not
    available when using HTTP boot.  The UEFI HTTP boot code does perform
    DHCP, but it pointlessly and unhelpfully throws away the DHCP packet
    and trashes the network interface configuration before handing over to
    the downloaded executable.
    
    Work around this imbecility by parsing and applying the few network
    configuration settings that are persisted into the loaded image's
    device path.  This is limited to very basic information such as the IP
    address, gateway address, and DNS server address, but it does at least
    provide enough for a functional routing table.
    
    Signed-off-by: Michael Brown <mcb30@ipxe.org>

 src/include/ipxe/settings.h  |   2 +
 src/interface/efi/efi_path.c | 241 +++++++++++++++++++++++++++++++++++++++++++
 2 files changed, 243 insertions(+)
stappers@paddy:~/src/githubforks/ipxe
$
```

(Consider me in lurking mode)

ChlorideCull, 2024-08-29:
It returns `ok` because this check passes, which passes if there's a return code of 0 for DHCP (or NDP, but I guess you don't have IPv6 set up on net0)

https://github.com/ipxe/ipxe/blob/748cab7745186ec6c770fb4d47b0e8c9f213e6df/src/usr/ifmgmt.c#L258-L259

Why would there be a return code of 0? Presumably, because `efi_path` is pulling something from your EFI firmware and setting the DHCP settings to that (because that's what it does if it finds something).

@maxatome Try enabling debug for `efi_path`, and also check what's set in `net0/dhcp/` with `config`. This mechanism is also probably why your `net2` interface magically gets the IP that was assigned to it during PXE.

I'll leave the rest to the experts, I'm done theorycrafting :)