ipxe/ipxe #1275: Stopping when using the RTL8111H network card

ask0rbin, 2024-08-08:
The loading of snponly.efi stops during startup. The assembly was performed using the command:
`make bin-x86_64-efi/snponly.efi EMBED=start.ipxe DEBUG="pci,nii,snp,snpnet,snponly,efi_driver,efi_debug,efi_init,efi_pci"`

Debag output:
```
...
EFIPCI 0000:2c:00.0 (10ec:8168 class 020000) has no driver
...
CHAINED PciRoot (0x0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/MAC(D843AE755C3E,0x0) is the chainloaded ManagedNetworkSb
EFIDRV PciRoot (0x0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/MAC(D843AE755C3E,0x0) has driver "MNPONLY"
EFIDRV PciRoot (0x0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/MAC(D843AE755C3E,0x0) disconnecting existing drivers
```

My network card is a realtek rtl8111h. vendor_id:device_id  - 10ec:8168 (rev 15)
lspci output:
```
2c:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 15)
```

The INDY Driver version in the BIOS:
```
Driver Name: Realtek UEFI UNDI Driver
Driver Version: 2.059
Driver Released Date: 2022/06/17
```

I've tried different build options:
**snp.efi**, **ipxe.efi** **10ec8168.efi**

NiKiZe, 2024-08-08:
If `snponly.efi` fails you can be almost certain that this is a bug in the firmware/ROM.

There is a veto list that could be used, but the 10ec:8168 is so common and often works just fine.

Reach out to your hardware manufacturer and report this issue to them.

I assume that with no debug options at buildtime, you have no output from iPXE at all?

NiKiZe, 2024-12-09:
Is this NIC on board? If so maybe this is more about firmware than the actual nic.

Maybe try and see if you are affected by the issue as mentioned in https://github.com/ipxe/ipxe/discussions/1348#discussioncomment-11507064

NiKiZe, 2026-04-21:
Stale, happy to reopen if new information comes in.

closed 2026-04-21 by NiKiZe