ipxe/ipxe #1289: Build failure with gcc-14: error: type defaults to ‘int’ in declaration of ‘FILE_LICENCE’ [-Wimplicit-int]

panlinux, 2024-08-28:
I'm trying to troubleshoot a build failure that is only happening with gcc-14. The same checkout, environment, OS, everything, works when I just switch to gcc-13.

With gcc-14. it's like the `include/compiler.h` file is not included for some reason. That missed the `FILE_LICENSE` macro definition, and the build fails at this step:
```
gcc -Wall -W -Wformat-nonliteral -O2 -g -I/usr/include -I/usr/include -I/usr/include -idirafter include -DEFI_TARGET_X64 util/elf2efi.c \
    -L/usr/lib -L/usr/lib -L/usr/lib -lbfd -ldl -lz -Wl,--no-warn-search-mismatch -o util/elf2efi64
In file included from util/elf2efi.c:33:
include/ipxe/efi/efi.h:24:1: error: type defaults to ‘int’ in declaration of ‘FILE_LICENCE’ [-Wimplicit-int]
   24 | FILE_LICENCE ( GPL2_OR_LATER );
      | ^~~~~~~~~~~~
```
With gcc-13, the exact same command -line works (just spits out a different warning, and moves on to the next file):
```
gcc-13 -Wall -W -Wformat-nonliteral -O2 -g -I/usr/include -I/usr/include -I/usr/include -idirafter include -DEFI_TARGET_X64 util/elf2efi.c \
    -L/usr/lib -L/usr/lib -L/usr/lib -lbfd -ldl -lz -Wl,--no-warn-search-mismatch -o util/elf2efi64
util/elf2efi.c: In function ‘elf2pe’:
util/elf2efi.c:369:9: warning: ‘__builtin_strncpy’ specified bound 8 equals destination size [-Wstringop-truncation]
  369 |         strncpy ( ( char * ) new->hdr.Name, section->name,
      |         ^
gcc-13 -Wall -W -Wformat-nonliteral -O2 -g -idirafter include -o util/efirom util/efirom.c
...
```
This is with the ubuntu package `ipxe-qemu-256k-compat_1.0.0+git-20150424.a25a16d.orig.tar.gz`[1]. It's the first time I'm troubleshooting this package, it looks like it's a git snapshot.

The build command in the ends seems to be this, from `debian/rules`:

```
cd src && make -j1 V=1 NO_WERROR=1 VERSION=1.0.0\+git-20150424.a25a16d-0ubuntu5 bin/82540em.rom
```

1. https://launchpad.net/ubuntu/+source/ipxe-qemu-256k-compat/1.0.0+git-20150424.a25a16d-0ubuntu5

NiKiZe, 2024-08-29:
Which is the exact GCC versions you are using?
GCC 14 reproduced issue in #1271

panlinux, 2024-08-29:
```
$ gcc --version
gcc (Ubuntu 14.2.0-3ubuntu1) 14.2.0
Copyright (C) 2024 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

mcb30, 2024-09-02:
This was fixed **approximately nine years ago** in commit https://github.com/ipxe/ipxe/commit/fdad22a1e

closed 2024-09-02 by mcb30

panlinux, 2024-09-02:
Oh, wow, I did try building master before, but it failed elsewhere unrelated. Sorry for not spotting the fix before, and it's truly shameful that ubuntu's package is so old.

panlinux, 2024-09-02:
Ok, got some history about this package in Ubuntu. It all started with a migration error and rom sizes (https://bugs.launchpad.net/ubuntu/+source/qemu/+bug/1713490), and since then ubuntu has shipped a "compat" variant of ipxe meant for those old machines, and this is the one that was failing to build. Proper ipxe non-compat package is way more current.

Sorry for the noise here, and thanks for the pointer to the patch!

mcb30, 2024-09-02:
> Oh, wow, I did try building master before, but it failed elsewhere unrelated. Sorry for not spotting the fix before, and it's truly shameful that ubuntu's package is so old.

:upside_down_face:

You might have been hitting #1219 building with GCC 14 on current master, due an issue that was fixed just now in commit https://github.com/ipxe/ipxe/commit/7f75d320f.  Could you retry on current master and see if it now works for you?

panlinux, 2024-09-02:
Right, that's the one I saw in master earlier. With 7f75d320f6d8ac7ec5185b2145da87f698aec273, it builds fine with gcc 14.2.0, thanks!