ipxe/ipxe #1293: GCC14 issues

purpleidea, 2024-08-30:
Similar to: https://github.com/ipxe/ipxe/issues/620

I get the following error:

```
  [BUILD] bin-x86_64-efi/acpi.o
In file included from include/string.h:23,
                 from include/ipxe/uaccess.h:25,
                 from core/acpi.c:28:
In function ‘__constant_memcpy’,
    inlined from ‘memcpy’ at arch/x86/include/bits/string.h:168:10,
    inlined from ‘trivial_memcpy_user’ at include/ipxe/uaccess.h:109:2,
    inlined from ‘memcpy_user’ at include/ipxe/efi/efi_uaccess.h:69:2,
    inlined from ‘copy_from_user’ at include/ipxe/uaccess.h:338:2,
    inlined from ‘acpi_sx_zsdt’ at core/acpi.c:217:3:
arch/x86/include/bits/string.h:79:32: error: array subscript ‘union <anonymous>[0]’ is partly outside array bounds of ‘union <anonymous>[1]’ [-Werror=array-bounds=]
   79 |                 dest_u->u32[0] = src_u->u32[0];
      |                 ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
core/acpi.c: In function ‘acpi_sx_zsdt’:
core/acpi.c:191:11: note: object ‘buf’ of size 4
  191 |         } buf;
      |           ^~~
cc1: all warnings being treated as errors
make: *** [Makefile.housekeeping:943: bin-x86_64-efi/acpi.o] Error 1

```

I tried this with version tag: `v1.21.1` but it has the same issue with git master.

I'm using gcc:

```
gcc --version
gcc (GCC) 14.2.1 20240801 (Red Hat 14.2.1-1)
...
```

On Fedora 40.

I can make a new issue for this if you prefer. You can this error running `make` or make with some bin/... target.

HTH

purpleidea, 2024-08-30:
FYI, running with NO_ERROR works around this bug:

```
make NO_WERROR=1 bin/ipxe.usb
```

For example...

stappersg, 2024-09-01:
> Similar to: #620

and simular to #1219

mcb30, 2024-09-02:
> ```
> arch/x86/include/bits/string.h:79:32: error: array subscript ‘union <anonymous>[0]’ is partly outside array bounds of ‘union <anonymous>[1]’ [-Werror=array-bounds=]
> ```
> 
> ```
> gcc --version
> gcc (GCC) 14.2.1 20240801 (Red Hat 14.2.1-1)
> ...
> ```
> 
> On Fedora 40.

I am unable to reproduce this error on Fedora 40, with the exact same version of `gcc`.

Your output shows a warning reported by `-Warray-bounds`.  This warning was explicitly disabled in commit https://github.com/ipxe/ipxe/commit/54c4c1d40 since it generates a very large number of false positives.

Please ensure that you are building from the latest git master, and that you are **not** customising `CFLAGS` to re-enable any warnings that are explicitly disabled in the iPXE Makefile.

closed 2024-09-02 by mcb30

purpleidea, 2024-09-03:
If you look at my above comment I wrote:

> I tried this with version tag: v1.21.1 but it has the same issue with git master.

Here's the exact error with git master:

```
  [BUILD] bin/etherfabric.o
In file included from drivers/net/etherfabric.c:36:
In function ‘falcon_xaui_link_ok’,
    inlined from ‘falcon_init_xmac’ at drivers/net/etherfabric.c:2355:14,
    inlined from ‘falcon_init_xmac’ at drivers/net/etherfabric.c:2314:1:
drivers/net/etherfabric.h:524:28: error: ‘reg.u32[0]’ may be used uninitialized [-Werror=maybe-uninitialized]
  522 |         (dword).u32[0] = ( ( (dword).u32[0]                                   \
      |                          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  523 |                              & ~EFAB_INPLACE_MASK32 (  0,  31, field ) )      \
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  524 |                            | EFAB_INSERT_FIELD32 (  0,  31, field, value ) ); \
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/net/etherfabric.c:2233:9: note: in expansion of macro ‘EFAB_SET_DWORD_FIELD’
 2233 |         EFAB_SET_DWORD_FIELD ( reg, FCN_XX_DISPERR, FCN_XX_DISPERR_RESET);
      |         ^~~~~~~~~~~~~~~~~~~~
drivers/net/etherfabric.c: In function ‘falcon_init_xmac’:
drivers/net/etherfabric.c:2213:22: note: ‘reg.u32[0]’ was declared here
 2213 |         efab_dword_t reg;
      |                      ^~~
cc1: all warnings being treated as errors
make: *** [Makefile.housekeeping:964: bin/etherfabric.o] Error 1

```

purpleidea, 2024-09-03:
(Which as of this writing is 301644ab480ab9787c617e1a94e19ca5c2774072)

mcb30, 2024-09-03:
> (Which as of this writing is 301644ab480ab9787c617e1a94e19ca5c2774072)

No, it isn't. As of the time of your comment, current git master was https://github.com/ipxe/ipxe/commit/7f75d320f6d8ac7ec5185b2145da87f698aec273, which is the fix for the issue that you have most recently reported in this thread (which is entirely separate from the issue originally reported in this thread).

purpleidea, 2024-09-03:
> No, it isn't. As of the time of your comment
You're totally right, my bad, sorry! It was the right sha1 from when I opened the issue, I forgot to `git pull`  since originally opening the ticket.

In any case, _actual_ git master now builds without errors, and sorry for the noise. I hope my original report was useful.

Thanks for the project, in the odd chance you're curious, I've been playing with ipxe so that machines that can't pxe/uefi netboot normally for various reasons can still use the https://github.com/purpleidea/mgmt/ provisioner as described here:

https://purpleidea.com/blog/2024/03/27/a-new-provisioning-tool/