ipxe/ipxe #1299: Compilation issue

aarnoud, 2024-09-10:
Applying this configuration:
```diff
diff --git a/src/config/console.h b/src/config/console.h
index 9f770d0..37ab4c7 100644
--- a/src/config/console.h
+++ b/src/config/console.h
@@ -34,8 +34,8 @@ FILE_LICENCE ( GPL2_OR_LATER_OR_UBDL );
  *
  */
 
-//#define	CONSOLE_SERIAL		/* Serial port console */
-//#define	CONSOLE_FRAMEBUFFER	/* Graphical framebuffer console */
+#define	CONSOLE_SERIAL		/* Serial port console */
+#define	CONSOLE_FRAMEBUFFER	/* Graphical framebuffer console */
 //#define	CONSOLE_SYSLOG		/* Syslog console */
 //#define	CONSOLE_SYSLOGS		/* Encrypted syslog console */
 //#define	CONSOLE_VMWARE		/* VMware logfile console */
diff --git a/src/config/general.h b/src/config/general.h
index b82e560..72316ee 100644
--- a/src/config/general.h
+++ b/src/config/general.h
@@ -56,7 +56,7 @@ FILE_LICENCE ( GPL2_OR_LATER_OR_UBDL );
 
 #define	DOWNLOAD_PROTO_TFTP	/* Trivial File Transfer Protocol */
 #define	DOWNLOAD_PROTO_HTTP	/* Hypertext Transfer Protocol */
-#undef	DOWNLOAD_PROTO_HTTPS	/* Secure Hypertext Transfer Protocol */
+#define	DOWNLOAD_PROTO_HTTPS	/* Secure Hypertext Transfer Protocol */
 #undef	DOWNLOAD_PROTO_FTP	/* File Transfer Protocol */
 #undef	DOWNLOAD_PROTO_SLAM	/* Scalable Local Area Multicast */
 #undef	DOWNLOAD_PROTO_NFS	/* Network File System Protocol */
@@ -125,8 +125,8 @@ FILE_LICENCE ( GPL2_OR_LATER_OR_UBDL );
 #define	IMAGE_PNG		/* PNG image support */
 #define	IMAGE_DER		/* DER image support */
 #define	IMAGE_PEM		/* PEM image support */
-//#define	IMAGE_ZLIB		/* ZLIB image support */
-//#define	IMAGE_GZIP		/* GZIP image support */
+#define	IMAGE_ZLIB		/* ZLIB image support */
+#define	IMAGE_GZIP		/* GZIP image support */
 //#define	IMAGE_UCODE		/* Microcode update image support */
 
 /*
@@ -149,24 +149,24 @@ FILE_LICENCE ( GPL2_OR_LATER_OR_UBDL );
 #define LOGIN_CMD		/* Login command */
 #define SYNC_CMD		/* Sync command */
 #define SHELL_CMD		/* Shell command */
-//#define NSLOOKUP_CMD		/* DNS resolving command */
+#define NSLOOKUP_CMD		/* DNS resolving command */
 //#define TIME_CMD		/* Time commands */
 //#define DIGEST_CMD		/* Image crypto digest commands */
 //#define LOTEST_CMD		/* Loopback testing commands */
 //#define VLAN_CMD		/* VLAN commands */
 //#define PXE_CMD		/* PXE commands */
-//#define REBOOT_CMD		/* Reboot command */
-//#define POWEROFF_CMD		/* Power off command */
+#define REBOOT_CMD		/* Reboot command */
+#define POWEROFF_CMD		/* Power off command */
 //#define IMAGE_TRUST_CMD	/* Image trust management commands */
 //#define IMAGE_CRYPT_CMD	/* Image encryption management commands */
 //#define PCI_CMD		/* PCI commands */
 //#define PARAM_CMD		/* Request parameter commands */
 //#define NEIGHBOUR_CMD		/* Neighbour management commands */
-//#define PING_CMD		/* Ping command */
-//#define CONSOLE_CMD		/* Console command */
-//#define IPSTAT_CMD		/* IP statistics commands */
+#define PING_CMD		/* Ping command */
+#define CONSOLE_CMD		/* Console command */
+#define IPSTAT_CMD		/* IP statistics commands */
 //#define PROFSTAT_CMD		/* Profiling commands */
-//#define NTP_CMD		/* NTP commands */
+#define NTP_CMD		/* NTP commands */
 //#define CERT_CMD		/* Certificate management commands */
 //#define IMAGE_MEM_CMD		/* Read memory command */
 #define IMAGE_ARCHIVE_CMD	/* Archive image management commands */
```

and running :
```bash
make bin/ipxe.usb EMBED=$CI_PROJECT_DIR/shadow.ipxe CERT=$CI_PROJECT_DIR/ca.pem,$CI_PROJECT_DIR/isrgrootx1.pem,$CI_PROJECT_DIR/lets-encrypt-r3.pem TRUST=$CI_PROJECT_DIR/ca.pem,$CI_PROJECT_DIR/isrgrootx1.pem,$CI_PROJECT_DIR/lets-encrypt-r3.pem DEBUG=x509
```

leads to this error (full log [here](https://cdn.artifacts.gitlab-static.net/00/bf/00bfc3c82a9f1e10e702ac609faa0d651d5068eff056f7a5a866f950f1615aaa/2024_09_10/7783331466/8547288862/job.log?response-content-type=text%2Fplain%3B%20charset%3Dutf-8&response-content-disposition=inline&Expires=1725961888&KeyName=gprd-artifacts-cdn&Signature=o77LcrJieYFQa5EPhFTlTzaM6qI=)) :
```
  [BIN] bin/usbdisk.bin
  [VERSION] bin/version.ipxe.hd.o
In file included from include/ipxe/features.h:6,
                 from core/version.c:33:
<command-line>: error: 'a183' undeclared here (not in a function)
include/ipxe/dhcp.h:551:58: note: in definition of macro 'DHCP_OPTION'
  551 | #define DHCP_OPTION( ... ) VA_ARG_COUNT ( __VA_ARGS__ ), __VA_ARGS__
      |                                                          ^~~~~~~~~~~
include/ipxe/features.h:71:9: note: in expansion of macro '__DHCP_FEATURE'
   71 |         __DHCP_FEATURE ( _name, feature_opt, __VA_ARGS__ )
      |         ^~~~~~~~~~~~~~
include/ipxe/features.h:69:9: note: in expansion of macro '_DHCP_FEATURE'
   69 |         _DHCP_FEATURE ( OBJECT, feature_opt, __VA_ARGS__ )
      |         ^~~~~~~~~~~~~
include/ipxe/features.h:106:9: note: in expansion of macro 'DHCP_FEATURE'
  106 |         DHCP_FEATURE ( DHCP_ENCAPSULATED ( DHCP_EB_VERSION ), __VA_ARGS__ )
      |         ^~~~~~~~~~~~
core/version.c:49:1: note: in expansion of macro 'FEATURE_VERSION'
   49 | FEATURE_VERSION ( VERSION_MAJOR, VERSION_MINOR, VERSION_PATCH );
      | ^~~~~~~~~~~~~~~
core/version.c:49:19: note: in expansion of macro 'VERSION_MAJOR'
   49 | FEATURE_VERSION ( VERSION_MAJOR, VERSION_MINOR, VERSION_PATCH );
      |                   ^~~~~~~~~~~~~
core/version.c:49:47: error: expected expression before ',' token
   49 | FEATURE_VERSION ( VERSION_MAJOR, VERSION_MINOR, VERSION_PATCH );
      |                                               ^
include/ipxe/dhcp.h:551:58: note: in definition of macro 'DHCP_OPTION'
  551 | #define DHCP_OPTION( ... ) VA_ARG_COUNT ( __VA_ARGS__ ), __VA_ARGS__
      |                                                          ^~~~~~~~~~~
include/ipxe/features.h:71:9: note: in expansion of macro '__DHCP_FEATURE'
   71 |         __DHCP_FEATURE ( _name, feature_opt, __VA_ARGS__ )
      |         ^~~~~~~~~~~~~~
include/ipxe/features.h:69:9: note: in expansion of macro '_DHCP_FEATURE'
   69 |         _DHCP_FEATURE ( OBJECT, feature_opt, __VA_ARGS__ )
      |         ^~~~~~~~~~~~~
include/ipxe/features.h:106:9: note: in expansion of macro 'DHCP_FEATURE'
  106 |         DHCP_FEATURE ( DHCP_ENCAPSULATED ( DHCP_EB_VERSION ), __VA_ARGS__ )
      |         ^~~~~~~~~~~~
core/version.c:49:1: note: in expansion of macro 'FEATURE_VERSION'
   49 | FEATURE_VERSION ( VERSION_MAJOR, VERSION_MINOR, VERSION_PATCH );
      | ^~~~~~~~~~~~~~~
core/version.c:67:48: error: expected expression before ';' token
   67 | const int product_minor_version = VERSION_MINOR;
      |                                                ^
make: *** [Makefile.housekeeping:1215: bin/version.ipxe.hd.o] Error 1
```

NiKiZe, 2024-09-10:
Always try to compile without any local modifications at all.
Always use latest git master. (What does `git log` show for you?)

After that, change one thing at a time.
Don't change default settings if you don't need to.

If you still have issues, do include version of gcc.

NiKiZe, 2024-09-10:
You are using a shallow git clone, see https://github.com/ipxe/ipxe/discussions/454#discussioncomment-1296930

aarnoud, 2024-09-11:
yes, this is fixed now, thanks !

closed 2024-12-10 by zube[bot]