ipxe/ipxe #1312: iPXE boot of KVM guest fails: Initramfs unpacking failed: broken padding

JKDingwall, 2024-09-30:
### Discussed in https://github.com/ipxe/ipxe/discussions/1244

<div type='discussions-op-text'>

<sup>Originally posted by **JKDingwall** June 27, 2024</sup>
Hi,

We have had intermittent issues in the past trying to boot under KVM (iPXE chainload) vs physical where the KVM boot would fail but the physical would succeed.  This could usually be resolved by rebuilding the initramfs image such that it ended up with a different size.  Today we've got a new problem with an environment using the Ubuntu `Linux kernel x86 boot executable bzImage, version 5.15.0-113-generic (buildd@lcy02-amd64-017) #123~20.04.1-Ubuntu SMP Wed Jun 12 17:33:13 UTC 2024` kernel where it fails with the message in the discussion title.  I have two workarounds:

1. reorder artefact download from kernel / initramfs (main) / initramfs (supplemetary cpio) to initramfs (main) / initramfs (supplemetary cpio) / kernel
2. pad the kernel to a 4k boundary with `truncate -s 11550720 vmlinuz` and keep the original order (original kernel size 11548296)

My suspicion is that somehow the combination of the in memory layout of the kernel / initramfs files produced by iPXE and the kernel decoding routine results in the initramfs being missed.  Are there any further suggestions on the debugging which can be carried out from here?

Two iPXE versions were tested: tag v1.21.1 and commit e965f179e1654103eca33feed7a9cc4c51d91be6

Full iPXE script:

```
#!ipxe
echo iPXE Boot String
show unixtime
kernel https://<boot_server_ip>:8080/vmlinuz initrd=certs.cpio initrd=initrd.img BOOTIF=<client_mac> ip=<client_ip>::<client_gw>:<netmask>:<hostname>::off netboot=http boot=casper locale=en_GB.UTF-8 setkmap=gb mirror/country=GB hostname=<hostname> fetch=https://<boot_server_ip>:8080/filesystem.squashfs audit=1 quiet splash  
imgverify vmlinuz https://<boot_server_ip>:8080/vmlinuz.sig

initrd https://<boot_server_ip>:8080/initrd.img
imgverify initrd.img https://<boot_server_ip>:8080/initrd.img.sig

initrd https://<boot_server_ip>:8080/<client_mac>/certs.cpio
imgverify certs.cpio https://<boot_server_ip>:8080/<client_mac>/certs.cpio.sig
boot ||
# If everything failed, give the user some options
echo Boot failed
prompt --key 0x02 --timeout 2000 Press Ctrl-B for the iPXE command line... && shell ||
```</div>

JKDingwall, 2024-09-30:
I tried to look at how the memory management differed according the NIC drivers which are in use between the two environments but didn't make much progress.  My hunch is that under KVM the driver may reuse a buffer without zeroing it but I was unable to actually show that.

NiKiZe, 2024-09-30:
You need to show that your problems is due to a bug i iPXE, and not a kernel or initrd one. (Untill that has shown I'm closing this issue, but happy to reopen, if we come to a different conclusion in discussion)

I will comment on the discussion, based on the info you have provided here, and close the issue in the meantime.