ipxe/ipxe #1316: Unable to load autoexec.ipxe due to missing domain name servers

lollipopman, 2024-10-06:
On our Dell R440 iPXE is unable to obtain the domain name server from EFI and so fails to load autoexec.ipxe. However, when testing with Grub, Grub is able to load additional files via EFI over HTTP. From my brief reading of some of the iPXE commit logs, my understanding is that iPXE only uses some of the EFI functions, whereas to my knowledge, Grub2 is using EFI entirely to try and load the file. Perhaps there is some way to fallback to Grub's behavior? Happy to provide additional debugging.

## Grub

Debian Grub2 version 2.12-5, bootnetx64.efi is able to load grubx64.efi using EFI HTTP:

## Logs

    # Web server logs
    10.64.48.139 - - [06/Oct/2024:18:40:06 +0000] "HEAD /tftpboot/trixie-installer/debian-installer/amd64/bootnetx64.efi HTTP/1.1" 200 0 "-" "UefiHttpBoot/1.0"
    10.64.48.139 - - [06/Oct/2024:18:40:06 +0000] "GET /tftpboot/trixie-installer/debian-installer/amd64/bootnetx64.efi HTTP/1.1" 200 957136 "-" "UefiHttpBoot/1.0"
    10.64.48.139 - - [06/Oct/2024:18:40:06 +0000] "GET /tftpboot/trixie-installer/debian-installer/amd64/revocations.efi HTTP/1.1" 404 153 "-" "UefiHttpBoot/1.0"
    10.64.48.139 - - [06/Oct/2024:18:40:06 +0000] "GET /tftpboot/trixie-installer/debian-installer/amd64//grubx64.efi HTTP/1.1" 200 2336192 "-" "UefiHttpBoot/1.0"

## iPXE

iPXE @3f4f8439 compiled with `make bin-x86_64-efi/snponly.efi DEBUG=efi_autoexec,efi_utils,efi_init,dns`, iPXE fails to load autoxec.ipxe:

## Logs

    >>Start HTTP boot over IPv4
      Station IP address is 10.64.48.139

      URI: http://apt.wikimedia.org/efiboot/snponly.efi
    EFI handle 0x60839020 systab 0x67bfe020
    EFI does not provide protocol ConsoleControl
    EFI protocol ef9fc172-a1b2-4693-b327-6d32fc416042 is at 0x55936348
    EFI protocol AcpiTable is at 0x65b260a8
    EFI protocol 8b843e20-8132-4852-90cc-551a4e4a7f1c is at 0x55a5bad0
    EFI configuration table eb9d2d30-2d88-11d3-9a16-0090273fc14d is at 0x6f7fe000
    EFI configuration table f2fd1544-9794-4a2c-992e-e5bbcf20e394 is at 0x686bc000
    EFI configuration table eb9d2d31-2d88-11d3-9a16-0090273fc14d is at 0x686be000
    EFI image base address 0x500f5000
    EFI image device path PciRoot(0x0)/Pci(0x1C,0x5)/Pci(0x0,0x0)/MAC(4CD98F60AF47,0x1)/IPv4(0.0.0.0,TCP,DHCP,10.64.48.139,10.64.48.1,255.255.252.0)/Uri(http://apt.wikimedia.org/efiboot/snponly.efi)
    EFIDEV PciRoot(0x0)/Pci(0x1C,0x5)/Pci(0x0,0x0) has no parent supporting SimpleNetwork: Error 0x7f45e08e (https://ipxe.org/7f45e08e)
    EFIDEV PciRoot(0x0)/Pci(0x1C,0x5)/Pci(0x0,0x0) has no parent supporting Nii31: Error 0x7f45e08e (https://ipxe.org/7f45e08e)
    EFIDEV PciRoot(0x0)/Pci(0x1C,0x5)/Pci(0x0,0x0) has no parent supporting ManagedNetworkSb: Error 0x7f45e08e (https://ipxe.org/7f45e08e)
    iPXE initialising devices...
    EFIDEV PciRoot(0x0)/Pci(0x1C,0x5)/Pci(0x0,0x0)/MAC(4CD98F60AF47,0x1)/IPv4(0.0.0.0,TCP,DHCP,10.64.48.139,10.64.48.1,255.255.252.0)/Uri(http://apt.wikimedia.org/efiboot/snponly.efi) has no parent supporting SimpleFileSystem: Error 0x7f45e08e (https://ipxe.org/7f45e08e)
    EFI PciRoot(0x0)/Pci(0x1C,0x5)/Pci(0x0,0x0)/MAC(4CD98F60AF47,0x1)/IPv4(0.0.0.0,TCP,DHCP,10.64.48.139,10.64.48.1,255.255.252.0)/Uri(http://apt.wikimedia.org/efiboot/snponly.efi) found no SimpleFileSystem: Error 0x7f45e08e (https://ipxe.org/7f45e08e)
    EFI PciRoot(0x0)/Pci(0x1C,0x5)/Pci(0x0,0x0)/MAC(4CD98F60AF47,0x1)/IPv4(0.0.0.0,TCP,DHCP,10.64.48.139,10.64.48.1,255.255.252.0)/Uri(http://apt.wikimedia.org/efiboot/snponly.efi) found ManagedNetworkSb on PciRoot(0x0)/Pci(0x1C,0x5)/Pci(0x0,0x0)/MAC(4CD98F60AF47,0x1)
    autoexec.ipxe...DNS not attempting to resolve "apt.wikimedia.org": no DNS servers
     Error 0x3e11628e (https://ipxe.org/3e11628e)
    EFI PciRoot(0x0)/Pci(0x1C,0x5)/Pci(0x0,0x0)/MAC(4CD98F60AF47,0x1)/IPv4(0.0.0.0,TCP,DHCP,10.64.48.139,10.64.48.1,255.255.252.0)/Uri(http://apt.wikimedia.org/efiboot/snponly.efi) could not download autoexec.ipxe: Error 0x3e11628e (https://ipxe.org/3e11628e)

    # Web server logs
    10.64.48.139 - - [06/Oct/2024:18:51:56 +0000] "HEAD /efiboot/snponly.efi HTTP/1.1" 200 0 "-" "UefiHttpBoot/1.0"
    10.64.48.139 - - [06/Oct/2024:18:51:56 +0000] "GET /efiboot/snponly.efi HTTP/1.1" 200 219136 "-" "UefiHttpBoot/1.0"

NiKiZe, 2024-10-06:
Start with no debug, what is the error you are having?

autoexec.ipxe is loaded the same way as iPXE is, which means, it is up to your efi firmware to get that to you. (This also means that iPXEs DNS logic is irrelevant)
This has mostly been tested with TFTP, where there is no hostnames involved, so surely could be a parsing issue of the path used.

autoexec.ipxe is quite new, the older way is to have your DHCP provide filename to a script when iPXE does DHCP discovery. Autoexec.ipxe is in that case expected to be a "file not found"

lollipopman, 2024-10-07:
Thanks @NiKiZe for taking a look, here is the output with no debugging enabled on the Dell, as a data point the same config works on one of our Supermicro servers.:

```
>>Start HTTP boot over IPv4
  Station IP address is 10.64.48.139

  URI: http://apt.wikimedia.org/efiboot/snponly.efi
iPXE initialising devices...
autoexec.ipxe... Error 0x3e11628e (https://ipxe.org/3e11628e)

iPXE 1.21.1+ (g3f4f8) -- Open Source Network Boot Firmware -- https://ipxe.org
Features: DNS HTTP iSCSI TFTP VLAN AoE EFI Menu

iPXE>                                    
```

NiKiZe, 2024-10-07:
Thanks, I would simply not use autoexec.ipxe, the message would still show but can be ignored.

And instead use DHCP filename set accordingly for iPXE clients. As described at https://ipxe.org/howto/chainloading#breaking_the_loop_with_the_dhcp_server

@mcb30 it seems that initial load uses iPXE DNS, while probably should use EFI fully?

mcb30, 2024-10-07:
> @mcb30 it seems that initial load uses iPXE DNS, while probably should use EFI fully?

@lollipopman As described at some length in the commit log for https://github.com/ipxe/ipxe/commit/165995b7e, while UEFI does technically provide an HTTP client abstraction, it is such an abomination of design failures that we choose not to go anywhere near it.  We rely instead upon creating a temporary MNP-based network interface, applying the settings (such as DNS servers) obtained from the UEFI HTTP boot device path, and then using our own HTTP client.

In your debug output, we see the line

```
EFI PciRoot(0x0)/Pci(0x1C,0x5)/Pci(0x0,0x0)/MAC(4CD98F60AF47,0x1)/IPv4(0.0.0.0,TCP,DHCP,10.64.48.139,10.64.48.1,255.255.252.0)/Uri(http://apt.wikimedia.org/efiboot/snponly.efi) found ManagedNetworkSb on PciRoot(0x0)/Pci(0x1C,0x5)/Pci(0x0,0x0)/MAC(4CD98F60AF47,0x1)
```

which indicates that we have found the MNP device handle.  The fact that we then get as far as the `could not download autoexec.ipxe` error message means that the creation of the temporary MNP device worked as expected.

I note that the device path (provided by your UEFI firmware) doesn't include any `Dns(...)` element.  This fully explains why iPXE has no DNS servers - your firmware has failed to convey the address of the DNS servers via the device path.

If you try with the same debug options on the known-working Supermicro server, my guess is that you'll see a `Dns(...)` element showing up within the device path.

Assuming that I'm correct, then this is a bug in the Dell UEFI firmware.  There may be a way to work around it, e.g. by finding some other place that the DNS server address is stored, or by deciding to (ab)use the router address as a DNS server address if the firmware fails to provide a DNS element, but this would definitely require some debug and development work with access to one of the faulty Dell servers.

lollipopman, 2024-10-07:
> > @mcb30 it seems that initial load uses iPXE DNS, while probably should use EFI fully?
> If you try with the same debug options on the known-working Supermicro server, my guess is that you'll see a `Dns(...)` element showing up within the device path.

You are correct, here is the supermicro output:

```
>>Checking Media Presence......
>>Media Present......
  URI: http://apt.wikimedia.org/efiboot/snponly.efi
  File Size: 219136 Bytes
  Downloading...100%EFI handle 0x5c75fe18 systab 0x6bdfe018
EFI protocol ConsoleControl is at 0x63e99e18
EFI protocol ef9fc172-a1b2-4693-b327-6d32fc416042 is at 0x645cf7c8
EFI protocol AcpiTable is at 0x5fa3c3a0
EFI protocol 8b843e20-8132-4852-90cc-551a4e4a7f1c is at 0x645dd850
EFI configuration table eb9d2d30-2d88-11d3-9a16-0090273fc14d is at 0x67add000
EFI configuration table f2fd1544-9794-4a2c-992e-e5bbcf20e394 is at 0x6b8f5000
EFI configuration table eb9d2d31-2d88-11d3-9a16-0090273fc14d is at 0x6b8f6000
EFI image base address 0x5b981000
EFI image device path PciRoot(0x4)/Pci(0x2,0x0)/Pci(0x0,0x0)/MAC(7CC255975A0E,0x1)/IPv4(0.0.0.0,TCP,DHCP,10.192.21.8,10.192.21.1,255.255.255.0)/Dns(10.3.0.1)/Uri(http://apt.wikimedia.org/efiboot/snponly.efi)
EFIDEV PciRoot(0x4)/Pci(0x2,0x0)/Pci(0x0,0x0) has no parent supporting SimpleNetwork: Error 0x7f45e08e (https://ipxe.org/7f45e08e)
EFIDEV PciRoot(0x4)/Pci(0x2,0x0)/Pci(0x0,0x0) has no parent supporting Nii31: Error 0x7f45e08e (https://ipxe.org/7f45e08e)
EFIDEV PciRoot(0x4)/Pci(0x2,0x0)/Pci(0x0,0x0) has no parent supporting ManagedNetworkSb: Error 0x7f45e08e (https://ipxe.org/7f45e08e)
iPXE initialising devices...
EFIDEV PciRoot(0x4)/Pci(0x2,0x0)/Pci(0x0,0x0)/MAC(7CC255975A0E,0x1)/IPv4(0.0.0.0,TCP,DHCP,10.192.21.8,10.192.21.1,255.255.255.0)/Dns(10.3.0.1)/Uri(http://apt.wikimedia.org/efiboot/snponly.efi) has no parent supporting SimpleFileSystem: Error 0x7f45e08e (https://ipxe.org/7f45e08e)
EFI PciRoot(0x4)/Pci(0x2,0x0)/Pci(0x0,0x0)/MAC(7CC255975A0E,0x1)/IPv4(0.0.0.0,TCP,DHCP,10.192.21.8,10.192.21.1,255.255.255.0)/Dns(10.3.0.1)/Uri(http://apt.wikimedia.org/efiboot/snponly.efi) found no SimpleFileSystem: Error 0x7f45e08e (https://ipxe.org/7f45e08e)
EFI PciRoot(0x4)/Pci(0x2,0x0)/Pci(0x0,0x0)/MAC(7CC255975A0E,0x1)/IPv4(0.0.0.0,TCP,DHCP,10.192.21.8,10.192.21.1,255.255.255.0)/Dns(10.3.0.1)/Uri(http://apt.wikimedia.org/efiboot/snponly.efi) found ManagedNetworkSb on PciRoot(0x4)/Pci(0x2,0x0)/Pci(0x0,0x0)/MAC(7CC255975A0E,0x1)
autoexec.ipxe...DNS 0x5b9b8be8 sending 10.3.0.1 query ID 0x903c for apt.wikimedia.org type A
DNS 0x5b9b8be8 received response ID 0x903c
DNS 0x5b9b8be8 found address 208.80.154.10
 ok
EFI PciRoot(0x4)/Pci(0x2,0x0)/Pci(0x0,0x0)/MAC(7CC255975A0E,0x1)/IPv4(0.0.0.0,TCP,DHCP,10.192.21.8,10.192.21.1,255.255.255.0)/Dns(10.3.0.1)/Uri(http://apt.wikimedia.org/efiboot/snponly.efi) loaded autoexec.ipxe (850 bytes)

```

> Assuming that I'm correct, then this is a bug in the Dell UEFI firmware. There may be a way to work around it, e.g. by finding some other place that the DNS server address is stored, or by deciding [...]

Makes sense, we will see if Dell is responsive on fixing the bug.

NiKiZe, 2024-10-07:
Any chance to use the cached DHCP packet, or are we to early here? (Also not sure how viable that would be in an IPv6 scenario?)

lollipopman, 2024-10-10:
I tested on some newer dell hardware and the domain server information is properly preserved:

- Failed: Dell R440 Bios version: 2.22.1, purchased 2019-04-01
- Succeeded: Dell R450 Bios version: 1.15.2, purchased 2023-04-10

Though I haven't tested our whole fleet, I am optimistic that newer dell BIOSes have fixed the issue.

NiKiZe, 2024-10-10:
Ok, so hopefully a rare case of non working for early implementations of HTTP boot? 🤞

Not really something that we want to spend time on unless there is more buggy firmwares? Or do we want to keep this open and look for possible workarounds?

deitch, 2025-01-21:
I built `ipxe.efi` with debug, as above. I then placed it on both tftp and http (same binary, just different protocols to access from same server). 

```
PXE -> ipxe.efi (over tftp) -> ipxe.efi (over http)
```

Here are the logs:

```
>>Start PXE over IPv4.
  Station IP address is 192.168.1.10

  Server IP address is 192.168.1.2
  NBP filename is ipxe.efi
  NBP filesize is 1008704 Bytes
 Downloading NBP file...

  NBP file downloaded successfully.
BdsDxe: loading Boot0003 "UEFI PXEv4 (MAC:525400123456)" from PciRoot(0x0)/Pci(0x2,0x0)/MAC(525400123456,0x1)/IPv4(0.0.0.0,0x0,DHCP,0.0.0.0,0.0.0.0
,0.0.0.0)
BdsDxe: starting Boot0003 "UEFI PXEv4 (MAC:525400123456)" from PciRoot(0x0)/Pci(0x2,0x0)/MAC(525400123456,0x1)/IPv4(0.0.0.0,0x0,DHCP,0.0.0.0,0.0.0.
0,0.0.0.0)
EFI handle 0x7e4dae18 systab 0x7f9ec018
EFI does not provide protocol ConsoleControl
EFI protocol ef9fc172-a1b2-4693-b327-6d32fc416042 is at 0x7f309ae8
EFI does not provide protocol c12b1873-ac17-4176-ac77-7ecb4defffec
EFI protocol AcpiTable is at 0x7f36cca0
EFI protocol 8b843e20-8132-4852-90cc-551a4e4a7f1c is at 0x7f38c420
EFI protocol 3152bca5-eade-433d-862e-c01cdc291f44 is at 0x7f312c90
EFI configuration table eb9d2d30-2d88-11d3-9a16-0090273fc14d is at 0x7fb7e000
EFI configuration table f2fd1544-9794-4a2c-992e-e5bbcf20e394 is at 0x7f91a000
EFI configuration table eb9d2d31-2d88-11d3-9a16-0090273fc14d is at 0x7f91c000
EFI image base address 0x7e232000
EFI image device path PciRoot(0x0)/Pci(0x2,0x0)/MAC(525400123456,0x1)/IPv4(0.0.0.0)
iPXE initialising devices...ok

iPXE 1.0.0+ -- Open Source Network Boot Firmware -- http://ipxe.org
Features: DNS FTP HTTP HTTPS iSCSI TFTP SRP AoE EFI Menu

Configuring (net0 52:54:00:12:34:56)......DNS servers: 192.168.1.3
 ok

http://192.168.1.2:8080/ipxe.efi... ok
EFI handle 0x7e4c9518 systab 0x7f9ec018
EFI does not provide protocol ConsoleControl
EFI protocol ef9fc172-a1b2-4693-b327-6d32fc416042 is at 0x7f309ae8
EFI does not provide protocol c12b1873-ac17-4176-ac77-7ecb4defffec
EFI protocol AcpiTable is at 0x7f36cca0
EFI protocol 8b843e20-8132-4852-90cc-551a4e4a7f1c is at 0x7f38c420
EFI protocol 3152bca5-eade-433d-862e-c01cdc291f44 is at 0x7f312c90
EFI configuration table eb9d2d30-2d88-11d3-9a16-0090273fc14d is at 0x7fb7e000
EFI configuration table f2fd1544-9794-4a2c-992e-e5bbcf20e394 is at 0x7f91a000
EFI configuration table eb9d2d31-2d88-11d3-9a16-0090273fc14d is at 0x7f91c000
EFI image base address 0x7df9e000
EFI image device path PciRoot(0x0)/Pci(0x2,0x0)/MAC(525400123456,0x1)/MAC(525400123456,0x1)
iPXE initialising devices...ok

iPXE 1.0.0+ -- Open Source Network Boot Firmware -- http://ipxe.org
Features: DNS FTP HTTP HTTPS iSCSI TFTP SRP AoE EFI Menu
```

Doesn't look all that different. Should I have compiled with different debug options?

stappersg, 2025-01-22:
> I built `ipxe.efi` with debug, as above.

What I found above, in the _openingsmessage_,: `DEBUG=efi_autoexec,efi_utils,efi_init,dns`

> Doesn't look all that different. Should I have compiled with different debug options?

I don't understand that question.  I did understand the _"Hi there, interesting problem you have, I want to help, I can build **iPXE** and willing to provide debug logging"_.

Now a new attempt to answer:

> Should I have compiled with different debug options?

First reproduce the problem without any debug options. A possible way to reproduce the problem, the problem this issue is about, is break server side, like breaking DNS. After all is this issue titled _Unable to load autoexec.ipxe due to missing domain name servers_.

@lollipopman 
> Though I haven't tested our whole fleet, I am optimistic that newer dell BIOSes have fixed the issue.

Please share with the iPXE community how much further human energy this issue, regarding you, deserves.

closed 2026-01-05 by lollipopman