ipxe/ipxe #1327: autoexec.ipxe error 0x3d126083

thesefer, 2024-10-28:
hi

I wanted to update my ipxe build from 2022 and on a fresh clone without any changes (and with changes) we receive the following error on vms and physical devices:

`autoexec.ipxe... Error 0x3d126083 (https://ipxe.org/3d126083)`

I didn't change anything on my dhcp (the old backup files work perfectly) but maybe I have missed some critical change? The menu still loads perfectly fine afterwards.

Scope Policies
- iPXE Configuration -> Boot\iPXE\iPXE.conf
- UEFI x64 -> Boot\iPXE\iPXE64.efi
- UEFI x86 -> Boot\iPXE\iPXE32.efi
- UEFI x86 & x64 -> Boot\iPXE\iPXE64.efi

iPXE.conf
```
#!ipxe
chain --autofree Boot\iPXE\boot.ipxe.cfg
```

And a menue in `boot.ipxe.cfg`

NiKiZe, 2024-10-28:
`autoexec.ipxe` is a new feature, it is downloaded before switching to iPXEs native network stack.
You are seeing a message about this file not being present at the same location as your ipxe binary.

However, it is in no way a requirement.

By searching after autoexec.ipxe you will find several other explanations. https://github.com/search?q=repo%3Aipxe%2Fipxe+autoexec.ipxe&type=issues such as #643 and others.

A small nit, iPXE uses scripts. Not config files, you are free to name them what you want, but some concepts is easier to understand when you think of them as dynamic scripts and not as static configurations.

mcb30, 2024-10-28:
Curiously, the TFTP server is returning an "illegal operation" error, instead of a "not found" error.  In the usual case of the `autoexec.ipxe` simply not existing, you would expect to see:
```
autoexec.ipxe... Not found
```
which is a less surprising message.

It would be interesting to know why your TFTP server is complaining about an illegal operation.  Do you see any errors relating to `autoexec.ipxe` in your TFTP server logs?

NiKiZe, 2024-10-28:
I would suspect that this is a windows server, which denies access outside the boot directory.

mcb30, 2024-10-28:
It's also worth noting that this looks like exactly the kind of setup for which the `autoexec.ipxe` feature was designed.  As far as I can tell, @thesefer is building iPXE with an embedded script that just does:
```
#!ipxe
chain --autofree Boot\iPXE\boot.ipxe.cfg
```
where `boot.ipxe.cfg` is placed in the same directory as the iPXE binaries.

So, if you build iPXE without any embedded script, and just rename `boot.ipxe.cfg` to `autoexec.ipxe`, then it should all behave as intended.

thesefer, 2024-10-28:
> I would suspect that this is a windows server, which denies access outside the boot directory.

Just to give a reply again, you are correct with this assumption!

Thank you both for the explaination.

@mcb30 I am not using any script, just changed some configs to fit our needs. Those files reside on the WDS/Server itself. I will adjust our DHCP accordingly.

NiKiZe, 2024-10-28:
> iPXE.conf
> 
> ```
> #!ipxe
> chain --autofree Boot\iPXE\boot.ipxe.cfg
> ```
> 

As I said, this is a script, not a config.

thesefer, 2024-10-29:
@mcb30
>It would be interesting to know why your TFTP server is complaining about an illegal operation. Do you see any errors relating to autoexec.ipxe in your TFTP server logs?

Just fyi. We are running Windows Server 2019 with DHCP enabled, not WDS. Unfortunately I cannot find any kind of log currently which would tell me what is happening here, probably would have to pcap it.

I tried renaming the file to from `boot.ipxe.cfg` to `autoexec.ipxe` and changing the line in `iPXE.conf` to no avail. I'll try tomorrow to deliver `autoexec.ipxe` directly instead. My dhcp setup is mostly build on https://github.com/mintsoft/iPXE-WDS-EFI-Boot-Menu using the Policies to deliver the iPXE.conf when requested as processing order 1 followed by the Arch depending boot files.

NiKiZe, 2024-10-29:
I assume that the TFTP server settings are the same as used for WDS and somewhat documented at https://ipxe.org/appnote/chainload_wds

Maybe there is further filename filters there that blocks the .ipxe extension (and thus you are using the .cfg extension to circumvent that)

mcb30, 2024-10-30:
The backwards slashes could be causing an issue with pathname resolution.  iPXE will try to load `autoexec.ipxe` from the same directory as itself by treating `autoexec.ipxe` as a relative URI to be resolved against its own base URI.  In your case, the base URI is the string `Boot\iPXE\iPXE64.efi`, and I suspect (without checking) that iPXE's path resolver will treat that as just a filename since the backwards slashes are not valid URI path directory separator characters.

You could try with `Boot/iPXE/iPXE64.efi` instead, hoping that your Windows-based TFTP server will accept standard forward slashes as directory separators.

thesefer, 2024-10-30:
Unfortunately still the same error even with forward slashes. It finds the file `Filename: Boot/iPXE/autoexec.ipxe`. But interestingly with forward slashes it cannot handle the chain anymore (with either `\` or `/` in the script).

```
#!ipxe
chain --autofree Boot\iPXE\boot.ipxe.cfg
```

If I place the menu in `autoexec.ipxe` instead without chain loading I still get the same error code `0x3d126083`. Access to this file is definitely given else the menu would not load. Hmm

mintsoft, 2024-12-12:
Just to chime in here with my understanding of what's going on:

The  `autoexec.ipxe error 0x3d126083` error appeared for me when I updated the `snponly.efi`. 
That's now a feature of iPXE that it will pull a file called autoexec.ipxe from the root of the TFTP server. 
Under default WDS config that'll error (with access violation) because of the `ReadFilter` registry key. 
If you modify the `ReadFilter` to include `autoexec.ipxe`, it will succeed in pulling the file. If you then put either the chain script or the content of `boot.ipxe.cfg` inside `autoexec.ipxe` it will execute that script.

There is, however, a key difference between the script being executed as `autoexec.ipxe` vs the original arrangement of ipxe doing a DHCP request and pulling its own option 67, so if the intention is to use the autoexec then the script needs to be different and include a dhcp step like:

```
#!ipxe
dhcp
chain --autofree Boot\iPXE\boot.ipxe.cfg
```

That works for me named `autoexec.ipxe` in the root of `RemoteInstall` and the `ReadFilter` regkey changed to include `autoexec.ipxe`. 

I probably would not recommend doing this, the ReadFilter registry keys (as far as I know) are unsupported, so personally I prefer to leave them alone and just ignore the error that appears as it's harmless anyway.

1soproni, 2025-01-03:
Hi @mintsoft 
What should autoexec.ipxe cointain? It was not mentioned in the tutorial I followed.

NiKiZe, 2025-01-03:
> Hi @mintsoft What should autoexec.ipxe cointain? It was not mentioned in the tutorial I followed.

`autoexec.ipxe` as just like any other iPXE script. It isn't required, you don't need to have it at all, but you can also migrate your normal scripts to this filename if you prefer.

mintsoft, 2025-01-03:
> Hi @mintsoft What should autoexec.ipxe cointain? It was not mentioned in the tutorial I followed.

If you read the comment above your comment (https://github.com/ipxe/ipxe/issues/1327#issuecomment-2540165968) it says in there what you _can_ put in `autoexec.ipxe` but as NiKiZe says, it's not required.