ipxe/ipxe #1331: make failure

mocallins, 2024-10-31:
I actually tried just make, to see what it would create, and i think all of them failed like this:

```
[piserve]:</<1>pi/build/ipxe/src/> make bin/ipxe.iso
  [BUILD] bin/__divdi3.o
gcc: error: unrecognized -march target: i386
gcc: note: valid arguments are: armv4 armv4t armv5t armv5te armv5tej armv6 armv6j armv6k armv6z armv6kz armv6zk armv6t2 armv6-m armv6s-m armv7 armv7-a armv7ve armv7-r armv7-m armv7e-m armv8-a armv8.1-a armv8.2-a armv8.3-a armv8.4-a armv8.5-a armv8.6-a armv8-m.base armv8-m.main armv8-r armv8.1-m.main armv9-a iwmmxt iwmmxt2 native
gcc: error: missing argument to ‘-march=’
gcc: error: unrecognized command-line option ‘-mpreferred-stack-boundary=2’
gcc: error: unrecognized command-line option ‘-mregparm=3’
gcc: error: unrecognized command-line option ‘-mrtd’
gcc: error: unrecognized command-line option ‘-m32’; did you mean ‘-mbe32’?
make: *** [Makefile.housekeeping:979: bin/__divdi3.o] Error 1
```

It would be awesome, that if you gave us the option to customize this iso with thhe  address of this manual installation.

NiKiZe, 2024-10-31:
You are running this on a raspberry pi?
What does `gcc --version` show?

mcb30, 2024-10-31:
> I actually tried just make, to see what it would create, and i think all of them failed like this:
>
> ```
> [piserve]:</<1>pi/build/ipxe/src/> make bin/ipxe.iso
>   [BUILD] bin/__divdi3.o
> gcc: error: [...]

Since you're running on what looks like a 32-bit ARM, you'll either need to build one of the 32-bit ARM targets, or install a cross compiler and build using a command such as

```
make CROSS=x86_64-linux-gnu- bin/ipxe.iso
```

Alternatively (and if what I've written doesn't immediately make sense to you), do the build on a standard x86 machine instead of on a Pi.

mocallins, 2024-10-31:
> You are running this on a raspberry pi? What does `gcc --version` show?

Yes that is correct, Raspberry Pi Zero.

```
[piserve]:</home/pi/> gcc --version
gcc (Raspbian 12.2.0-14+rpi1) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

NiKiZe, 2024-10-31:
> >   
> > Alternatively (and if what I've written doesn't immediately make sense to you), do the build on a standard x86 machine instead of on a Pi.
> 
> Seems like installing a cross compiler, if available, rather that installing all of this on an x86 system, although not impossible, as long as it can be a VM, lol

Any Linux environment on x86 with a compiler would do the job, not much to install... Building on the pi would probably take quite some time.

mocallins, 2024-10-31:
> Any Linux environment on x86 with a compiler would do the job, not much to install... Building on the pi would probably take quite some time.

Also finding an error in bin/.rom.defs:12968:
```
  12966 # NIC   168c0015        168c,0015       Atheros 5212 x15
  12967 DRIVER_168c0015 = ath5k
  12968 # R7
  12969 DRIVER_168c0017 = ath5k
  12970 ROM_TYPE_168c0017 = pci
  12971 ROM_DESCRIPTION_168c0017 = "Atheros 5212 x17"
```
so i just commented and getting further.
Compiling with `make CROSS=arm-linux-gnueabihf- ARCH=arm bin/ipxe.iso`

NiKiZe, 2024-10-31:
If you are building `bin/` that is `i386` you will need a i386 compiler, not ARM!
Any error messages you get from invalid build options is not worth pursuing in any way.

Building for i386 is usualy done (and supported) in a x86 environment. If you want to build for arm, then you need to use the arm build target, not bin (which is a i386 alias). See https://ipxe.org/appnote/buildtargets

What is your end goal?

mocallins, 2024-10-31:
> If you are building `bin/` that is `i386` you will need a i386 compiler, not ARM! Any error messages you get from invalid build options is not worth pursuing in any way.
>
> Building for i386 is usualy [...]

At this point i'm not really sure, chasing the problems, got kind of confusing and lost my train of thought

mocallins, 2024-10-31:
> > > If you are building `bin/` that is `i386` you will need a i386 compiler, not ARM! Any error messages you get from invalid build options is not worth pursuing in any way.
> > > Building for i386 is usualy done (and supported) in a x86 environment. If you want to build for arm, then you need to use the arm build target, not bin (which is a i386 alias). See https://ipxe.org/appnote/buildtargets
> > > What is your end goal?
> > 
> > 
> > At this point i'm not really sure, chasing the problems, got kind of confusing and lost my train of thought
> 
> I was thinking creating a .rom/.mrom image for a VM to boot, but that's looking far more involved. Then i thought maybe just as .iso, to boot with

Is there a nice clean / easy way to create / modify  the / an .iso to automatically chain to myserver/menu.ipxe   ?

mocallins, 2024-10-31:
> Is there a nice clean / easy way to create / modify the / an .iso to automatically chain to myserver/menu.ipxe ?

And based on the buildtargets reference, i am now attempting  `make  bin-arm32-efi/ipxe.iso` , this the correct approach, compiling on this platform ?

NiKiZe, 2024-10-31:
Are you going to use ipxe to boot an arm32 based device? Or what will you use the ipxe build for?

mocallins, 2024-10-31:
> Are you going to use ipxe to boot an arm32 based device? Or what will you use the ipxe build for?

No not going to use with arn device.
For now it will just be some ESXI based VM's.

mocallins, 2024-10-31:
Welp, that compile completed.
And i booted to that ipxe.iso , and it just takes me to  _boot:_
I was thinking i would get the IPXE prompt.

Anyway, i think we can close this issue

I would still like to pursue a custom bootable .iso and would appreciate any info to get me in that direction.

NiKiZe, 2024-10-31:
So you are most likely going to boot an x86 based VM?
Then you have to build for that architecture.

closed 2025-01-30 by zube[bot]