ipxe/ipxe #1333: Network interfaces not detected on AWS EC2 m7 instance type

fayak, 2024-11-04:
Hello,

I have some issue similar to https://github.com/ipxe/ipxe/issues/255 where I cannot boot an EC2 m7 instance type with ipxe, it doesn't find any network interface.
Running lspci gives:
- `27:00.0 Ethernet controller [0200]: Amazon.com, Inc. Elastic Network Adapter (ENA) [1d0f:ec20]`
Following issue 255 I tried:
```
diff --git a/src/config/ioapi.h b/src/config/ioapi.h
index a1498482..3df90e14 100644
--- a/src/config/ioapi.h
+++ b/src/config/ioapi.h
@@ -11,8 +11,8 @@ FILE_LICENCE ( GPL2_OR_LATER_OR_UBDL );
 
 #include <config/defaults.h>
 
-//#undef       PCIAPI_PCBIOS           /* Access via PCI BIOS */
-//#define      PCIAPI_DIRECT           /* Direct access via Type 1 accesses */
+#undef PCIAPI_PCBIOS           /* Access via PCI BIOS */
+#define        PCIAPI_DIRECT           /* Direct access via Type 1 accesses */
```
but it refuses to compile with
```
In file included from include/ipxe/dma.h:13,
                 from include/ipxe/pci.h:15,
                 from core/null_pci.c:33:
arch/x86/include/ipxe/pcicloud.h:24:24: error: redefinition of ‘pci_can_probe’
   24 | PCIAPI_INLINE ( cloud, pci_can_probe ) ( void ) {
      |                        ^~~~~~~~~~~~~
include/ipxe/api.h:35:58: note: in definition of macro ‘SINGLE_API_NAME’
   35 | #define SINGLE_API_NAME( _prefix, _api_func ) _prefix ## _api_func
      |                                                          ^~~~~~~~~
include/ipxe/pci_io.h:40:9: note: in expansion of macro ‘SINGLE_API_INLINE’
   40 |         SINGLE_API_INLINE ( PCIAPI_PREFIX_ ## _subsys, _api_func )
      |         ^~~~~~~~~~~~~~~~~
arch/x86/include/ipxe/pcicloud.h:24:1: note: in expansion of macro ‘PCIAPI_INLINE’
   24 | PCIAPI_INLINE ( cloud, pci_can_probe ) ( void ) {
      | ^~~~~~~~~~~~~
arch/x86/include/ipxe/pcidirect.h:34:25: note: previous definition of ‘pci_can_probe’ with type ‘int(void)’
   34 | PCIAPI_INLINE ( direct, pci_can_probe ) ( void ) {
      |                         ^~~~~~~~~~~~~
include/ipxe/api.h:35:58: note: in definition of macro ‘SINGLE_API_NAME’
   35 | #define SINGLE_API_NAME( _prefix, _api_func ) _prefix ## _api_func
      |                                                          ^~~~~~~~~
include/ipxe/pci_io.h:40:9: note: in expansion of macro ‘SINGLE_API_INLINE’
   40 |         SINGLE_API_INLINE ( PCIAPI_PREFIX_ ## _subsys, _api_func )
      |         ^~~~~~~~~~~~~~~~~
arch/x86/include/ipxe/pcidirect.h:34:1: note: in expansion of macro ‘PCIAPI_INLINE’
   34 | PCIAPI_INLINE ( direct, pci_can_probe ) ( void ) {
      | ^~~~~~~~~~~~~
  [BUILD] bin/pinger.o
make: *** [Makefile.housekeeping:979: bin/null_pci.o] Error 1
make: *** Waiting for unfinished jobs....
```
I am not clever enough to understand and solve this issue however.

mcb30, 2024-11-04:
Build with `CONFIG=cloud` (and no local changes to `config/ioapi.h`) for booting in EC2.  Alternatively, just use the published iPXE images as listed at https://ipxe.org/howto/ec2

fayak, 2024-11-04:
Let my try with an official AMI you people published if I face the issue as well.

Also, I should have mentioned that the AMI I built works on a m5 instance type for example

fayak, 2024-11-04:
```iPXE 1.21.1+ (ga1830) -- Open Source Network Boot Firmware -- https://ipxe.org
Features: DNS HTTP HTTPS iSCSI TFTP SRP AoE ELF MBOOT PXE bzImage Menu PXEXT
Amazon EC2 - iPXE boot via user-data
CPU: GenuineIntel Intel(R) Xeon(R) Platinum 8488C
net0: 0e:46:11:65:25:cf using ena-vf on 0000:27:00.0 (Ethernet) [closed]
  [Link:up, TX:0 TXE:0 RX:0 RXE:0]
DHCP attempt 1
Configuring (net0 0e:46:11:65:25:cf)...... ok
net0: 10.8.9.220/255.255.255.0 gw 10.8.9.1
net0: fe80::c46:11ff:fe65:25cf/64
http://169.254.169.254/latest/user-data... Not found (https://ipxe.org/2d0c613b)
net0: 0e:46:11:65:25:cf using ena-vf on 0000:27:00.0 (Ethernet) [open]
  [Link:up, TX:15 TXE:0 RX:7 RXE:0]
  ```
  
  Ok so it seems like IPXE can work with m7 instances. Now I need to find the difference between officially published image and mine, the later being able to boot on m5 but not m7.
  Wrong report then, sorry to have bothered you. I will try to find the problem and update this issue for future record

mcb30, 2024-11-04:
> sorry I edited my comment right after you posted. I indeed built with `make CONFIG=cloud bin/ipxe.usb EMBED=script.ipxe -j$(nproc)`

I think you may need to use `bin-x86_64-pcbios/ipxe.usb` (as documented at https://ipxe.org/howto/ec2#building_from_source).  From memory, some of the larger instance types place the PCI BARs above 4GB, so the 32-bit build of iPXE cannot access them.

fayak, 2024-11-04:
:facepalm:

Cheers, it works now. So I understand correctly, I used i386 aka x86 32 bits which somehow was good enough for most of my use cases, but for this new kind of host, you need AMD64 aka x86_64 to access addresses above 2**32.
I'm not sure how I came up with the 32 bits version, I've been using this for a few years already so I forgot.

Thanks a lot for the fast answer, and sorry for the misleading issue :) I'll close it

closed 2024-11-04 by fayak