ipxe/ipxe #1337: Intel x550 inconsistent failure, No configuration methods succeeded

ianb-mp, 2024-11-14:
When iPXE attempts DHCP initially it always fails. A tcpdump on my DHCP server shows no traffic from the client on the first attempt. If I drop to a shell and immediately run `dhcp` it always works on the second attempt. TCPdump on my dhcp server for the second attempt is below.

<details>
<summary>tcpdump</summary>

```console
IP (tos 0x0, ttl 252, id 48, offset 0, flags [none], proto UDP (17), length 420)
    10.1.218.1.67 > 10.5.33.252.67: [udp sum ok] BOOTP/DHCP, Request from 10:f9:20:e9:2b:92, length 392, hops 1, xid 0x6fedc66e, secs 4, Flags [Broadcast] (0x8000)
	 Gateway-IP 10.1.218.1
	 Client-Ethernet-Address 10:f9:20:e9:2b:92
	 Vendor-rfc1048 Extensions
	   Magic Cookie 0x63825363
	   DHCP-Message Option 53, length 1: Discover
	   MSZ Option 57, length 2: 1472
	   ARCH Option 93, length 2: 7
	   NDI Option 94, length 3: 1.3.10
	   Vendor-Class Option 60, length 32: "PXEClient:Arch:00007:UNDI:003010"
	   User-Class Option 77, length 4: 
	     instance#1: ERROR: invalid option
	   Parameter-Request Option 55, length 24: 
	     Subnet-Mask, Default-Gateway, Domain-Name-Server, LOG
	     Hostname, Domain-Name, RP, MTU
	     NTP, Vendor-Option, Vendor-Class, TFTP
	     BF, Option 119, Option 128, Option 129
	     Option 130, Option 131, Option 132, Option 133
	     Option 134, Option 135, Option 175, Option 203
	   T175 Option 175, length 39: 177.5.1.128.134.21.99.235.3.1.21.1.23.1.1.39.1.1.36.1.1.19.1.1.17.1.1.21.1.1.38.1.1.27.1.1.18.1.1
	   Client-ID Option 61, length 7: ether 10:f9:20:e9:2b:92
	   GUID Option 97, length 17: 0.149.211.55.208.246.51.71.102.173.224.117.67.6.24.208.86
IP (tos 0xc0, ttl 64, id 7144, offset 0, flags [none], proto UDP (17), length 401)
    10.5.33.252.67 > 10.1.218.1.67: [bad udp cksum 0x1192 -> 0x1019!] BOOTP/DHCP, Reply, length 373, hops 1, xid 0x6fedc66e, secs 4, Flags [Broadcast] (0x8000)
	 Your-IP 10.1.218.144
	 Server-IP 10.5.33.252
	 Gateway-IP 10.1.218.1
	 Client-Ethernet-Address 10:f9:20:e9:2b:92
	 Vendor-rfc1048 Extensions
	   Magic Cookie 0x63825363
	   DHCP-Message Option 53, length 1: Offer
	   Server-ID Option 54, length 4: 10.5.33.252
	   Lease-Time Option 51, length 4: 3600
	   RN Option 58, length 4: 1800
	   RB Option 59, length 4: 3150
	   BF Option 67, length 45: "http://10.5.25.26/esp/pxeboot/ipxe/boot.ipxe^@"
	   Subnet-Mask Option 1, length 4: 255.255.255.0
	   BR Option 28, length 4: 10.1.218.255
	   Default-Gateway Option 3, length 4: 10.1.218.1
	   Domain-Name Option 15, length 14: "i.megaport.com"
	   TFTP Option 66, length 12: "10.5.33.252^@"
	   Domain-Name-Server Option 6, length 8: 10.2.1.4,10.2.1.5
IP (tos 0x0, ttl 252, id 50, offset 0, flags [none], proto UDP (17), length 420)
    10.1.218.1.67 > 10.5.33.252.67: [udp sum ok] BOOTP/DHCP, Request from 10:f9:20:e9:2b:92, length 392, hops 1, xid 0x6fedc66e, secs 10, Flags [Broadcast] (0x8000)
	 Gateway-IP 10.1.218.1
	 Client-Ethernet-Address 10:f9:20:e9:2b:92
	 Vendor-rfc1048 Extensions
	   Magic Cookie 0x63825363
	   DHCP-Message Option 53, length 1: Discover
	   MSZ Option 57, length 2: 1472
	   ARCH Option 93, length 2: 7
	   NDI Option 94, length 3: 1.3.10
	   Vendor-Class Option 60, length 32: "PXEClient:Arch:00007:UNDI:003010"
	   User-Class Option 77, length 4: 
	     instance#1: ERROR: invalid option
	   Parameter-Request Option 55, length 24: 
	     Subnet-Mask, Default-Gateway, Domain-Name-Server, LOG
	     Hostname, Domain-Name, RP, MTU
	     NTP, Vendor-Option, Vendor-Class, TFTP
	     BF, Option 119, Option 128, Option 129
	     Option 130, Option 131, Option 132, Option 133
	     Option 134, Option 135, Option 175, Option 203
	   T175 Option 175, length 39: 177.5.1.128.134.21.99.235.3.1.21.1.23.1.1.39.1.1.36.1.1.19.1.1.17.1.1.21.1.1.38.1.1.27.1.1.18.1.1
	   Client-ID Option 61, length 7: ether 10:f9:20:e9:2b:92
	   GUID Option 97, length 17: 0.149.211.55.208.246.51.71.102.173.224.117.67.6.24.208.86
IP (tos 0xc0, ttl 64, id 7543, offset 0, flags [none], proto UDP (17), length 401)
    10.5.33.252.67 > 10.1.218.1.67: [bad udp cksum 0x1192 -> 0x1013!] BOOTP/DHCP, Reply, length 373, hops 1, xid 0x6fedc66e, secs 10, Flags [Broadcast] (0x8000)
	 Your-IP 10.1.218.144
	 Server-IP 10.5.33.252
	 Gateway-IP 10.1.218.1
	 Client-Ethernet-Address 10:f9:20:e9:2b:92
	 Vendor-rfc1048 Extensions
	   Magic Cookie 0x63825363
	   DHCP-Message Option 53, length 1: Offer
	   Server-ID Option 54, length 4: 10.5.33.252
	   Lease-Time Option 51, length 4: 3600
	   RN Option 58, length 4: 1800
	   RB Option 59, length 4: 3150
	   BF Option 67, length 45: "http://10.5.25.26/esp/pxeboot/ipxe/boot.ipxe^@"
	   Subnet-Mask Option 1, length 4: 255.255.255.0
	   BR Option 28, length 4: 10.1.218.255
	   Default-Gateway Option 3, length 4: 10.1.218.1
	   Domain-Name Option 15, length 14: "i.megaport.com"
	   TFTP Option 66, length 12: "10.5.33.252^@"
	   Domain-Name-Server Option 6, length 8: 10.2.1.4,10.2.1.5
IP (tos 0x0, ttl 252, id 52, offset 0, flags [none], proto UDP (17), length 432)
    10.1.218.1.67 > 10.5.33.252.67: [udp sum ok] BOOTP/DHCP, Request from 10:f9:20:e9:2b:92, length 404, hops 1, xid 0x6fedc66e, secs 18, Flags [Broadcast] (0x8000)
	 Gateway-IP 10.1.218.1
	 Client-Ethernet-Address 10:f9:20:e9:2b:92
	 Vendor-rfc1048 Extensions
	   Magic Cookie 0x63825363
	   DHCP-Message Option 53, length 1: Request
	   MSZ Option 57, length 2: 1472
	   ARCH Option 93, length 2: 7
	   NDI Option 94, length 3: 1.3.10
	   Vendor-Class Option 60, length 32: "PXEClient:Arch:00007:UNDI:003010"
	   User-Class Option 77, length 4: 
	     instance#1: ERROR: invalid option
	   Parameter-Request Option 55, length 24: 
	     Subnet-Mask, Default-Gateway, Domain-Name-Server, LOG
	     Hostname, Domain-Name, RP, MTU
	     NTP, Vendor-Option, Vendor-Class, TFTP
	     BF, Option 119, Option 128, Option 129
	     Option 130, Option 131, Option 132, Option 133
	     Option 134, Option 135, Option 175, Option 203
	   T175 Option 175, length 39: 177.5.1.128.134.21.99.235.3.1.21.1.23.1.1.39.1.1.36.1.1.19.1.1.17.1.1.21.1.1.38.1.1.27.1.1.18.1.1
	   Client-ID Option 61, length 7: ether 10:f9:20:e9:2b:92
	   GUID Option 97, length 17: 0.149.211.55.208.246.51.71.102.173.224.117.67.6.24.208.86
	   Server-ID Option 54, length 4: 10.5.33.252
	   Requested-IP Option 50, length 4: 10.1.218.144
IP (tos 0xc0, ttl 64, id 7950, offset 0, flags [none], proto UDP (17), length 401)
    10.5.33.252.67 > 10.1.218.1.67: [bad udp cksum 0x1192 -> 0x0d0b!] BOOTP/DHCP, Reply, length 373, hops 1, xid 0x6fedc66e, secs 18, Flags [Broadcast] (0x8000)
	 Your-IP 10.1.218.144
	 Server-IP 10.5.33.252
	 Gateway-IP 10.1.218.1
	 Client-Ethernet-Address 10:f9:20:e9:2b:92
	 Vendor-rfc1048 Extensions
	   Magic Cookie 0x63825363
	   DHCP-Message Option 53, length 1: ACK
	   Server-ID Option 54, length 4: 10.5.33.252
	   Lease-Time Option 51, length 4: 3600
	   RN Option 58, length 4: 1800
	   RB Option 59, length 4: 3150
	   BF Option 67, length 45: "http://10.5.25.26/esp/pxeboot/ipxe/boot.ipxe^@"
	   Subnet-Mask Option 1, length 4: 255.255.255.0
	   BR Option 28, length 4: 10.1.218.255
	   Default-Gateway Option 3, length 4: 10.1.218.1
	   Domain-Name Option 15, length 14: "i.megaport.com"
	   TFTP Option 66, length 12: "10.5.33.252^@"
	   Domain-Name-Server Option 6, length 8: 10.2.1.4,10.2.1.5
```
</details>

[image]

I have an identical server on another network with the same issue. Both servers talk to the same DHCP server via DHCP relay.

I'm using `ipxe.efi` downloaded from: https://boot.ipxe.org/

NiKiZe, 2024-11-14:
If you follow the URL in the error message from "no configuration methods succeeded" you will find several suggestions, one of them is about STP. Make sure your switch isn't blocking for a few seconds after link up.

If none of the suggestions on that page help.. is this NIC shared by any BCM controller or similar?

ianb-mp, 2024-11-15:
Thanks @NiKiZe . I've been able to confirm that 'spanning-tree portfast' is **not** enabled on the switch ports these servers connect to. Unfortunately, I can't enable it at this time in order to test the theory. If portfast was the culprit, wouldn't the initial pxeboot (prior to iPXE loading) also fail? And do you happen to know if it's possible to compile ipxe with a longer initial wait/retry time so that it would work even with portfast disabled?

NiKiZe, 2024-11-15:
Ok good, closing since this is unlikely to be an iPXE,issue, but rather a infrastructure one.

iPXE has short timeouts and not that many retries to not slow down boot process to much if there is no DHCP servers providing boot information.

Initial firmware network boot probably has longer timeouts and more retries. Once iPXE starts it initializes it's native NIC driver which brings the link down and then later up again, resetting the stp timer.

There is config options that can be used to change the DHCP timeouts and retries.

closed 2025-02-13 by zube[bot]