ipxe/ipxe #1340: make failure, looks like looping.

mocallins, 2024-11-17:
Trying to perform _make everything EMBED=embed.ipxe_
And it looks like it is looping.

```
make[2]: 'bin-x86_64-efi/ipxe.efi' is up to date.
make[2]: 'bin-x86_64-efi/ipxe.efidrv' is up to date.
make[2]: 'bin-x86_64-efi/ipxe.efirom' is up to date.
  [DEPS] interface/linux/linux_umalloc.c
  [DEPS] interface/linux/linux_timer.c
  [DEPS] interface/linux/linux_time.c
  [DEPS] interface/linux/linux_sysfs.c
  [DEPS] interface/linux/linux_smbios.c
  [DEPS] interface/linux/linuxprefix.c
  [DEPS] interface/linux/linux_pci.c
  [DEPS] interface/linux/linux_nap.c
  [DEPS] interface/linux/linux_entropy.c
  [DEPS] interface/linux/linux_console.c
  [DEPS] interface/linux/linux_acpi.c
  [DEPS] drivers/linux/tap.c
  [DEPS] drivers/linux/slirp.c
  [DEPS] drivers/linux/af_packet.c
  [DEPS] interface/linux/linux_umalloc.c
  [DEPS] interface/linux/linux_timer.c
  [DEPS] interface/linux/linux_time.c
  [DEPS] interface/linux/linux_sysfs.c
  [DEPS] interface/linux/linux_smbios.c
  [DEPS] interface/linux/linuxprefix.c
  [DEPS] interface/linux/linux_pci.c
  [DEPS] interface/linux/linux_nap.c
  [DEPS] interface/linux/linux_entropy.c
  [DEPS] interface/linux/linux_console.c
  [DEPS] interface/linux/linux_acpi.c
  [DEPS] drivers/linux/tap.c
  [DEPS] drivers/linux/slirp.c
  [DEPS] drivers/linux/af_packet.c
  [DEPS] interface/linux/linux_umalloc.c
  [DEPS] interface/linux/linux_timer.c
  [DEPS] interface/linux/linux_time.c
  [DEPS] interface/linux/linux_sysfs.c
  [DEPS] interface/linux/linux_smbios.c
  [DEPS] interface/linux/linuxprefix.c
  [DEPS] interface/linux/linux_pci.c
  [DEPS] interface/linux/linux_nap.c
  [DEPS] interface/linux/linux_entropy.c
  [DEPS] interface/linux/linux_console.c
  [DEPS] interface/linux/linux_acpi.c
  [DEPS] drivers/linux/tap.c
  [DEPS] drivers/linux/slirp.c
  [DEPS] drivers/linux/af_packet.c
^Cmake[1]: *** [Makefile.housekeeping:166: bin-i386-linux/tap.linux] Interrupt
make: *** [Makefile:170: everything] Interrupt```

mocallins, 2024-11-17:
In other words, don't use the  _EMBED_

NiKiZe, 2024-11-17:
Do use `EMBED` if you want to on targets where it makes sense, but it might not on `everything`
I'm wasn't saying "don't use embed", I did write to try it without modifications first. If it is working then find out which modification is triggering the loop.

What is your goal here? Are you running tests for development, or are you trying to build boot targets for booting?

mocallins, 2024-11-17:
I've tried with just _makeverything_ , and still looping at the same place

mocallins, 2024-11-17:
Well i was thinking that i was going to get a v2.x build , so just wanted to have them already done, for anything i might try.
when i did it manually for _bin/_, i do get a v1.21+, and it does give me the ping command, so i can at least  test if i'm getting all the routing and dns stuff correct.
So many moving parts, lol

mocallins, 2024-11-17:
Also from this page https://gist.github.com/rikka0w0/50895b82cbec8a3a1e8c7707479824c1
I enabled nfs along with ping and that goes into a booting loop.
I don't know that it is trying nfs, i just enabled both because it was on the page, i was figuring if i wasn't using nfs it would just work as normal. 
Any thoughts ?

NiKiZe, 2024-11-28:
Where should the 2.x come from?
Latest version of iPXE is 1.21.1 g5202f 5202f83345e17c56da4935205f0a97bd4aa20714
That is current git master

mocallins, 2024-12-01:
Ok, got it,  Thanks

closed 2024-12-01 by mocallins

kohlschuetter, 2025-03-17:
@mocallins The loop you encountered can happen if you're trying to build the linux binary variant but no Linux headers could be found (this is part of "everything" but the loop will also occur on bin.../ipxe.linux, for example).

On Alpine Linux, try: `apk add linux-headers`.