ipxe/ipxe #1342: make: *** [Makefile.housekeeping:1248: bin-x86_64-efi/ipxe.efi.tmp] Error 1

duferdias, 2024-11-21:
I'm trying to compile "*.efi" binaries and before finishing the prompt shows this error:

`ld: warning: bin-x86_64-efi/ipxe.efi.tmp has a LOAD segment with RWX permissions
bin-x86_64-efi/blib.a(librm.o): in function `pm_esp':
(.data.pm_esp+0x0): relocation truncated to fit: R_X86_64_32 against symbol `_estack' defined in .stack section in bin-x86_64-efi/blib.a(stack.o)
bin-x86_64-efi/blib.a(librm.o): in function `real_to_prot':
(.text16.real_to_prot+0x1): relocation truncated to fit: R_X86_64_PC16 against symbol `enable_a20' defined in .text16.early section in bin-x86_64-efi/blib.a(liba20.o)
(.text16.real_to_prot+0x8): relocation truncated to fit: R_X86_64_16 against symbol `rm_ds' defined in .text16.data.rm_ds section in bin-x86_64-efi/blib.a(librm.o)
(.text16.real_to_prot+0x22): relocation truncated to fit: R_X86_64_16 against `.bss16.rm_virt_addrs'
(.text16.real_to_prot+0x2e): relocation truncated to fit: R_X86_64_16 against `.bss16.rm_virt_addrs'
(.text16.real_to_prot+0x33): relocation truncated to fit: R_X86_64_16 against `.bss16.rm_virt_addrs'
(.text16.real_to_prot+0x39): relocation truncated to fit: R_X86_64_16 against `.data16.gdt'
(.text16.real_to_prot+0x58): relocation truncated to fit: R_X86_64_32 against `.text.real_to_prot'
bin-x86_64-efi/blib.a(librm.o): in function `r2p_pmode':
(.text.real_to_prot+0x10): relocation truncated to fit: R_X86_64_32 against `.data.pm_esp'
(.text.real_to_prot+0x17): relocation truncated to fit: R_X86_64_32 against symbol `idtr32' defined in .data section in bin-x86_64-efi/blib.a(librm_mgmt.o)
(.text.real_to_prot+0x1f): additional relocation overflows omitted from the output
make: *** [Makefile.housekeeping:1248: bin-x86_64-efi/ipxe.efi.tmp] Error 1
rm bin-x86_64-efi/version.ipxe.efi.o`

mcb30, 2024-11-21:
You have attempted to enable some feature that is BIOS-only, which cannot work in a UEFI environment. 

Start with the default, unmodified configuration, and make only the configuration changes that you actually need.

closed 2024-11-21 by mcb30

duferdias, 2024-11-21:
I had enabled CPUID_SETTINGS and MEMMAP_SETTINGS in settings.h to get processor and RAM information. From what I saw this feature cannot be compiled in efi.

mcb30, 2024-11-21:
You should find that CPUID_SETTINGS works.  MEMMAP_SETTINGS will unfortunately not work under UEFI.