ipxe/ipxe #1345: Google compute instances with >1 gVNIC interface are failing to find link

stoo-davies, 2024-11-25:
I saw issue #1274 but this appears to be different as the instances don't even get to the DHCP stage.

I brought up 2 c4-standard-2 instances using the image `projects/ipxe-images/global/images/ipxe-20240917` and a pxeboot metadata script which just prints PROGRESS to test.    Besides the number of NICs, there are no other differences.

The serial console output is below....

Single NIC instance...
```
[1miPXE 1.21.1+ (g59d12)[0m -- Open Source Network Boot Firmware -- [36mhttps://ipxe.org[0m
Features: DNS HTTP HTTPS iSCSI TFTP SRP AoE ELF MBOOT PXE bzImage Menu PXEXT
Google Compute Engine - iPXE boot via metadata
CPU: GenuineIntel INTEL(R) XEON(R) PLATINUM 8581C CPU @ 2.30GHz
net0: 42:01:c0:a8:64:3e using gve on 0000:00:03.0 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086101)]
Waiting for link-up on net0... ok
Configuring (net0 42:01:c0:a8:64:3e)...... ok
net0: 192.168.100.62/255.255.255.255 gw 192.168.100.1
net0: fe80::4001:c0ff:fea8:643e/64
http://metadata.google.internal/computeMetadata/v1/instance/attributes/ipxeboot... ok
PROGRESS
Booting from Floppy...
Boot failed: could not read the boot disk
```

Dual NIC instance...
```
[1miPXE 1.21.1+ (g59d12)[0m -- Open Source Network Boot Firmware -- [36mhttps://ipxe.org[0m
Features: DNS HTTP HTTPS iSCSI TFTP SRP AoE ELF MBOOT PXE bzImage Menu PXEXT
Google Compute Engine - iPXE boot via metadata
CPU: GenuineIntel INTEL(R) XEON(R) PLATINUM 8581C CPU @ 2.30GHz
net0: 42:01:c0:a8:64:3d using gve on 0000:00:03.0 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086101)]
net1: 42:01:c0:a8:01:16 using gve on 0000:00:04.0 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086101)]
Waiting for link-up on net0...
```

It will hang there forever.   Similarly if I put `iflinkwait --timeout 5000 net0 ||` in the embedded script it hangs forever at that point, never reaching the echo after it.

This appears to be an issue with the gve driver, as a very dirty hack of making gve_probe() immediately exit for any NIC other than the first one allows us to get past this point.   There are other problems later on, probably as a result of the hack, but I just wanted to do some tests.

The scripts I used to create the instances are below if anyone wants to try and recreate the issue...

```
gcloud compute instances bulk create \
    --count=1 \
    --name-pattern=pxetest-1nic-## \
    --project=$PROJECT \
    --image-project=ipxe-images \
    --image=ipxe-20240917 \
    --machine-type=c4-standard-2 \
    --restart-on-failure \
    --network-interface=nic-type=GVNIC,subnet=${SYS_SUBNET} \
    --metadata-from-file=ipxeboot=/tmp/ipxe_script \
    --provisioning-model=STANDARD \
    --zone=us-central1-c \
    --on-host-maintenance=TERMINATE \
    --service-account=${SERVICE_ACCOUNT} \
    --scopes=https://www.googleapis.com/auth/cloud-platform \
    --boot-disk-size=200 \
    --labels=goog-ec-src=vm_add-gcloud

gcloud compute instances bulk create \
    --count=1 \
    --name-pattern=pxetest-2nic-## \
    --project=$PROJECT \
    --image-project=ipxe-images \
    --image=ipxe-20240917 \
    --machine-type=c4-standard-2 \
    --restart-on-failure \
    --network-interface=nic-type=GVNIC,subnet=${SYS_SUBNET} \
    --network-interface=nic-type=GVNIC,subnet=${DATA_SUBNET} \
    --metadata-from-file=ipxeboot=/tmp/ipxe_script \
    --provisioning-model=STANDARD \
    --zone=us-central1-c \
    --on-host-maintenance=TERMINATE \
    --service-account=${SERVICE_ACCOUNT} \
    --scopes=https://www.googleapis.com/auth/cloud-platform \
    --boot-disk-size=200 \
    --labels=goog-ec-src=vm_add-gcloud

```

stoo-davies, 2024-11-25:
Building with DEBUG=gve gets past the link wait stage, I noticed the same thing happened when I had put my own debug prints in, which made me think it could be a timing issue.    However, in either case, it then fails at the DHCP stage, and gives the following error link...

```
[1miPXE 1.21.1+ (ge9a2)[0m -- Open Source Network Boot Firmware -- [36mhttps://ipxe.org[0m
Features: DNS HTTP HTTPS iSCSI TFTP SRP AoE ELF MBOOT PXE bzImage Menu PXEXT
Google Compute Engine - iPXE boot via metadata
CPU: GenuineIntel INTEL(R) XEON(R) PLATINUM 8581C CPU @ 2.30GHz
net0: 42:01:c0:a8:64:cf using gve on 0000:00:03.0 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086101)]
net1: 42:01:c0:a8:01:1d using gve on 0000:00:04.0 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086101)]
[31mGVE 0xffffffffeb10d8f0 IRQs at [bff5a000,bff5a080)
[0m[31mGVE 0xffffffffeb10d8f0 events at [bff5b000,bff5b008)
[0m[31mGVE 0xffffffffeb10d8f0 TX using QPL 0x18ae5458 with 8/1024 descriptors
[0m[31mGVE 0xffffffffeb10d8f0 QPL 0x18ae5458 at [bfe43000,bfe47000)
[0m[31mGVE 0xffffffffeb10d8f0 TX descriptors at [bfe3e000,bfe42000)
[0m[31mGVE 0xffffffffeb10d8f0 RX using QPL 0x18ae5258 with 64/1024 descriptors
[0m[31mGVE 0xffffffffeb10d8f0 QPL 0x18ae5258 at [bfe1d000,bfe3d000)
[0m[31mGVE 0xffffffffeb10d8f0 RX descriptors at [bfe1a000,bfe1c000)
[0m[31mGVE 0xffffffffeb10d8f0 RX completions at [bfe09000,bfe19000)
[0mWaiting for link-up on net0...[31mGVE 0xffffffffeb10d8f0 IRQ 0 doorbell +0x0000
[0m[31mGVE 0xffffffffeb10d8f0 IRQ 1 doorbell +0x0040
[0m[31mGVE 0xffffffffeb10d8f0 TX doorbell +0xc000 event counter 1
[0m[31mGVE 0xffffffffeb10d8f0 RX doorbell +0x4000 event counter 0
[0m ok
Configuring (net0 42:01:c0:a8:64:cf)...[32mGVE 0xffffffffeb10f410 IRQ 0 doorbell +0x0000
[0m[32mGVE 0xffffffffeb10f410 IRQ 1 doorbell +0x0040
[0m[32mGVE 0xffffffffeb10f410 AQ 0x01 command 0x0003 failed: 0xfffffff7
[0m[32m00000000 : 00 00 00 03 ff ff ff f7-00 00 00 00 00 00 00 00 : ................
00000010 : 00 00 00 00 bf f5 90 00-00 00 00 00 00 00 10 00 : ................
00000020 : 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00 : ................
00000030 : 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00 : ................
[0m[32mGVE 0xffffffffeb10f410 AQ error: Input/output error (https://ipxe.org/1dd65739)
[0m[32mGVE 0xffffffffeb10f410 startup failed: Input/output error (https://ipxe.org/1dd65739)
[0m
```

mcb30, 2024-11-25:
I haven't investigated yet, but my guess is that some IDs (eg QPL IDs) may need to be global across all NICs in the system.

stoo-davies, 2024-12-02:
I tried the changes from Kal, I was able to bring up a 9 interface GPU instance successfully.

closed 2024-12-03 by mcb30

mcb30, 2024-12-03:
Fixed (via #1354), and I've pushed updated public images to Google Cloud

PR #1354 (2024-12-03): [gve] Run startup process only while device is open
The startup process is scheduled to run when the device is opened and terminated (if still running) when the device is closed.  It assumes that the resource allocation performed in gve_open() has taken place, and that the admin and transmit/receive data structure pointers are therefore valid.

The process initialisation in gve_probe() erroneously calls process_init() rather than process_init_stopped() and will therefore schedule the startup process immediately, before the relevant resources have been allocated.

This bug is masked in the typi [... cut]