ipxe/ipxe #1349: autoexec. ipxe - Using fbtftp_server on Rocky Linux 9

allancarlosvr, 2024-11-27:
I tried to configure a new lab, following and replicate the tutorial:

https://bluebanquise.com/tutorials/sysadmin_servers_farm_deployment/

I passed for the next phase, but still showing error in PXE Boot process, but refer autoexec.ipxe Not found

Mode details in the image attached
[image]

NiKiZe, 2024-11-27:
Not an error, just information you are not using `autoexec.ipxe`

For more information you can search other discussions

allancarlosvr, 2024-11-27:
> Not an error, just information you are not using `autoexec.ipxe`
> 
> For more information you can search other discussions

Understood, but the boot don't make progress after this message

NiKiZe, 2024-11-27:
Ok, so you get a hang on iPXE initialising devices.
That is something different.

Which version of iPXE are you using, have you made sure that you have tried the latest?

Some things to try to narrow down the cause:
Try snponly.efi instead.
Build with something like DEBUG=efi_pci,efi_driver

And on what kind of machine is this?

allancarlosvr, 2024-12-21:
I discovered the issue @NiKiZe, and fixed it was because in my VM c001, I had checked option Enable EFI in the System Tab from Virtual Box GUI. When I disabled this option loaded normally my PXE Boot

NiKiZe, 2024-12-21:
Happy to hear that your intended use case was solved.
If your environment can't work with EFI then yes you should keep using pcbios.

However using EFI should not cause an issue or hang, and I would not call "having EFI enabled" being the issue.

closed 2025-02-25 by zube[bot]