ipxe/ipxe #1358: Older Lenovo laptops hanging after autoexec.ipxe due to Dhcp6 Veto

NiKiZe, 2024-12-09:
For some time, older Lenovo laptops have been hanging during boot. (both on IPv4 and IPv6)

Bisect gave first bad commit cb95b5b378b1a61d10770965d82dba535b6be242
Reverting and it boots fine.
As exepected with `DEBUG=efi_veto` we have
```
EFIVETO manufacturer is "LENOVO"
EFIVETO DHCP6 Protocol Driver is vetoed (Dhcp6)
```

This was added due to #1192

Potential fix for these machines, but feels fragile and there might be firmware on both sides of the fence here.
```patch
--- a/src/interface/efi/efi_veto.c
+++ b/src/interface/efi/efi_veto.c
@@ -507,13 +507,19 @@ efi_veto_vmware_uefipxebc ( EFI_DRIVER_BINDING_PROTOCOL *binding __unused,
 static int efi_veto_dhcp6 ( EFI_DRIVER_BINDING_PROTOCOL *binding __unused,
                            EFI_LOADED_IMAGE_PROTOCOL *loaded __unused,
                            EFI_COMPONENT_NAME_PROTOCOL *wtf __unused,
-                           const char *manufacturer __unused,
+                           const char *manufacturer,
                            const CHAR16 *name ) {
        static const CHAR16 dhcp6[] = L"DHCP6 Protocol Driver";
+       static const char *lenovo = "LENOVO";

-       /* Check driver name */
+       /* Check manufacturer and driver name */
+       if ( ! manufacturer )
+               return 0;
        if ( ! name )
                return 0;
+       /* At least some LENOVO laptopts hangs if this is vetoed */
+       if ( strcmp ( manufacturer, lenovo ) == 0 )
+               return 0;
        if ( memcmp ( name, dhcp6, sizeof ( dhcp6 ) ) != 0 )
                return 0;
```

Above has been tested as a fix for older 20BK (T460s) booted over both IPv4 and IPv6.
And also verified to not break on newer 21HD (T14) with recent firmware.
Untested case is newer hardware with non up to date firmware.

I have not had any issues with any Lenovo machine before commit cb95b5b37, but the original issue is only with IPv6 and that is probably still quite rare.

excedra, 2025-10-31:
> > git commit id is [0ddd830](https://github.com/ipxe/ipxe/commit/0ddd83069363f411881442857d5971654638a986) in the master branch.
> > Lenovo Type Number is 21L5-001SGE for example, which is a ThinkPad L14 G5. Installed BIOS is 1.37, which is currently newest version available by Lenovo.
> 
> AMD Ryzen

What's wrong about the Ryzen?

mcb30, 2025-11-04:
Looks as though the EDK2 bug for which the veto was added has never been addressed.  The old EDK2 bugzilla is gone, but the bug lives on as https://github.com/tianocore/edk2/issues/10506

NiKiZe, 2026-04-06:
Trying to verify this
* 20AA0014MD - Still randomly hangs, but has also worked once - This is not a device we should care about.
* 23301F4 - No hang on v4, still on v6 (expected)
* 20BV001VMS - No hang

NiKiZe, 2026-04-06:
if I see any hangs that is reproducable, and at least somewhat can narrow down the cause, there will be new issues, unless they fit into any of the already open ones, which I will again go thru time permitting.