ipxe/ipxe #1372: iPXE hangs at EFI:Stub Measured initrd data into PCR 9

laurwls, 2024-12-26:
I upgraded our FOG project server  to 1.5.10. 1626 which uses iPXE 1.21.1 (g5de5d).  Tried to capture an image but iPXE hangs up on the EFI:Stub Measured initrd data into PCR 9 step.  
[image]

Rolled FOG server back to the version that is using iPXE 1.21.1 ( g47159) and everything works fine.  
[image]

Any ideas?

NiKiZe, 2024-12-26:
Seems you are having issues with the Linux kernel?
Exactly which kernel version is it?
Could you do a git bisect between the 2 versions to find the exact commit that is causing the change in behaviour?

msteggink, 2025-03-21:
This issue doesn't seem to be related to Linux kernel or distro. I'm running into issues with different systems (different vendors) and different Linux kernels: RHEL9, Ubuntu 22, Ubuntu 24, kernels 6.8.0-31-generic, 6.2.0-35-generic, 5.15.0-25-generic, 5.14.0-427.37.1.el9_4.x86_64 all the same issue.

msteggink, 2025-03-31:
For me it seemed it was not related to the kernel. I replaced ipxe.efi with snp.efi and it works. Seems to be related to the driver in iPXE? I compiled it with debug flags and this is what I got:

[image]

[image]

NiKiZe, 2025-03-31:
@msteggink could you more precisely describe exactly the issue you are seeing.
The "EFI stub" messages mentioned by OP is from the Linux kernel, eighter there is a kernel bug, or firmware bug in play here. We shouldn't rule out that there is also a iPXE bug, but even if there is, the kernel should deal with it, and yes there is plenty of EFI bugs in older kernels.

To even debug this in iPXE, make sure this is reproducible with iPXE master, and then possibly bisect to known good version to find what changed.

mcb30, 2025-03-31:
> For me it seemed it was not related to the kernel. I replaced ipxe.efi with snp.efi and it works. Seems to be related to the driver in iPXE? I compiled it with debug flags and this is what I got:

Which precise git commit of iPXE did you build from, and what was your full build command line?

lbonrath, 2026-11-10:
Hi guys,

I can reproduce this issue on a Lenovo ThinkBook 16p G6 IAX and performed
a git bisect between a known-good iPXE revision and current master.

Hardware:
Lenovo ThinkBook 16p G6 IAX

Boot target:
Ubuntu 24.04.5 live-server amd64
UEFI / Secure Boot disabled

Symptom:
The boot consistently hangs after:

EFI stub: Loaded initrd from LINUX_EFI_INITRD_MEDIA_GUID device path
EFI stub: Measured initrd data into PCR 9

Booting the same Ubuntu ISO via Ventoy works.

I also tested snp.efi and reproduced the same behavior.

Current master is still affected:

6763bcb21 [efi] Allow direct PCI device probing when no PCI I/O protocol exists

I bisected the issue and found the first bad commit to be:

c3376f86457120e360161ce19e6862f31980c2f2
[efi] Drop to external TPL for calls to ConnectController()

I additionally verified the boundary manually:

c3376f864571^  -> boots successfully
c3376f864571   -> hangs
6763bcb21      -> hangs

Build command used during the bisect:

make bin-x86_64-efi/snp.efi \
    EMBED=/tmp/embed.ipxe \
    DEBUG=tftp,open,xhci,usb

The original boot entry is essentially:

kernel <url>/vmlinuz
initrd <url>/initrd
imgargs vmlinuz initrd=initrd ip=dhcp url=${ubuntu_server_iso} ...
boot

Removing all kernel arguments did not change the behavior.

Secure Boot is disabled. Disabling/changing kernel debug options did not
produce any additional kernel output.

I also tested chainloading GRUB and booting the same vmlinuz/initrd from
a GRUB standalone EFI image. The system still hangs immediately after
the EFI stub messages.

The bisect therefore points very specifically at c3376f864571.

NiKiZe, 2026-11-10:
@lbonrath do you have the model or P/N of the device,
also the full Lenovo firmware version in use (with the firmware version string it is easy to find the model as well, so just grab the firmware version string)

lbonrath, 2026-11-10:
@NiKiZe sure, here are the device and firmware details:

Machine Type: 21R0
Product Name: ThinkBook 16p G6 IAX
Product SKU: LENOVO_MT_21R0_BU_idea_FM_ThinkBook 16p G6 IAX
Firmware Version: R2CN61WW
Firmware Date: 09/20/2026

Let me know if you need any additional firmware or hardware information.