ipxe/ipxe #1373: Loaded image location possibly not passed to chainloaded grub?

deitch, 2025-01-02:


deitch, 2025-01-02:
Sorry, hit wrong button. Updating issue.

mcb30, 2025-01-02:
> To be honest, grub is in there only because of the challenges of getting my installer.iso mounted and running booting within iPXE.

This seems like a much easier problem to address.  Extract the kernel and initrd from within the installer.iso and boot those directly from iPXE, without involving GRUB.  This will also make the boot a lot faster, since you won't be needlessly downloading the whole ISO into RAM just for the sake of extracting the kernel and initrd files at boot time.

deitch, 2025-01-02:
Hi @mcb30 thanks for such a rapid response!

> This seems like a much easier problem to address

I am not sure it is. Even if it is, I think it would be good to know why the behaviour we are seeing above is happening. 

> This seems like a much easier problem to address

Is it? An iso has a whole filesystem on it. In this case, it is an installer, whose iso filesystem looks like:

```
$ sudo mount installer.iso /mnt
$ ls /mnt
EFI  bin  boot  config  config.img  containers  dev  etc  home  init  lib  media  mnt  opt  persist  persist.img  proc  root  rootfs.img  run  sbin  srv  sys  tmp  usr  var
```

There is its own grub on it, which, in the end, loads an initrd and a kernel and boots the kernel (albeit with some complex configuration; installers 🤷‍♂️ ), and then uses all of the information in the iso filesystem to do installation.

How would I do that directly from iPXE?

NiKiZe, 2025-01-02:
What is the message you get from grub when it fails?
To be clear we are talking about the boot location/path of the efi binary?

Historically I have a abused wimboot to get grub to find it's config file, if I have had to use grub.

Have you actually been able to use "the iso" from within the booted kernel at all?
The booted kernel is unlikely (without special patches) to map the iso into a virtual block device.

So as said before at several places, never try to boot an iso unless you are using sanboot over iscsi. With Linux, always extract the relevant files from iso.

deitch, 2025-01-02:
> What is the message you get from grub when it fails?

It fails to find its grub.cfg. It tries at `tftp://192.168.1.2/EFI/BOOT/grub.cfg` instead of at `http://...`. If I place the `grub.cfg` file in the tftpboot directory, it succeeds, but that only is an option in a testbed.

> To be clear we are talking about the boot location/path of the efi binary?

I will try to clarify:

* `ipxe.efi` is available via `tftp` (because that is all many older PXE firmwares support)
* `BOOTX64.EFI` (grub) is available via http
* `grub.cfg` is available via http
* `installer.iso` is available via http

> Have you actually been able to use "the iso" from within the booted kernel at all?

Yes, quite. If I start with a device which supports iPXE natively, then everything is http from the beginning and works, grub sees its `$prefix` as http, etc. If I "cheat" and place the grub files - `BOOTX64.EFI`, `grub.cfg` and `installer.iso` in tftp, it also works, although sometimes the size of the iso can hit limits.

Everything is fine when all http, or all tftp (minus the size issues). It is the "need to start tftp to get to iPXE, then http to the rest" that misses the "you are loaded via http". Which is what led me to believe that it is possibly an EFI Loaded Image issue, although that is not much more than a semi-educated guess.

> never try to boot an iso unless you are using sanboot over iscsi. With Linux, always extract the relevant files from iso

I am open to better options, although this is distributing an installer _en masse_ over a network. tftp is fine for local network boot when needed, but deploy it to many nodes across a globally distributed private network, let alone across the Internet, and http is the only choice.

How else would you do it? Would you extract the entire filesystem, and make it available to the booting node how?

Either way, I think it would be good to figure out if we are missing something in chainbooting and figuring it out.

NiKiZe, 2025-01-02:
You said that "it works" but is the contents of the iso that was loaded by grub, also available in your "installer" and does the install of the contents from that virtual iso work 100% of the way and not only the pre kernel stage.

Since the answer of the above usually is, "no it's not available after kernel boot" you do extract the iso and then either preload the data you need in the iPXE stage, or provide an URL to the initrd scripts and installer to know where to fetch any next stage files.

deitch, 2025-01-02:
> You said that "it works" but is the contents of the iso that was loaded by grub, also available in your "installer" and does the install of the contents from that virtual iso work 100% of the way and not only the pre kernel stage

Yes. I included the `grub.cfg` further up, repeating it here:

```
echo "Downloading installer. This may take some time. Please wait patiently."
loopback loop0 ($cmddevice)/installer.iso
set root=loop0
set isnetboot=true
export isnetboot
configfile ($root)/EFI/BOOT/grub.cfg
```

It creates a loopback, so it has access to the entire ISO filesystem, and then loads the `grub.cfg` from within there, and it has access to everything, since it is mounted at `loop0`. grub supports using loopback devices like this.

> you do extract the iso

Or mount it as a loopback, as above.

If there is a cleaner automated way to get this directly through iPXE over http, I would be more than happy to eliminate grub, although eventually it will launch grub. The iso contains a rather complex configuration (see the last line in the `grub.cfg`, which loads the one on the ISO via the loopback), so eventually, it all becomes grub.

Either way, I think the core issue is, is chaining missing some EFI variable setting? How do I even check for and test that?

NiKiZe, 2025-01-02:
Yes grub has... But does the booted Linux kernel? (3rd time asking)

No, you wouldn't start grub at all, just migrate whatever options you need so that iPXE can tell your firmware to load the Linux kernel in EFI mode directly.

deitch, 2025-01-02:
> yes, grub has... but does the booted Linux kernel?

I don't understand. Grub has what?

In the end, I think this iso will boot grub, because there is a really complex boot menu. Think of how a lot of installers work with lots of variants. 

So the current boot is:

1. PXE to iPXE via tftp 
1. iPXE loads cfg via tftp
1. iPXE chains to grub via http
1. Grub loads config file via http (or should, this fails because it tries tftp)
1. That config tells grub to loopback mount iso, then read grub config file from loop mount
1. Does rest of installer by booting installer kernel with options

I'd be really happy to simplify it, but that grub cfg is used for other installation methods as far as I know. And in any case, figuring out why the loaded image or whatever isn't set right for grub to pick it up is worthwhile.

NiKiZe, 2025-01-02:
So you only want the iso to boot grub?
You do not need the contents of the iso to be available when the kernel has booted?

deitch, 2025-01-02:
Sure we need the contents of the iso. Once the iso is loopback mounted, the grub config in it loads the kernel from it and initrd and runs an install process. 

Conveniently it's identical to just booting the ISO off a cd device or similar. 

But I keep getting distracted. Isn't the key question, why is chain (apparently) not setting the EFI vars such that grub picks it up? Whatever problem I'm trying to solve in my specific case, isn't that secondary to the primary issue?

NiKiZe, 2025-01-02:
> Sure we need the contents of the iso. Once the iso is loopback mounted, the grub config in it loads the kernel from it and initrd and runs an install process.
>

Ok, and you have verified that the contents actually is there?

> Conveniently it's identical to just booting the ISO off a cd device or similar.

No it's not. A physical storage is present, which the kernel knows how to find.
In the case of loopback in grub, in theory the iso data is stored in RAM, and the kernel needs to identify this region, make sure it isn't corrupted, and then connect this as a block device.
I have never seen this being done other than with memdisk and out of kernel patches, so please understand that I'm hesitant to think you actually do see the iso content from within the booted kernel.

> 
> But I keep getting distracted. Isn't the key question, why is chain (apparently) not setting the EFI vars such that grub picks it up? Whatever problem I'm trying to solve in my specific case, isn't that secondary to the primary issue?

No efi vars, this should be in the boot path, but you are trying to do something that many have tried before and it can't work, unless you can show that it would work in your case, there is no point in wasting time on anything else.

deitch, 2025-01-02:
Yeah I can confirm that it works. If I boot from a device with support for http network booting built in, or if I put everything via tftp (avoid the "protocol switch"), then it runs all the way though. Not going to claim it is ideal, but it works. 

If you're saying it shouldn't work, I'm not sure what to respond. Your knowledge of the low level details exceeds mine. 

> No efi vars, this should be in the boot path

I don't understand. When I chain to grub, grub is setting its cmdpath to `tftp,192.168.1.2`. That's not hard coded in, it must be getting that from somewhere. That has to be EFI vars, doesn't it? And shouldn't those be set initially to tftp and that address by UEFI firmware when it loads ipxe.efi, and shouldn't it get changed when ipxe chains to a different .EFI file? 

Forget for a minute that it is grub. Any EFI app I chain, shouldn't it get EFI vars indicating where it (the EFI app) was loaded from, and not where the first one in the chain (ipxe.efi) was loaded from?

deitch, 2025-01-02:
I'm more than happy to test my theory that the EFI vars are not set correctly when chain loading. I'm just at a loss as to how to do that. Maybe compile an EFI app that just reports all the variables it finds, chain load it from ipxe, and see? I don't know how to write that.

Starrry, 2025-01-16:
Hello @deitch , I encountered the same problem as you. Since the system kernel is in ELF format, it cannot be booted directly with iPXE and needs to call GRUB. I am using `chain http` to load `grub.efi`. However, after loading, the cmdpath is `(tftp, 172.0.0.1)`, which prevents it from loading `grub.cfg` located in the same directory as `grub.efi`. I look forward to your solution.

NiKiZe, 2025-01-16:
> Hello @deitch , I encountered the same problem as you. Since the system kernel is in ELF format, it cannot be booted directly with iPXE and needs to call GRUB. I am using `chain http` to load [...]

Easy, get a kernel that has CONFIG_EFI_STUB enabled, drop grub.
If you still need to load grub, do so by abusing the powers of wimboot to get a virtual in memory filesystem to put needed files on.

Starrry, 2025-01-17:
@NiKiZe Thank you for your reply. I want to start a kernel based on the loong64 architecture, and neither of these methods is easy for me.I am trying to use other methods to work around this issue.

deitch, 2025-01-17:
@Starrry yup, exactly that. My contention is that this may not be just a grub issue, but rather a question of what it reads from. Meaning either iPXE misses setting some EFI var (and therefore it will affect other types of chain where we switch protocols) and so we can fix it there, or possibly something that is inherent to EFI.

It also is possible it is a grub error.

I was hoping to narrow it down by figuring out exactly what EFI var grub is reading, which would make this conversation more productive.

Here is a bit of a trail.

It starts [here](https://git.savannah.gnu.org/gitweb/?p=grub.git;a=blob;f=grub-core/kern/main.c;h=d29494d54fba7140eb8529e5ec4ffc840a85bd0a;hb=HEAD#l133) where it gets set from `grub_machine_get_bootlocation`:

```c
  grub_machine_get_bootlocation (&fwdevice, &fwpath);

  if (fwdevice)
    {
      char *cmdpath;

      cmdpath = grub_xasprintf ("(%s)%s", fwdevice, fwpath ? : "");
      if (cmdpath)
	{
	  grub_env_set ("cmdpath", cmdpath);
	  grub_env_export ("cmdpath");
	  grub_free (cmdpath);
	}
    }
```

For EFI, `grub_machine_get_bootlocation` is set [here](https://git.savannah.gnu.org/gitweb/?p=grub.git;a=blob;f=grub-core/kern/efi/init.c;h=1637077e1e96499c9df77d4da7c9011d3dba3dbf;hb=HEAD#l130):

```c
void
grub_machine_get_bootlocation (char **device, char **path)
{
  grub_efi_loaded_image_t *image = NULL;
  char *p;

  image = grub_efi_get_loaded_image (grub_efi_image_handle);
  if (!image)
    return;
  *device = grub_efidisk_get_device_name (image->device_handle);
  if (!*device && grub_efi_net_config)
    {
      grub_efi_net_config (image->device_handle, device, path);
      return;
    }

  *path = grub_efi_get_filename (image->file_path);
  if (*path)
    {
      /* Get the directory.  */
      p = grub_strrchr (*path, '/');
      if (p)
        *p = '\0';
    }
}
```

And then `grub_efi_get_loaded_image ` is set [here](https://git.savannah.gnu.org/gitweb/?p=grub.git;a=blob;f=grub-core/kern/efi/efi.c;h=b93ae3aba12db0337e5713634456f448e602f4bd;hb=HEAD#l158):

```c
grub_efi_loaded_image_t *
grub_efi_get_loaded_image (grub_efi_handle_t image_handle)
{
  return grub_efi_open_protocol (image_handle,
				 &loaded_image_guid,
				 GRUB_EFI_OPEN_PROTOCOL_GET_PROTOCOL);
}
```

And then `grub_efi_open_protocol ` is set [here](https://git.savannah.gnu.org/gitweb/?p=grub.git;a=blob;f=grub-core/kern/efi/efi.c;h=b93ae3aba12db0337e5713634456f448e602f4bd;hb=HEAD#l100):

```c
void *
grub_efi_open_protocol (grub_efi_handle_t handle,
			grub_guid_t *protocol,
			grub_efi_uint32_t attributes)
{
  grub_efi_boot_services_t *b;
  grub_efi_status_t status;
  void *interface;

  b = grub_efi_system_table->boot_services;
  status = b->open_protocol (handle,
			     protocol,
			     &interface,
			     grub_efi_image_handle,
			     0,
			     attributes);
  if (status != GRUB_EFI_SUCCESS)
    return 0;

  return interface;
}
```

Finally, `GRUB_EFI_OPEN_PROTOCOL_GET_PROTOCOL ` is set [here](https://git.savannah.gnu.org/gitweb/?p=grub.git;a=blob;f=include/grub/efi/api.h;h=b686e8afeed2b4c65ed35a7799e7073d38d35148;hb=HEAD#l58):

```c
#define GRUB_EFI_OPEN_PROTOCOL_GET_PROTOCOL		0x00000002
```

And `GRUB_EFI_LOADED_IMAGE_GUID ` is set [here](https://git.savannah.gnu.org/gitweb/?p=grub.git;a=blob;f=include/grub/efi/api.h;h=b686e8afeed2b4c65ed35a7799e7073d38d35148;hb=HEAD#l77):

```c
#define GRUB_EFI_LOADED_IMAGE_GUID	\
  { 0x5b1b31a1, 0x9562, 0x11d2, \
    { 0x8e, 0x3f, 0x00, 0xa0, 0xc9, 0x69, 0x72, 0x3b } \
  }
```

@NiKiZe do you know how to interpret it? What is grub looking for, and how can I check it? I couldn't find anything that matches when booting PXE->iPXE->Shell.

NiKiZe, 2025-01-17:
AFAIK there is no vars, it should be a image path, but grub defaulting to tftp on 127... Is just odd.

deitch, 2025-01-17:
It gets the IP address correctly and knows the protocol was tftp, so it must have found something.

Do you know how to read that? I know conceptually is is reading the EFILoadedImage, but I am not really sure what that means, or how to retrieve that, e.g. if I use the grub shell. Or, for that matter, to write a simple EFI program that would get loaded, spit the vars out to console, and then just hang would be fine too.

NiKiZe, 2025-01-17:
You keep writing "vars" there is no such thing.
See the EFI image device path, examples https://github.com/ipxe/ipxe/issues/1316

My guess is grub has just bad defaults based on some part of the path, which makes it default to tftp

deitch, 2025-01-17:
> You keep writing "vars" there is no such thing.

Fair enough.

> See the EFI image device path, examples https://github.com/ipxe/ipxe/issues/1316

Yes! That example is solid. 

```
EFI image device path PciRoot(0x0)/Pci(0x1C,0x5)/Pci(0x0,0x0)/MAC(4CD98F60AF47,0x1)/IPv4(0.0.0.0,TCP,DHCP,10.64.48.139,10.64.48.1,255.255.252.0)/Uri(http://apt.wikimedia.org/efiboot/snponly.efi)
```

The example you shared comes from iPXE. How can I print that out? Actually, now that I think about it, I could replicate the situation by doing:

PXE --> iPXE over tftp --> iPXE over http (this one prints that out)

Could I do that? Sounds much easier than trying to compile something special. Is it this part? `make bin-x86_64-efi/snponly.efi  DEBUG=efi_autoexec,efi_utils,efi_init,dns`, could I do the same with `ipxe.efi`?

> My guess is grub has just bad defaults based on some part of the path, which makes it default to tftp

It has to be either that or within iPXE. I suspect you are right. If I can do the above test, I can provide the value it is reading, and then take it up with the grub folks.

NiKiZe, 2025-01-17:
If you can show that iPXE fails the early autoexec.ipxe load when chained from iPXE then there is something that iPXE probably needs to fix.

Before going to grub tho, let's make sure we know exactly what needs fixing.
And a heads up, if you mention iPXE to grub folks, they will just say, "use grub for everything"

deitch, 2025-01-17:
> heads up, if you mention iPXE to grub folks, they will just say, "use grub for everything"

Heh, I am trying really hard not to get in the middle of anything. There is a "bridge" between them - probably EFI Loaded Image, like you showed - so I am trying to use that. I want to show, "iPXE set it to the Uri xxxx", and that proves iPXE did it right (or wrong), and then I can go to the grub side with it and show, "Uri is xxxx but grub reads it as y, there is a bug here."

"Just the facts, ma'am" (gosh, that was a really old show, before my time, but still a good line).

So I can chain iPXE -> autoexec.ipxe? What could I chainload that would show that output line like above? This:

```
EFI image device path PciRoot(0x0)/Pci(0x1C,0x5)/Pci(0x0,0x0)/MAC(4CD98F60AF47,0x1)/IPv4(0.0.0.0,TCP,DHCP,10.64.48.139,10.64.48.1,255.255.252.0)/Uri(http://apt.wikimedia.org/efiboot/snponly.efi)
```

deitch, 2025-01-17:
If I hadn't said it before, thanks for helping @NiKiZe 😄

deitch, 2025-01-20:
I tried to follow your suggestion in the linked issue. I built `snponly.efi` with debug enabled (I probably could have done the same with `ipxe.efi`, but might as well stick to the recommendation exactly). 

```
make bin-x86_64-efi/snponly.efi DEBUG=efi_utils,efi_init,dns
```

I then chained: PXE -> ipxe.efi (over tftp) -> snponly.efi (debug, over http)

Logs:

```
iPXE 1.0.0+ -- Open Source Network Boot Firmware -- http://ipxe.org
Features: DNS FTP HTTP HTTPS iSCSI TFTP SRP AoE EFI Menu
Configuring (net0 52:54:00:12:34:56)...... ok
Booting from tftp://192.168.1.2/ipxe.efi.cfg
tftp://192.168.1.2/ipxe.efi.cfg... ok
http://192.168.1.2:8080/snponly.efi... ok
EFI handle 0x7e4ca118 systab 0x7f9ec018
EFI does not provide protocol ConsoleControl
EFI protocol ef9fc172-a1b2-4693-b327-6d32fc416042 is at 0x7f309ae8
EFI protocol AcpiTable is at 0x7f36cca0
EFI protocol 8b843e20-8132-4852-90cc-551a4e4a7f1c is at 0x7f38c420
EFI protocol 3152bca5-eade-433d-862e-c01cdc291f44 is at 0x7f312c90
EFI configuration table eb9d2d30-2d88-11d3-9a16-0090273fc14d is at 0x7fb7e000
EFI configuration table f2fd1544-9794-4a2c-992e-e5bbcf20e394 is at 0x7f91a000
EFI configuration table eb9d2d31-2d88-11d3-9a16-0090273fc14d is at 0x7f91c000
EFI image base address 0x7e17d000
EFI image device path PciRoot(0x0)/Pci(0x2,0x0)/MAC(525400123456,0x1)/MAC(525400123456,0x1)
iPXE initialising devices...ok
```

The loaded image device path is in the second to last line, it doesn't even include the Uri, unlike the linked issue, which has:

```
EFI image device path PciRoot(0x0)/Pci(0x1C,0x5)/Pci(0x0,0x0)/MAC(4CD98F60AF47,0x1)/IPv4(0.0.0.0,TCP,DHCP,10.64.48.139,10.64.48.1,255.255.252.0)/Uri(http://apt.wikimedia.org/efiboot/snponly.efi)
```

Suggestions as to what I can do next?

NiKiZe, 2025-01-20:
The linked issue is when firmware does http boot.
Here you had different iPXE loaded first, and then loaded one with debug enabled, would be nice to see both with debug enabled for comparison.

@mcb30 shouldn't iPXE include ip and maybe url in the device path somehow?
Maybe make device path, or part of device path configurable?

deitch, 2025-01-20:
Sure, I will hit that right away.

deitch, 2025-01-20:
Pretty much the same thing:

```
Filename: ipxe.efi
tftp://192.168.1.2/ipxe.efi... ok
ipxe.efi : 1008704 bytes [EFI]
EFI handle 0x7e4ca518 systab 0x7f9ec018
EFI does not provide protocol ConsoleControl
EFI protocol ef9fc172-a1b2-4693-b327-6d32fc416042 is at 0x7f309ae8
EFI does not provide protocol c12b1873-ac17-4176-ac77-7ecb4defffec
EFI protocol AcpiTable is at 0x7f36cca0
EFI protocol 8b843e20-8132-4852-90cc-551a4e4a7f1c is at 0x7f38c420
EFI protocol 3152bca5-eade-433d-862e-c01cdc291f44 is at 0x7f312c90
EFI configuration table eb9d2d30-2d88-11d3-9a16-0090273fc14d is at 0x7fb7e000
EFI configuration table f2fd1544-9794-4a2c-992e-e5bbcf20e394 is at 0x7f91a000
EFI configuration table eb9d2d31-2d88-11d3-9a16-0090273fc14d is at 0x7f91c000
EFI image base address 0x7df9e000
EFI image device path PciRoot(0x0)/Pci(0x2,0x0)/MAC(525400123456,0x1)/MAC(525400123456,0x1)
iPXE initialising devices...ok
```

Did I miss something in the debug build?

NiKiZe, 2025-01-20:
I expect to see the output from both iPXE boots.
One "EFI image device path" for each instance of iPXE 
But if the last output is from firmware, then grub should behave identical, regardless of if it was booted from firmware directly, or from iPXE, it would not get a http path.

deitch, 2025-01-21:
Nah that's from ipxe.efi. It only appears when I compiled it with the DEBUG= options, and you can see it loading from the first lines I included. 

The lines from the second load over http are identical to what was earlier. I can rerun and paste both here in one output later.

NiKiZe, 2026-03-16:
Thanks for taking the time to verify, really appreciate it!