ipxe/ipxe #1377: `boot || goto` does not work

edgood1, 2025-01-08:
I have this logic in my ipxe scripts, which aims to prevent machines from getting stuck when the server isn't serving images.  I'm pretty sure that it worked on a previous version of ipxe but no longer works as expected:

```
set attempt:int32 1

:start
imgfree
initrd http://<url>/initrd.igz <params> || goto retry
kernel http://<url>/kernel.vmlinuz || goto retry
boot || goto retry

:retry
iseq ${attempt} 5 && reboot
inc attempt
echo "Boot failed, waiting 10 seconds before retrying (Attempt ${attempt} of 5)...
sleep 10
goto start
```

I've recently found several machines stuck on boot failures when I accidentially injected a syntax error in the initrd line, which inaccurately reports something like "Image too large". In any event, I would expect the machines to enter a reboot loop but they just get stuck and require a manual reboot. Am I missing something obvious here?

NiKiZe, 2025-01-08:
Can you be more precise in what kind of syntax error you injected, as well as well as the stage and full error message of the `Image too large`

If your prams does introduce a iPXE script syntax error, then the logic could become something different.
If not I would expect the message you was seeing to be coming from the kernel or it's init scripts.

A tip, if you aren't already, is to give the kernel a panic=30 or similar for automatic reboots at that stage as well.

edgood1, 2025-01-09:
I did some testing on a libvirt VM and was able to replicate the problem under a few different scenarios:

1. If the ipxe script gets truncated, ipxe presumably fails to parse it and throws the error `Could not boot image: Result too large  (https://ipxe.org46022085)` which is confusing. 
Here's the script, which gets truncated after the kernel line:
```
#!ipxe

set attempt:int32 1

echo ==== FastTrack iPXE LifeCycle script for demo_vm_2 in state defined ====
sleep 10

dhcp net10 || dhcp || goto retry

:start
imgfree
echo Downloading and extracting images.

imgextract --name netboot-amd64 http://192.168.100.1:8080/image_files/netboot-amd64.igz || goto retry
imgextract --name lifecycle_overlay http://192.168.100.1:8080/image_files/lifecycle_overlay.igz || goto retry
kernel http://192.168.100.1:8080/image_files/netboot-amd64.vmlinuz ramdisk_size=16778240 root=/dev/ram0 rw initrd=netboot-amd64 initrd=lifecycle_overlay %
```
Screenshot of the resulting boot:
[image]

The above is the case I mentioned in my previous post. My syntax error actually rendered an an incomplete ipxe script, not an invalid kernel parameter as I originally stated. The script was truncated right after the kernel line, and never rendered the `boot` statement. It's worth noting that the `||` seems to work fine in the `dhcp net10` line. 

2. The more concerning case, which is the reason for my opening this issue is that if the kernel or any of the files fail to download I see a `Not found` error but then `Could not boot: Result too large` followed by `Could not boot image: Reult too large`. ipxe never enters any of the  "or" `||` statements. I intentionally renamed the kernel file to something that doesn't exist on the web server. Here's the complete script, but with a bad filename for the kernel image:
```
#!ipxe

set attempt:int32 1

echo ==== FastTrack iPXE LifeCycle script for demo_vm_2 in state defined ====
sleep 10

dhcp net10 || dhcp || goto retry

:start
imgfree
echo Downloading and extracting images.

imgextract --name netboot-amd64 http://192.168.100.1:8080/image_files/netboot-amd64.igz || goto retry
imgextract --name lifecycle_overlay http://192.168.100.1:8080/image_files/lifecycle_overlay.igz || goto retry
kernel http://192.168.100.1:8080/image_files/Xnetboot-amd64.vmlinuz ramdisk_size=16778240 root=/dev/ram0 rw initrd=netboot-amd64 initrd=lifecycle_overlay server_name=demo_vm_2 console=ttyS0 console=tty0 nousb api_url=http://192.168.100.1:8080/api/v1 lifecycle_state=defined api_uuid=0894ee24-dd40-4785-9eff-761e3e9de929 || goto retry
boot || goto retry

:retry
iseq ${attempt} 5 && reboot
inc attempt
echo "Boot failed, waiting 10 seconds before retrying (Attempt ${attempt} of 5)...
sleep 10
goto start
```

and the resulting boot:
[image]

3.) The behaviour seems to be the same a the previous scenario if I change the name of one of the images requested by `imgextract`. I added an X to the first imagename.
```
#!ipxe

set attempt:int32 1

echo ==== FastTrack iPXE LifeCycle script for demo_vm_2 in state defined ====
sleep 10

dhcp net10 || dhcp || goto retry

:start
imgfree
echo Downloading and extracting images.

imgextract --name netboot-amd64 http://192.168.100.1:8080/image_files/Xnetboot-amd64.igz || goto retry
imgextract --name lifecycle_overlay http://192.168.100.1:8080/image_files/lifecycle_overlay.igz || goto retry
kernel http://192.168.100.1:8080/image_files/netboot-amd64.vmlinuz ramdisk_size=16778240 root=/dev/ram0 rw initrd=netboot-amd64 initrd=lifecycle_overlay server_name=demo_vm_2 console=ttyS0 console=tty0 nousb api_url=http://192.168.100.1:8080/api/v1 lifecycle_state=defined api_uuid=0894ee24-dd40-4785-9eff-761e3e9de929 || goto retry
boot || goto retry

:retry
iseq ${attempt} 5 && reboot
inc attempt
echo "Boot failed, waiting 10 seconds before retrying (Attempt ${attempt} of 5)...
sleep 10
goto start

```
[image]

4.) I switched `imgextract` to `initrd` and see the same behaviour
```
#!ipxe

set attempt:int32 1

echo ==== FastTrack iPXE LifeCycle script for demo_vm_2 in state defined ====
sleep 10

dhcp net10 || dhcp || goto retry

:start
imgfree
echo Downloading and extracting images.

initrd http://192.168.100.1:8080/image_files/Xnetboot-amd64.igz || goto retry
initrd http://192.168.100.1:8080/image_files/lifecycle_overlay.igz || goto retry
kernel http://192.168.100.1:8080/image_files/netboot-amd64.vmlinuz ramdisk_size=16778240 root=/dev/ram0 rw initrd=netboot-amd64 initrd=lifecycle_overlay server_name=demo_vm_2 console=ttyS0 console=tty0 nousb api_url=http://192.168.100.1:8080/api/v1 lifecycle_state=defined api_uuid=0894ee24-dd40-4785-9eff-761e3e9de929 || goto retry
boot || goto retry

:retry
iseq ${attempt} 5 && reboot
inc attempt
echo "Boot failed, waiting 10 seconds before retrying (Attempt ${attempt} of 5)...
sleep 10
goto start
```

[image]

5.) If I stop the webserver after ipxe downloads the first image I see the similar behaviour when the second image fails to download:
[image]

My initial claim that `boot || goto` doesn't work might not have been completely accurate. It's actually the `kernel` `initrd` `imgextract` commands that don't honour the `||` upon failure. I'm not sure how to test the `boot ||` case.

The expected behaviour is that if any of the commands fail we jump to the `:retry` block to prevent the machine from getting stuck. In my observations they will fail and get stuck. 

 I'm fairly certain that this worked as expected on a previous version of ipxe . A secondary concern are the misleading `Result too large` errors.

Please let me know if I can provide more information.

NiKiZe, 2025-01-09:
All iPXE errors do include a ipxe.org url, include that and also follow that link for advice, it's there for a reason ;)

What I think happens in your first case is that the script exits, and then your outer script that called the inner one might try to continue executing. Another clue here is that iPXE does continue to execute some kind of script in most of your examples.

I understand that you dont want this to happen at all, but with a corrupt script, iPXE behaviour will become unreliable. To reduce this risk, do define your failure handler earlier...

Do you maybe have the retry label already defined in outer scripts?

~I would still expect something on the kernel line to be unexpected, non ascii, what you can do is to only fetch kernel with kernel and then use imgargs to split.~ I believe more in the duplicate retry labels.

edgood1, 2025-01-10:
Ah! I forgot to include the inner script. This is what gets served by tftp to standard pxe clients. It has similar goto logic. I'm going to run some tests to see if eliminating the goto logic in the parent script has any impact on the issue. 

```
#!ipxe

set attempt:int32 1

echo ========= iPXE Chainloader ===============

:start

dhcp || goto baseretry

chain http://${net0/gateway}:8080/api/v1/ipxe/${serial}?mfg=${manufacturer}&model=${product} || goto baseretry

:baseretry
iseq ${attempt} 5 && reboot
inc attempt
echo "iPXE chainloading failed, retrying in 10 seconds. (Attempt ${attempt} of 5 before reboot)...
sleep 10
goto start
```

There may very well be some sort of control shift problem happening here.

edgood1, 2025-01-11:
I removed everything in the parent script except the `dhcp` and `chain` lines and it didn't change the previous bevavior. Do you think that using nested scripts is a contributing factor?

edgood1, 2025-01-14:
I figured out the problem. by adding `||` after `iseq ${attempt} 5 && reboot` it now works as expected. I will close this out as user error 😜

closed 2025-01-14 by edgood1