ipxe/ipxe #1385: iPXE freezes shortly after launching on Soekris 4501

Kurlon, 2025-01-22:
So first, yup, ye olde hardware, I'm likely the only one these days playing with it, fully aware there won't be much heavy lifting going on to make this work.

Box is a Soekris 4501, AMD Elan (486DX-4 equivalent shoved into a SoC) with 64MB RAM. The box supports booting off Compact Flash or PXE via any one of three National Semiconductor DP8381 NICs. My goal is to get the box booting Gentoo via iSCSI, so far my compromise is booting via pxelinux to an NFS root, iSCSI volume is then attached for swap 'cause of course the kernel can't be told an iSCSI LUN to snag via boot options.

I've tried using iPXE both chain loading it over PXE and writing it to CF and it fails the same way with both: It starts, identifies the active NIC as UNDI, asks me to hit Ctrl-B to get to a prompt... and never responds to anything ever again. If I append a script to it I can get it to do DHCP and report an address before hanging. It seems to live for a very brief interval after start up. I've played around with custom config setups, mostly disabling the regular pc console and just using serial, I've not been able to get debug output to display though to show where it's hanging. Same behavior using undionly.

Any suggestions on how to debug this further?

Kurlon, 2025-01-22:
Example boot log:

```
comBIOS ver. 1.33  20080103  Copyright (C) 2000-2007 Soekris Engineering.

net45xx

0064 Mbyte Memory                        CPU Elan SC520 133 Mhz 

Pri Mas  HYPERSTONE FLASH DISK           LBA 997-16-32  255 Mbyte

Slot   Vend Dev  ClassRev Cmd  Stat CL LT HT  Base1    Base2   Int 
-------------------------------------------------------------------
0:00:0 1022 3000 06000000 0006 2280 00 00 00 00000000 00000000 
0:16:0 13A3 0020 0B400000 0116 0280 10 3C 00 A0000000 A0002000 10
0:18:0 100B 0020 02000000 0107 0290 00 3F 00 0000E001 A0010000 11
0:19:0 100B 0020 02000000 0107 0290 00 3F 00 0000E101 A0011000 05
0:20:0 100B 0020 02000000 0107 0290 00 3F 00 0000E201 A0012000 09

 5 Seconds to automatic boot.   Press Ctrl-P for entering Monitor.

comBIOS Monitor.   Press ? for help.

> boot F0

NSC DP83815/DP83816 Fast Ethernet UNDI, v1.03                                  
Copyright (C) 2002, 2003 National Semiconductor Corporation
All rights reserved.

Pre-boot eXecution Environment  PXE-2.0 (build 082)
Copyright (C) 1997-2000  Intel Corporation

CLIENT MAC ADDR: 00 00 24 C4 E8 C0  
CLIENT IP: 192.168.192.17  MASK: 255.255.255.0  DHCP IP: 192.168.192.9         
GATEWAY IP: 192.168.192.1 
PXE->EB: !PXE at 9E64:0060, entry point at 9E64:00F6                           
         UNDI code segment 9E64:1396, data segment 9B73:2F10 (621-639kB)
         UNDI device is PCI 00:12.0, type DIX+802.3
         639kB free base memory after PXE unload
iPXE initialising devices...

iPXE 1.21.1+ (gc929) -- Open Source Network Boot Firmware -- https://ipxe.org
Features: DNS HTTP iSCSI TFTP SRP AoE ELF MBOOT PXE bzImage Menu PXEXT

Press Ctrl-B for the iPXE command line...
```

And no response from there, I have to hard cycle the machine. I'm not having any luck building older releases to see if they'll play ball, they all fail with various compiler issues. I may roll back to period correct distro releases to match the release dates to see if that helps.

NiKiZe, 2025-01-22:
TIVOLI_VMM_WORKAROUND comes to mind,
But do consider checking out gPXE era version, and bisect from there to find if there is some specific change that has caused boot to fail.

Challenge will be to find compiler compatible with that old code.

Kurlon, 2025-01-22:
So, now I'm a bit extra confused... did some digging, the last time a 486 was confirmed to have worked was 2022, and even better the test hardware was what I am using: https://x.com/ipxe/status/1488889177559212047?lang=en

Commit https://github.com/ipxe/ipxe/commit/bc35b24e3 should work. Dug out an Ubuntu 20.04 box to build on, cloned, started a git bisect with that commit as the good point and went through every test and... none worked, same behavior across the board. Did a git reset --hard bc35b24e3 and, still failed the same way?

I was able to build v1.20.1 from the tarball, it doesn't even initialize, same with v1.21.1 from the tarball. v1.0.0 won't build, will need an older setup to try with.

Using Archive.org, grabbed ipxe.pxe from May 2022, just after the Feb commit that should have the fix, no joy, same bug I've been chasing. Earliest snap from 2014, doesn't initialize. Tried all available after the May 2022 snap, same bug I'm chasing.

I'm thinking I'm hitting both a code issue AND a build tool issue at the same time now?

Kurlon, 2025-01-22:
Ubuntu 10.04 will build v.0.9.4 with some warnings, undionly.kpxe doesn't do anything after being loaded. I couldn't build the desired commit due to binutils / ar issues, binutils this old doesn't understand deterministic archives.

Building bc35b24 on Ubuntu 14.04 however, there are signs of life. I can get to the ipxe command line on my Soekris 4501 now. Continuing to poke, but it appears I've got a viable test point finally.

NiKiZe, 2026-03-16:
I'll keep an eye out for devices like this and if I find one try to narrow down the issue.
Fixing it depends on the impact on other code.