ipxe/ipxe #1389: Error "TLS 0xXXXXXX could not generate random data: Invalid argument" after multiple HTTPS chain loading PHP interactions. --Error code: Could not start download: Invalid argument (1c25e002).

nethlon, 2025-01-24:
**Background and Issue Description:**
We use an iPXE USB key to boot, then chain load PHP interactions via HTTPS connections to input information, followed by chain loading different system images. When booting in Legacy Mode (BIOS), after chain loading PHP several times (about 7-8 times), iPXE displays the error shown in the screenshot. If there are only two or three simple PHP connections, we can successfully boot the system image. However, if we connect to PHP multiple times (around 7-8 times, after DRBG reseed appears), this error always occurs. We have tried this on different models including DELL Server R660, R760, and notebooks (with different network cards), all experiencing the same issue.
However, using the same configuration parameters to compile ipxe.efi for UEFI boot mode, we do not encounter this problem at all.

[image]

**The PHP chain loading code is as follows, FYI:**
> echo ":retry_boot1\n";
>echo "sleep 3\n"; 
>echo 'chain [https://${bootip}/ipxeboot/1.php?tc=${serial:uristring}](https://${bootip}/ipxeboot/1.php?tc=$%7bserial:uristring%7d) || goto retry_boot1';`

> echo ":retry_boot2\n"; 
> echo "sleep 3\n"; 
> echo 'chain [https://${bootip}/ipxeboot/2.php?S=${S_Number:uristring}&V=${V_Number:uristring}](https://${bootip}/ipxeboot/2.php?S=$%7bS_Number:uristring%7d&V=$%7bV_Number:uristring%7d) || goto retry_boot2';`

**We compiled ipxe.lkrn with the following parameters:**
> #define DOWNLOAD_PROTO_HTTPS 
> /* Enable certificate management */ 
> #define CERT_CMD 
> #define IMAGE_TRUST_CMD 
> #define IMAGE_CRYPT_CMD 
> /* Enable PING_CMD */ 
> #define PING_CMD

**We compiled ipxe.lkrn using this command:**
> make bin/ipxe.lkrn TRUST=ca.crt DEBUG=device,x509,tls,https,rbg,drbg,random,entropy,efi_entropy

We have used several of the latest release versions to compile, all of which resulted in the same error (latest version: c2f21a21852741e869b5a64d02d6a49ef16a94cd). We also tried the following modifications without success:
1. In /include/ipxe, modifying the value of DRBG_SECURITY_STRENGTH to 64:
> -- #define DRBG_SECURITY_STRENGTH 128
> ++ #define DRBG_SECURITY_STRENGTH 64`

2. In /crypto/drbg.c, modifying the drbg_reseed method:
> -- unsigned int entropy_bits = DRBG_SECURITY_STRENGTH;
> ++ unsigned int entropy_bits = ( ( 3 * DRBG_SECURITY_STRENGTH + 1 ) / 2 );

We appreciate the convenience IPXE has brought us and are glad to see recent releases focusing on encrypted transmission. We hope this issue can be resolved as well. Thank you very much!

NiKiZe, 2025-01-24:
The relevant message is actually invalid argument https://ipxe.org/1c25e002
Notice the not to be ignored URL.

There is not enough random data to stay secure... I think you can type random thinks on the keyboard, or have more random packages on the network.

nethlon, 2025-01-24:
Hi Nilsson, 

Thank you for your prompt reply.
In fact, during the process of connecting to PHP interactions, we need to input some data using the keyboard. Why is there still a lack of entropy in such a scenario? Can I solve this problem by simulating the input of more random values through a PHP script?

such as: 
for ($i = 0; $i < 10; $i++) {
    echo bin2hex(random_bytes(4)) . "\n";  // Generates 4 random bytes (8 hex characters)
}

nethlon, 2025-02-05:
I hope you have the opportunity to review these questions,  I'm committed to conducting further experiments based on your recommendations. Appreciate any guidance you can offer. 

> 1. Could you advise how to input random data during the process? Are there any similar cases with successful solutions that you're aware of?
> 2. Is it possible to lower the encryption strength to match the reduced entropy situation, for example, by adjusting the DRBG_SECURITY_STRENGTH value?
> 3. Could you please explain how to add more random packages from the network?
> 4. Are there any other methods to increase entropy during the IPXE chain load process? For example, by modifying parts of the source code?

Thank you in advance for your support.

mcb30, 2025-02-07:
>     1. Could you advise how to input random data during the process? Are there any similar cases with successful solutions that you're aware of?

There is no way to inject additional entropy within iPXE.  Keystroke data etc are not used, only the selected entropy source (which for BIOS, will be based on timer skew between the nanosecond-scale CPU cycle counter and the millisecond-scale legacy PIC timer).

>     2. Is it possible to lower the encryption strength to match the reduced entropy situation, for example, by adjusting the DRBG_SECURITY_STRENGTH value?

No: only a single security strength is supported.

>     3. Could you please explain how to add more random packages from the network?

This is also not possible to to in iPXE.

>     4. Are there any other methods to increase entropy during the IPXE chain load process? For example, by modifying parts of the source code?

Unfortunately not.

I'm interested to know how you came to the conclusion that it is reseeding that is causing the issue.  In your screenshot, I see the message "DRBG 0x.... not valid", which indicates that `state->valid == 0` at that point.  From reinspecting the code, nothing ever clears `state->valid`: it is set to 1 at the end of a successful call to `drbg_instantiate()` and then can never be cleared.

This doesn't align with your description of a problem that arises only after a few successful attempts to generate entropy.  Could you please share more details of how you concluded that reseeding is when the problem arises?

NiKiZe, 2025-02-07:
Sorry I was wrong here,
Could it be that you have some other failure which then starts a new instance of iPXE, or that you have multiple layers of iPXE and due to a failure in inner it exits to the outer?

nethlon, 2025-02-08:
@mcb30 @NiKiZe Hello team, I'm glad to have both of you here and thank you for your helpful feedback.

> I'm interested to know how you came to the conclusion that it is reseeding that is causing the issue. In your screenshot, I see the message "DRBG 0x.... not valid", which indicates that state->valid [...]

> This doesn't align with your description of a problem that arises only after a few successful attempts to generate entropy. Could you please share more details of how you concluded that reseeding is when the problem arises?

After reinspecting the code, I believe the issue occurs in drbg_generate() rather than drbg_reseed(). Therefore, I retract my previous statement that reseeding is causing the issue.

The issue occurs in BIOS (Legacy) boot mode, while UEFI boot mode works fine.

Let me walk you through the entire process to provide a better understanding of how the issue arises. This might help us pinpoint the source of the problem more effectively.

1. iPXE boots up, rdrand gathers 192 bits in 24 samples:
[image]

2. System chain load 1.php to input the booting information:
[image]

3. CChain load 2.php to input the information again:
[image]

4. Chain load 3.php to verify if the inputs match, then prompt to enter "Y/N" to confirm the inputs are correct. Here, I enter "N" to simulate incorrect information and repeat the interaction:
[image]

5. "4.php" receives the "N" input, then chain loads back to "1.php". During this, we can see the DRBG was reseeded, and the entropy reduced to "128 bits in 16 samples". Input the booting information again:
[image]

6. Chain load to 2.php to input the information again. We can see the DRBG was reseeded, and the entropy remains at "128 bits in 16 samples":
[image]

7. Chain load to 3.php, entered "N" to repeat the interaction. During this time, we can see the DRBG was reseeded, and the entropy still remains at "128 bits in 16 samples":
[image]

8. System failed to chain load the next php page, with the following error message:
[image]

9. The system retries to chain load the next php page every 3 seconds, and the error message repeats:
```
    echo ":retry_boot4\n"; 
    echo "sleep 3\n";
    echo 'chain https://${bootip}/ipxeboot/4.php?S='.$S.'&V='.$V.'&C='.$C.'&Y=${Confirmed:uristring}&OS='.$OS.' || goto retry_boot4';
```
[image]

To adapt to the decreased entropy, I tried changing the DRBG Algorithm, but the situation remained unchanged.
```
//#define HMAC_DRBG_ALGORITHM HMAC_DRBG_SHA256  /* masked by nethlon on 2025/02/06 */
//#define HMAC_DRBG_ALGORITHM HMAC_DRBG_SHA512  /* added by nethlon on 2025/02/06 */
#define HMAC_DRBG_ALGORITHM HMAC_DRBG_SHA1  /* added by nethlon on 2025/02/06 */
```

Appreciate any further support you can provide. Thank you.

mcb30, 2025-02-10:
>     1. iPXE boots up, rdrand gathers 192 bits in 24 samples:
>
>     5. "4.php" receives the "N" input, then chain loads back to "1.php". During this, we can see the DRBG was reseeded, and the entropy reduced to "128 bits in 16 samples". Input the booting information again:

This is not a reduction in entropy, but simply reflects the different entropy input requirements between instantiation and reseeding.

For a DRBG with a 128 bit security strength, instantiation requires 128 bits of min-entropy from the entropy source plus an additional 64 bits of min-entropy as a nonce.  The nonce is permitted to be generated from the same entropy source, and so we do this in iPXE by simply requesting 192 bits of min-entropy during instantiation.

Reseeding does not require a nonce, and so only 128 bits are requested from the entropy source when reseeding occurs.

>     6. Chain load to 2.php to input the information again. We can see the DRBG was reseeded, and the entropy remains at "128 bits in 16 samples":

I am curious to know why you are encountering such frequent reseeding.  The default value for `HMAC_DRBG_RESEED_INTERVAL` is 1024 requests, so you should not be seeing it happen at all with only the small number of requests in your example.

I have tried a quick test script to cause repeated DRBG generation:

```
set count:int32 0
:loop
inc count
echo Attempt ${count}...
imgfetch https://boot.ipxe.org/hw.ipxe || goto failed
imgfree
goto loop
:failed
shell
exit
```

I have successfully run this for over 2000 iterations with no problems.  Even if I reduce `HMAC_DRBG_RESEED_INTERVAL` to an artifically low value to trigger frequent reseeding, I still encounter no problems:

```
Attempt 2010...
https://boot.ipxe.org/hw.ipxe...DRBG 0x4bd5c0 generate
 ok
Attempt 2011...
https://boot.ipxe.org/hw.ipxe...DRBG 0x4bd5c0 generate
DRBG 0x4bd5c0 reseed
ENTROPY rdrand enabled
ENTROPY rdrand passed 5 startup tests
ENTROPY rdrand disabled
ENTROPY rdrand gathered 128 bits in 16 samples
 ok
Attempt 2012...
https://boot.ipxe.org/hw.ipxe...DRBG 0x4bd5c0 generate
 ok
Attempt 2013...
https://boot.ipxe.org/hw.ipxe...DRBG 0x4bd5c0 generate
DRBG 0x4bd5c0 reseed
ENTROPY rdrand enabled
ENTROPY rdrand passed 5 startup tests
ENTROPY rdrand disabled
ENTROPY rdrand gathered 128 bits in 16 samples
 ok
```

So, since I am unable to reproduce the issue, and since you are experiencing frequent reseeding attempts that ought not to occur, I think at this point I have to ask what modifications you have made to the iPXE code.  Please minimise the number of modifications that you have made, and see if you can still reproduce the problem.

nethlon, 2025-02-11:
@mcb30 Hi Machael,
Thank you for explaining the details.I haven’t made any additional changes to the iPXE code

**I compiled ipxe.lkrn with the following parameters:**
```
#define DOWNLOAD_PROTO_HTTPS
/* Enable certificate management /
#define CERT_CMD
#define IMAGE_TRUST_CMD
#define IMAGE_CRYPT_CMD

/* Enable PING_CMD */
#define PING_CMD
```

**I used the following command to compile**
`make bin/ipxe.lkrn TRUST=ca.crt DEBUG=device,x509,tls,https,rbg,drbg,random,entropy,efi_entropy`

I tried to run the same quick test script in my environment and got the same results as you.

After conducting further testing, I now have some potential clues to share with you. Below are the test cases I performed:
**Test Case 1:**
In the 1st case, I I modified the test1.php to repeatedly chain-load test2.php. It worked without issues. 
```
test1.php
<?php
header("Content-Type: text/plain");
echo "#!ipxe\n";
echo "echo \n";
//echo "set counter:int32 0 \n";
echo ":loop \n";
echo "echo \n";
echo "inc counter \n";
echo "echo Attempt \${counter}...\n";
echo "echo This is test1.php...\n";
echo "echo Counter in test1: \${counter}\n";
echo "sleep 1\n";
echo "chain [https://\${bootip}/ipxeboot/test2.php](https://${bootip}/ipxeboot/test2.php) || goto failed\n";
echo "goto loop\n";
echo ":failed\n";
echo "shell\n";
echo "exit\n";
?>
```
```
test2.php
<?php
header("Content-Type: text/plain");
echo "#!ipxe\n";
echo "echo This is test2.php...\n";
echo "echo Counter in test2: \${counter}\n";
//echo "chain [https://\${bootip}/ipxeboot/test1.php](https://${bootip}/ipxeboot/test1.php)";
?>
```
**Test Case 2:**
When I added a chain-load command in "test2.php" to return to "test1.php",  the script stopped after 6 iterations:
```
test2.php
<?php
header("Content-Type: text/plain");
echo "#!ipxe\n";
echo "echo This is test2.php...\n";
echo "echo Counter in test2: \${counter}\n";
echo "chain [https://\${bootip}/ipxeboot/test1.php](https://${bootip}/ipxeboot/test1.php)";
?>
```

**Test Case 3:**
I modified "test1.php" to chain-load itself (i.e., recursive chain-loading). The system triggered reseeding after 10 iterations and stopped.
```
test1.php
<?php
header("Content-Type: text/plain");
echo "#!ipxe\n";
echo "echo \n";
//echo "set counter:int32 0 \n";
echo ":loop \n";
echo "echo \n";
echo "inc counter \n";
echo "echo Attempt \${counter}...\n";
echo "echo This is test1.php...\n";
echo "echo Counter in test1: \${counter}\n";
echo "sleep 1\n";
echo "chain [https://\${bootip}/ipxeboot/test1.php](https://${bootip}/ipxeboot/test1.php) || goto failed\n";
echo "goto loop\n";
echo ":failed\n";
echo "shell\n";
echo "exit\n";
?>
```

From these tests, I found that repeated chain-loading of the same file eventually triggered reseeding. 
However, the same scripts worked perfectly fine in a UEFI environment.

I hope these findings provide some clues to help identify and resolve the issue. Thank you!

mcb30, 2025-02-11:
@nethlon Please avoid nesting multiple levels of iPXE script, and your problem will go away.  You may want to investigate the `--replace` and `--autofree` options to the `chain` command: these let you effectively replace the current script with the new script (rather than nesting the call to the new script) and automatically free scripts after execution.

closed 2025-02-11 by mcb30

nethlon, 2025-02-13:
@mcb30 @NiKiZe 
After applying both options to the chain loading, the issue was resolved.
In fact, I had previously added the '--autofree' option to the chain load, but I overlooked the usage of '--replace'. Moreover, since the same script worked fine when using HTTP protocol and in the UEFI environment, I didn't investigate further in this direction.
Thank you very much for your assistance in solving this tricky problem.