ipxe/ipxe #1390: The imgfree command is required to chain boot

purpleidea, 2025-01-26:
First of all, thanks for ipxe, it's such a useful tool.

I've been using it with https://github.com/purpleidea/mgmt/ and I got stuck on an issue. Long story short, it turns out some other folks I know had the same issue. We tracked it down. I wrote up the details here:

https://purpleidea.com/blog/2025/01/23/working-around-an-ipxe-issue/

TL;DR: You need an extra `imgfree` command, where it's not expected. It turns out the netboot.xyz also uses this workaround.

I'd appreciate an explanation (if this is not a bug) and a docs patch if that's appropriate.

/cc @frebib 

Thanks again

NiKiZe, 2025-01-26:
You are correct that you in some cases do need `imgfree` to give the kernel the correct initrd.
Or to be more precise, have the correct initrds loaded when the kernel boots, one way to work around left behinds from earlier parts of script is to use `imgfree`

Why is that? To know you will have to first run `imgstat` to check what you have put there to begin with.
Remove the `imgfree` you have now and add `imgstat` and `pause` before your `boot`

This is actually very much expected, so not a bug, but I think it is better if we jump into that once we have `imgstat` output from your specific failure case.
I wish I could say that it also is documented, but this is a mix of old `initrd=` missing on command line, and newer posts with same or similar root causes as here.
https://github.com/search?q=repo%3Aipxe%2Fipxe+not+syncing%3A+VFS&type=discussions

Did this change? Well kind of yes.
In the past for EFI boot you needed to tell the kernel which initrd filename to load, but iPXE has another nice feature, appending additional files into the existing cpio archive, this has "always" worked in pcbios, but was "recently" added for EFI, around this time support for protocol telling the kernel about the initrd without giving it explicitly on command line was added.

purpleidea, 2025-01-26:
As one example, reading these docs: https://ipxe.org/howto/fedora
It never occurred to me that I would need imgfree, and it doesn't mention that there.
I also didn't see anything on the https://ipxe.org/cmd/imgfree page which might be useful for some users.

How to reproduce? Use an ipxe on a USB key, and then use that to `chain http://.../menu.ipxe` which contains kernel/initrd/boot lines.
Without the imgfree, I always get the VFS panic.

If sending a patch to the above pages would be a helpful doc fix, please let me know and I'll be happy to contribute one.

Thanks again!

NiKiZe, 2025-01-26:
Please provide `imgstat` output for when this happens.

NiKiZe, 2025-01-28:
I do believe that the examples mentioned have been used in combination with other examples.
The examples mentioned is unlikely to miss behave on their own.

Rather, other examples, maybe use of background images, might need to be updated to use `imgfree`, or to not use `--keep` (but this is just a guess without seeing `imgstat` output)

purpleidea, 2025-01-29:
(I'm just travelling omw to FOSDEM/CfgMgmtCamp, so I won't be able to get to this until after this trip when I'm back at real hardware...)

purpleidea, 2026-01-03:
zomg. Sorry it's been so long. Been kind of buried trying to start/run an open source company.

I don't want to make any promise other than I'm happy using the `imgfree` workaround like netboot.xyz and some other folks I know are using in production.

Thanks again for IPXE it's really amazing work.

NiKiZe, 2026-01-03:
Yes `imgfree` is needed if something else is leaving behind "files" that is not expected to be there when for example a kernel boots.
This was behaving differently before in EFI but should not be much of a difference in PCBIOS.

So the question should be, what is left behind that needs to be cleaned, and what is leaving it behind?
Should you maybe have some `--autofree` in chains to scripts?

But by grabbing `imgstat` we can find an alternate (correct) solution, and also narrow down where something else has gone wrong, documentation wise. (We can not patch documentation, if we don't know which documentation needs patching)

I don't see this as an real issue so closing, but if there is bad examples somewhere that is causing this, and we can help fix, I'm happy to do what can be done. But I will need to get pointers to which examples that is ;)

purpleidea, 2026-10-09:
/cc @jordansissel