ipxe/ipxe #1391: Dell PowerEdge R860 exception

defo89, 2025-01-27:
Hi, I am trying boot a new batch of Dell servers via the onboard NIC:
```
Model: PowerEdge R860
BIOS: Version 2.4.4
iDRAC Firmware Version: 7.10.90.00
```

I've built iPXE from [this](https://github.com/ipxe/ipxe/tree/c2f21a21852741e869b5a64d02d6a49ef16a94cd) commit.
`make bin-x86_64-efi/ipxe.efi DEBUG=pci,efi_pci`

DHCP  and .efi file fetch via TFTP is working: 
[image]

After downloading kernel and initrd I get an exception screen: 
[image]

Has anyone faced similar issues?

Not sure if this is the cause, but before starting kernel/initrd download EFI debug logs: 

[image]

NiKiZe, 2025-01-27:
Your first picture shows messages from Linux kernel, what steps have been done to narrow this doen to an iPXE issue?

Is there a management interface, and is that possibly shared with the NIC used to boot iPXE?

Have you tried snponly.efi to see if the output changes in any way?

defo89, 2025-01-27:
Thanks for the hint! I have build an snponly image `make bin-x86_64-efi/snponly.efi` and was able to boot all 9 servers of same model successfully. 

Does it mean that drivers for onboard nics included in `ipxe.efi` image might have some incompatibility?

defo89, 2025-02-07:
@NiKiZe your suspicion about the firmware bug was correct. After a session with Dell, here are some findings:
- even though we boot from onboard 1Gb NICs, there is an issue with `ipxe.efi` during loading or releasing drivers for 100Gb cards (`Broadcom BCM57508 2x100G QSFP PCIE`)
- `ipxe.efi` worked when we disabled PCI devices (where 100Gb cards are connected) in BIOS (Integrated Devices - Slot Disablement in iDRAC)
- `ipxe.efi` worked when 100Gb card firmware was downgraded from `23.11.16.22` (latest) to `22.92.06.10`
- when we upgraded back to `23.11.16.22` - RSOD again when using `ipxe.efi`

We will see whether Dell/Broadcom will have a solution for this in the next firmware- but at least this issue will hopefully help others if they encounter RSOD :)

avk999, 2025-03-05:
I believe I've got the same issue with a server having broadcom 57504-based card.
The server is PowerEdge 6715 with BIOS 1.0.7 (also tried with 1.0.5, no difference). It has several cards including the built-in one:
```
61:00.0 Ethernet controller: Broadcom Inc. and subsidiaries BCM57504 NetXtreme-E 10Gb/25Gb/40Gb/50Gb/100Gb/200Gb Ethernet (rev 12)
        DeviceName: NIC in Slot 5 Port 1-1
        Subsystem: Broadcom Inc. and subsidiaries NetXtreme-E BCM57504 4x25G OCP3.0
        Flags: bus master, fast devsel, latency 0, IRQ 87, NUMA node 0, IOMMU group 31
        Memory at 382e05030000 (64-bit, prefetchable) [size=64K]
        Memory at 382e04000000 (64-bit, prefetchable) [size=16M]
        Memory at 382e05058000 (64-bit, prefetchable) [size=32K]
        Expansion ROM at 7c000000 [disabled] [size=256K]
        Capabilities: [48] Power Management version 3
        Capabilities: [50] Vital Product Data
        Capabilities: [a0] MSI-X: Enable+ Count=74 Masked-
        Capabilities: [ac] Express Endpoint, MSI 00
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [13c] Device Serial Number 6c-92-cf-ff-fe-4c-63-60
        Capabilities: [150] Power Budgeting <?>
        Capabilities: [160] Virtual Channel
        Capabilities: [180] Vendor Specific Information: ID=0000 Rev=0 Len=020 <?>
        Capabilities: [1b0] Latency Tolerance Reporting
        Capabilities: [1b8] Alternative Routing-ID Interpretation (ARI)
        Capabilities: [230] Transaction Processing Hints
        Capabilities: [300] Secondary PCI Express
        Capabilities: [200] Precision Time Measurement
        Capabilities: [358] Physical Layer 16.0 GT/s <?>
        Capabilities: [388] Lane Margining at the Receiver <?>

61:00.0 0200: 14e4:1751 (rev 12)
        DeviceName: NIC in Slot 5 Port 1-1
        Subsystem: 14e4:5045

 ethtool -i  enp97s0f0np0
driver: bnxt_en
version: 5.14.0-284.11.1.el9_2.x86_64
firmware-version: 231.0.154.0/pkg 23.11.16.22
expansion-rom-version:
bus-info: 0000:61:00.0
supports-statistics: yes
supports-test: yes
supports-eeprom-access: yes
supports-register-dump: yes
supports-priv-flags: no

(3 other ports omitted for brewity), PCI id is 0

```

When I use snponly.efi everything works as it should. Using ipxe.efi (commit `be3a78eaf804a2c437aa055d5c1e2f4a1310a0c1`) fails:

1. ifstat doesn't show that Broadcom card at all
2. After successfull fetching of kernel and initrd the last message I see is 'EFI Stub: measuring initrd into PCR9` and the system resets.

After disabling the card in iDrac I was able to boot the system with ipxe.efi.

I've intercepted via serial console EFIPCI and PCI debug messages related to this card:
```
EFIPCI 0000:61:00.0 (14e4:1751 class 020000) has driver "14e4-1751"
EFIPCI 0000:61:00.0 (14e4:1751 class 020000) has driver "14e4-1751"
EFIPCI 0000:61:00.0 (14e4:1751 class 020000) has driver "14e4-1751"
0000:61:00.0 (14e4:1751) has driver "14e4-1751"
0000:61:00.0 has mem 382e05030000 io 0 irq 255
0000:61:00.0 device not enabled by BIOS! Updating PCI command 0406->0407
0000:61:00.0 latency timer is unreasonably low at 0. Setting to 32.
EFIPCI 0000:61:00.0 using driver "14e4-1751"
EFIPCI 0000:61:00.1 (14e4:1751 class 020000) has driver "14e4-1751"
EFIPCI 0000:61:00.1 (14e4:1751 class 020000) has driver "14e4-1751"
EFIPCI 0000:61:00.1 (14e4:1751 class 020000) has driver "14e4-1751"
0000:61:00.1 (14e4:1751) has driver "14e4-1751"
0000:61:00.1 has mem 382e05020000 io 0 irq 255
0000:61:00.1 device not enabled by BIOS! Updating PCI command 0006->0007
0000:61:00.1 latency timer is unreasonably low at 0. Setting to 32.
EFIPCI 0000:61:00.1 using driver "14e4-1751"
EFIPCI 0000:61:00.2 (14e4:1751 class 020000) has driver "14e4-1751"
EFIPCI 0000:61:00.2 (14e4:1751 class 020000) has driver "14e4-1751"
EFIPCI 0000:61:00.2 (14e4:1751 class 020000) has driver "14e4-1751"
0000:61:00.2 (14e4:1751) has driver "14e4-1751"
0000:61:00.2 has mem 382e05010000 io 0 irq 255
0000:61:00.2 device not enabled by BIOS! Updating PCI command 0006->000y
0000:61:00.2 latency timer is unreasonably low at 0. Setting to 32.
EFIPCI 0000:61:00.2 using driver "14e4-1751"
EFIPCI 0000:61:00.3 (14e4:1751 class 020000) has driver "14e4-1751"
EFIPCI 0000:61:00.3 (14e4:1751 class 020000) has driver "14e4-1751"
EFIPCI 0000:61:00.3 (14e4:1751 class 020000) has driver "14e4-1751"
0000:61:00.3 (14e4:1751) has driver "14e4-1751"
0000:61:00.3 has mem 382e05000000 io 0 irq 255
0000:61:00.3 device not enabled by BIOS! Updating PCI command 0006->0007
0000:61:00.3 latency timer is unreasonably low at 0. Setting to 32.
EFIPCI 0000:61:00.3 using driver "14e4-1751"
```

"Device not enabled by BIOS" confuses me a bit because the card slot is not disabled (but the card is not enabled for PXE boot).

I wasn't able to capture anything from kernel after `Measuring initrd into PCR9` even with using `earlyprintk` to redirect to serial, probably crashes even earlier.

NiKiZe, 2025-03-05:
And what's the outcome for both of you when using snp.efi? (Different from snponly.efi)

avk999, 2025-03-06:
> And what's the outcome for both of you when using snp.efi? (Different from snponly.efi)

snp.efi on the virtual cd worked, ifstat has shown all the interfaces and the system booted. Thank you for the tip @NiKiZe , I guess this will do at least as a temporary solution.

Can you please explain the difference between snp.efi and ipxe.efi?