ipxe/ipxe #1394: Google compute a3-ultragpu-8g instances are having gVNIC interface issues after loading linux from iPXE

stoo-davies, 2025-01-31:
If I bring up an a3-ultragpu-8g instance in GCP with a single NIC (gVNIC/gve) and boot into linux from a boot disk, the interface is functional.
However, if I load the same linux image from the network with iPXE, the NIC is not functional once linux is loaded.

If the instance has 2 interfaces, after booting via iPXE the primary interface which was used by iPXE to download the kernel/initrd is not functional, the additional interface is.

I have raised a case with GCP support, they gave me the following info...
-- SNIP --
Issue Statement:
IPXE uses third-party "gve" driver to work with Google's gvnic device, which initialized into the legacy mode. When loaded into the ubuntu image the new driver expect a clear state from the gvnic device while it seems IPXE doesn't clean up the driver state clearly which makes new driver busy waiting the hardware resources.

AI: IPXE team to help to reset the old "gve" driver status to unblock new driver load successfully
-- SNIP --

Is there some additional cleanup that can be added to the gve code to reset the NIC back to a clean state when booting?

The iPXE boot looks normal (single nic example)...

```
[1m[37m[40miPXE 1.21.1+ (g24db39)[0m[37m[40m -- Open Source Network Boot Firmware -- [0m[36m[40mhttps://ipxe.org[0m[37m[40m
Features: DNS HTTP HTTPS iSCSI TFTP VLAN SRP AoE EFI Menu
Google Compute Engine - iPXE boot via metadata
CPU: GenuineIntel INTEL(R) XEON(R) PLATINUM 8581C CPU @ 2.10GHz
net0: 42:01:0a:8e:00:19 using gve on 0000:00:12.0 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
Waiting for link-up on net0... ok
Configuring (net0 42:01:0a:8e:00:19)...... ok
net0: 10.142.0.25/255.255.128.0 gw 10.142.0.1
net0: fe80::4001:aff:fe8e:19/64
http://metadata.google.internal/computeMetadata/v1/instance/attributes/ipxeboot... ok
http://10.142.0.2:8080/ipxe... ok
http://10.142.0.2:8080/tftpboot/images/default-image/vmlinuz... ok
http://10.142.0.2:8080/tftpboot/images/default-image/initrd... 84%     ok
EFI stub: Loaded initrd from LINUX_EFI_INITRD_MEDIA_GUID device path
EFI stub: Measured initrd data into PCR 9
``` 

However, the Linux GVE driver then shows an error...
```
[   48.706364] gve 0000:00:12.0: Driver is running with DQO RDA queue format.
[   48.706368] gve 0000:00:12.0: MAC addr: 42:01:0a:8e:00:19
[   48.706379] gve 0000:00:12.0: TX queues 16, RX queues 16
[   48.706381] gve 0000:00:12.0: Max TX queues 16, Max RX queues 16
[   78.739966] gve 0000:00:12.0: AQ command failed with status -12
[   78.739975] gve 0000:00:12.0: could not setup device_resources: err=-62
``` 

I can run with DEBUG=gve or other options if you need that output.
I am running the public GCP iPXE image, project ipxe-images, family ipxe-uefi-x86-64.

NiKiZe, 2025-01-31:
Is this a new issue with new or updated kernel driver? If so GCP would probably like to fix that regression.

On top of that they can reach out with information on which state the NIC needs to be left in, while also taking in to account for example an iSCSI boot scenario.

stoo-davies, 2025-01-31:
The linux kernel driver is not new, we are using the same Ubuntu inbox module (1.0.0) that we've used in the past.     We did also try with the latest dkms module from github (1.4.5) which behaved the same way.

I will point Google support at this issue for comment.

mcb30, 2025-02-07:
@stoo-davies iPXE will reset the NIC by writing a zero to the admin queue page frame number register (at offset 0x10).  This is the documented method for performing a reset according to https://github.com/torvalds/linux/blob/master/Documentation/networking/device_drivers/ethernet/google/gve.rst#device-resets 👍 

> A device reset is triggered by writing 0x0 to the AQ PFN register. This causes the device to release all resources allocated by the driver, including the AQ itself.

Please ask Google support to update their documentation if it is not correct, or at least to comment here to say what else they believe iPXE ought to be doing.