ipxe/ipxe #1402: Support for the Dell Broadcom 57504 Quad-Port 10/25 GbE

arun2arunraj, 2025-02-09:
hi Team,

We are trying to install RockyLinux 8.10 for the Broadcom NIC via ipxe image.

Model Name : Dell Broadcom 57504 Quad-Port 10/25 GbE 

We are getting the below error message from the hardware. Please help to get it sorted.

NOTE : Unfortunately, we dont find the above hardware in the following page. https://ipxe.org/appnote/hardware_drivers

NiKiZe, 2025-02-09:
You should grab output of `ifstat`
You should also collect the pciid of the device.

Your output shows..
NETDEV net0 unknown network protocol aaaa
NETDEV net0 failed to receive 0x0: Operation not supported (https://ipxe.org/3c086083)
Nothing here says it is an error, how did you come to such conclusion?

All errors in iPXE does contain these urls, make sure you include them, as text, if issues are created, also do read those pages.

In this case it does seem that you have created a debug build of iPXE probably due to some original issue.
It is always better to include the actual problem you are sseing, and not some debug output from something that might not at all be relevant.

There is probably 2 ways to get rid of the specific message, either put the device on an isolated network without random traffic (so you don't get packets with protocol field set to aaaa), or don't use debug builds. (As the url in the message properly does explain)
Not all messages are errors.

Are you running DHCP and does that fail, or what is the actually issue?

arun2arunraj, 2025-02-09:
hi

Thanks for the quick reply. We are getting the protocol not supported message while the packet received to the nic. 

Any help would be appreciated

NiKiZe, 2025-02-09:
You are posting debug output from a non error, while still not posting the ipxe.org url, and not reading that url.

You will have to start with explaining the actual issue you are having. As in, what are you trying to do, and what isn't working as you expect it to.

To be clear, the below output from your picture, is only there because you have enabled debug options that shows details that you shouldn't care about unless you are actually having issues with a TCP flow, and if you actually do, you should isolate it on its own network so you can debug only that traffic, and not random garbage  that is totally normal in a production environment.
NETDEV net4 failed to receive 0x0: Error 0x440e6083 (https://ipxe.org/440e6083)

NiKiZe, 2025-02-09:
In your edit, you have output from a script so hard to say what it thinks it is failing on, you would have to share the script.

But better than that, in your first picture you did ifopen net0, maybe you should do ifopen net2, but the question is then still what do you expect to happen that doesn't, for most that would be a dhcp configuration attempt. If not you probably have static configuration, but you still need something that results in "this isn't working" and you have to tell us what that is.

arun2arunraj, 2025-02-09:
hi @NiKiZe 

Noted on the 0x440e6083, I have updated the comment exactly when you have revert back in parallel. Apologize for the edit, Thought to have a more focused and aligned conversation in single place.

The error message which we have faced is following for your reference below. just to debug, we have enabled debug message to understand the error in better way, basically we are just opening a boot interface and we are stuck with the issue.

Our IPXE script :

Our objective : plumb an IPV6 IP for each of the interface and try to pull the kernel image to boot the hardware if the connection is successful. We are facing issue while we open the interface to plumb an IPv6 IP.

`#!ipxe
echo "List of interfaces on this BM :_: "
ifstat

set idx:int32 0
:loop isset ${net${idx}/mac} || goto shell
  echo "Test net${idx} with MAC ${net${idx}/mac}, can be used...."
  set bootif net${idx}
  ifopen ${bootif}
  set ${bootif}/ip6:ipv6 <CLUSTER_IP>
  set ${bootif}/len6:int8 64
  set ${bootif}/gateway6:ipv6 <CLUSTER_GATEWAY>
  ifstat ${bootif}
  ping --count 5 <CLUSTER_GATEWAY> || echo "Ping failed"
  echo "Wait 30 Sec..."
  sleep 30
  ifstat ${bootif}
  ping --count 5 <CLUSTER_GATEWAY> || echo "Ping failed"
  kernel http://<PASSWORD>:<USERNAME>@<URI>/artifactory/docker-local/cloud/dummy/test_infra/v0.6/repo/test/0.6.3/rocky_810/isolinux/vmlinuz && goto bootkernel || ifclose ${bootif}
  inc idx && goto loop

:shell
  echo "Could not fetch kernel from any device. Can not continue. CHECK NETWORK CONNECTIVITY and try again."
 shell

:bootkernel
  echo "${bootif} Detected & Connected. Booting with MAC ${net${idx}/mac}"
  initrd http://<PASSWORD>:<USERNAME>@<URI>/artifactory/docker-local/cloud/dummy/test_infra/v0.6/repo/test/0.6.3/rocky_810/isolinux/initrd.img
  set initrd initrd.img

  iseq dell dell || goto dell_end
   echo "Proceeding to boot ROCKY LINUX "
  set initrd initrd.magic

:dell_end

  route

  imgstat

  kernel http://<PASSWORD>:<USERNAME>@<URI>/artifactory/docker-local/cloud/dummy/test_infra/v0.6/repo/test/0.6.3/rocky_810/isolinux/vmlinuz ipv6.disable=0 selinux=0 inst.selinux=0 kssendmac iommu=pt amd_iommu=on inst.stage2=<PASSWORD>:<USERNAME>@<URI>/artifactory/docker-local/cloud/dummy/test_infra/v0.6/repo/test/0.6.3/rocky_810 inst.ks=https://<URI>:30047/artifact/api/v1/bmaas/build/rocky_810/l997testms10102.nw.de.local/config/l997testms10102.nw.de.local.ks inst.repo=http://<PASSWORD>:<USERNAME>@<URI>/artifactory/docker-local/cloud/dummy/test_infra/v0.6/repo/test/0.6.3/rocky_810 inst.sshd=1 network initrd=initrd.img rd.net6.timeout.route=30 net.naming-scheme=rhel-8.4 ksdevice=${net${idx}/mac} ip=<CLUSTER_IP>::<CLUSTER_GATEWAY>:64:l997testms10102.nw.de.local::none  console=tty0 console=ttyS1,115200
  initrd http://<PASSWORD>:<USERNAME>@<URI>/artifactory/docker-local/cloud/dummy/test_infra/v0.6/repo/test/0.6.3/rocky_810/isolinux/initrd.img
  boot
`

NiKiZe, 2025-02-10:
In your first case you did `ifopen net0` without any issue.
The question there is if the rest of your configuration would have worked.

For `net2`, have you tried doing this manually?
Does `net2` happen to be used also by the firmware for remote management?
Is `net2` in this case the same nic that was used to boot iPXE in the first place? If yes on the last one, please test if snponly.efi behaves differently in any way.

arun2arunraj, 2025-02-10:
hi @NiKiZe 

For us net0 is ConnectX-DX which works to open without any issues. Where as net2 is 14e4-1751 (Broadcom nic). Ifstat provides the interface name in my screenshots

That's where looking for your support whether we have support for this nic or not. If there is no support what is required to be enabled.

Thanks 
Arun Raj. R

NiKiZe, 2025-02-10:
Is the nic known to be working, yes, in some scenarios.
ifstat does show the native driver.

Is there bugs in some firmware, yes, thats why I asked several questions that you didn't answer.

You included a picture of net0, I assumed that was from a different machine but the same nic.

Main points from
https://github.com/ipxe/ipxe/issues/1402#issuecomment-2646673380
* Does it work if you configure manually, maybe something has a race condition like setup.
* Is there a management interface shared on this NIC, which firmware then claims and prevents setup.
* Is there any difference if you use snponly.efi or snp.efi

arun2arunraj, 2025-02-14:
hi @NiKiZe ,

Can you please help to debug this issue further (manual iPXE prompt itself) ? We have tried to boot the kernel via chain, imgexec and boot commands. All the boot commands are rebooting the node instead of booting from the kernel. The server boot preference and booting orders are configured correct. This is completely new hardware and we assumed only the NIC making the problem. But we couldn't narrow it down as of now. 

Any help would be much appreciated.

``#!ipxe
set bootif net0
ifopen net0
set net0/ip6:ipv6 fbce:abc:1150:3Bf1:b401:2002:1:0
set net0/len6:int8 64
set net0/gateway6:ipv6 fbce:abc:1150:3Bf1::ffff
ifstat net0
route

kernel http://testartifactory.local:@/artifactory/docker-local/cloud/dummy/test_infra/v0.6/repo/test/0.6.3/rocky_810/isolinux/vmlinuz ipv6.disable=0 selinux=0 inst.selinux=0 kssendmac iommu=pt amd_iommu=on inst.stage2=:@/artifactory/docker-local/cloud/dummy/test_infra/v0.6/repo/test/0.6.3/rocky_810 inst.ks=https://testartifactory.local:30047/artifact/api/v1/bmaas/build/rocky_810/l997testms10102.nw.de.local/config/l997testms10102.nw.de.local.ks inst.repo=http://:@/artifactory/docker-local/cloud/dummy/test_infra/v0.6/repo/test/0.6.3/rocky_810 inst.sshd=1 network initrd=initrd.img rd.net6.timeout.route=30 net.naming-scheme=rhel-8.4 ksdevice=${net${idx}/mac} ip=<CLUSTER_IP>::<CLUSTER_GATEWAY>:64:l997testms10102.nw.de.local::none console=tty0 console=ttyS1,115200
initrd http://:@/artifactory/docker-local/cloud/dummy/test_infra/v0.6/repo/test/0.6.3/rocky_810/isolinux/initrd.img
imgstat

boot`

arun2arunraj, 2025-02-16:
hi @NiKiZe ,

Thanks for your help and support.

imgstat shows the SOURCE of TRUTH for the kernel as well as the initrd.

If we boot just the kernel without initrd, Our hardware is stuck and frozen in the boot terminal of IPXE as attached. I guess, this is expected. If we execute to boot for the same along with the initrd, the hardware is rebooting instead of loading from the RAM disk. Is that mean , we are having issue in the initrd ? if yes, how to debug and get this going for next steps. Any help would be much appreciated.

NiKiZe, 2025-02-16:
Verify that your kernel and initrd works, probably by removing all cmdline options.

I would expect you to get full kernel output, including "VFS not found" message from the kernel when booting without initrd.
If booting with initrd, I would expect its scripts to output what is going on.
Detailed help on how to debug your kernel or initrd is out of scope of this issue.

On that point, I have still not gotten an actuall description of what kind of issue you are having with iPXE, everything so far boils down to scripts or other things not working for you, but without giving actuall iPXE issues, and very little in terms of answering questions to try and help further.

arun2arunraj, 2025-02-22:
We are not getting the "VFS not found" message when booting without initrd. understand that this is more on the kernel side. Just trying to see the same works if we use iso based installation. Only we face issue when we do UEFI based installation for the iPXE.

**EFI stub: Loaded initrd from LINUX_EFI_INITRD_MEDIA_GUID device path
EFI stub: Measured initrd data into PCR 9**

NiKiZe, 2025-02-22:
So what **does** `imgstat` show before your `boot`
Could you also do `cat /proc/cmdline` after booting the working kernel, and then again post your iPXE script?

Since there have been no more info on the reported issue, even after asking multiple times for details, I'm closing the issue itself.

closed 2025-02-22 by NiKiZe