ipxe/ipxe #1405: iPXE hangs when enabling USB_CMD for xhci-pch

disjustin, 2025-02-14:
I'm encountering an error where iPXE hangs when initializing xhci-pch after enabling USB_CMD.

using latest iPXE version as of 2/13/25
Steps to Reproduce:

```bash
wget https://github.com/ipxe/ipxe/archive/refs/heads/master.zip
unzip master.zip
cd ipxe-master/src
vi config/general.h
```

enable USB_CMD

```diff
-//#define USB_CMD		/* USB commands */
+//#define USB_CMD		/* USB commands */
```

```bash
make bin/ipxe.lkrn bin-x86_64-efi/ipxe.efi DEBUG=pci
# provide to bootloader filepath
cp -p bin-x86_64-efi/ipxe.efi /var/lib/tftpboot/ipxe.efi
```

Here is the hardware configuration
- [X11SSL-F](https://www.supermicro.com/manuals/motherboard/C236/MNL-1785.pdf)
- [X11DPU](https://www.supermicro.com/manuals/motherboard/C620/MNL-1865.pdf)
- Onboard LAN Option ROM Type: EFI

Error line:
```diff
0000:00:14.0 (8086:a12f) has driver "xhci-pch"
0000:00:14.0 has mem df700000 io 0 irq 255
-0000:00:14.0 device not enabled by BIOS! Updating PCI command 0006->0007
0000:00:14.0 latency timer is unreasonably low at 0. Setting to 32.
```

This line does not appear after disabling USB_CMD

NiKiZe, 2025-02-14:
Messages about potential issues are NOT error by themselves.
Are you saying that after this output iPXE hangs?

This is ipxe.efi that hangs?

disjustin, 2025-02-14:
@NiKiZe yes, ipxe hangs indefinitely after this line is printed as we waited ~1hr for multiple systems to load at this stage. Normal operation should take a few seconds to initialize devices
[image]