ipxe/ipxe #1407: HTTP no usable certificate with ipxe.efi whereas undionly.kpxe is fine

Poulpatine, 2025-02-14:
Hi,

I'm trying to boot over HTTP with ipxe.
I've built sources for undionly.kpxe and ipxe.efi.

With undionly.kpxe (so in BIOS mode) I can fetch my kernel and initrd from https://pxe.factory.talos.dev but with ipxe.efi I can not with an error :

`X509 chain 0xc5d6cca8 found no usable certificates`.

Both binaries were built using the same `config/general.h` and build options.

A simple test to reproduce the problem is to use the iPXE prompt and simply type:
```
imgfetch https://pxe.factory.talos.dev
```
Thanks

Poulpatine, 2025-02-14:
Both (BIOS and UEFI) tests have been made on the same hardware and UEFI problem has later been reproduced on another hardware.

I've tried to show unixtime on both BIOS and UEFI and it's correct.
I've compiled sources from master branch.

Here is the console output (OCR extracted from a picture and manually corrected so a few mistakes may remain).

```
iPXE> imgfetch https://pxe.factory.talos.dev/

TLS 0xc5d70588 created session pxe.factory.talos.dev

https://pxe.factory.talos.dev..TS 0xc5d70588 using protocol version 3.3

TLS 0xc5d70588 selected pubkey-rsa-aes_cbc-128-sha1

TS 0xc5d70588 new session ID:
[...]
X509 chain 0xc5d6c348 added X509 0xc5d6f808 "pxe.factory.talos.dev"

TLS 0xc5d70588 found certificate pxe.factory.talos.dev

X509 chain 0xc5d6c348 added ¥509 Oxc5d76008 "ZeroSsL RSA Domain Secure Site CA"

TLS 0xc5d70588 found certificate ZeroSsL RSA Domain Secure Site CA

X509 chain 0xc5d6c348 added X509 Oxc5d71808 "USERTrust RSA Certification Authority"

TLS 0xc5d70583 found certificate USERTrust RSA Certification Authority

X509 chain 0xc5d6c348 found no usable certificates

X509 chain 0xc5d6c348 found no usable certificates

X509 chain 0xc5d6c348 found no usable certificates

. [XCRT pxe.factory.talos.dev]X509 chain 0xc5d6c348 found no usable certificates

TLS 0xc5d70588 certififcate validation failed: Permission denied (https://ipxe.org/0216eb8f)

Permission denied (https://ipxe.org/0216eb8f)
```

Poulpatine, 2025-02-15:
I've been able to send logs over syslog.

Here are the UEFI logs:
```
ipxe: iPXE> imgfetch https://pxe.factory.talos.dev
ipxe: TLS 0xc5d71008 created session pxe.factory.talos.dev
ipxe: https://pxe.factory.talos.dev...Sending plaintext data:
ipxe: c5d6f948 : 01 00 00 b5 03 03 67 7d-b0 67 e1 58 1c 45 5d d7 : ......g}.g.X.E].
ipxe: c5d6f958 : a4 b1 d7 6a ae e4 c4 3f-ba 79 b3 e9 65 5b e7 18 : ...j...?.y..e[..
ipxe: c5d6f968 : 6b f2 a0 7a 08 4c 20 67-7d b0 67 e1 58 1c 45 5d : k..z.L g}.g.X.E]
ipxe: c5d6f978 : d7 a4 b1 d7 6a ae e4 c4-3f ba 79 b3 e9 65 5b e7 : ....j...?.y..e[.
ipxe: c5d6f988 : 18 6b f2 a0 7a 08 4c 00-24 c0 2f c0 30 c0 27 c0 : .k..z.L.$./.0.'.
ipxe: c5d6f998 : 28 c0 13 c0 14 00 9e 00-9f 00 67 00 6b 00 33 00 : (.........g.k.3.
ipxe: c5d6f9a8 : 39 00 9c 00 9d 00 3c 00-3d 00 2f 00 35 01 00 00 : 9.....<.=./.5...
ipxe: c5d6f9b8 : 48 00 00 00 1a 00 18 00-00 15 70 78 65 2e 66 61 : H.........pxe.fa
ipxe: c5d6f9c8 : 63 74 6f 72 79 2e 74 61-6c 6f 73 2e 64 65 76 00 : ctory.talos.dev.
ipxe: c5d6f9d8 : 01 00 01 04 00 0d 00 0c-00 0a 02 01 03 01 04 01 : ................
ipxe: c5d6f9e8 : 05 01 06 01 ff 01 00 01-00 00 23 00 00 00 0a 00 : ..........#.....
ipxe: c5d6f9f8 : 08 00 06 00 17 00 18 00-1d                      : .........       
ipxe: Received plaintext data:
ipxe: c5d6f980 : 02 00 00 56 03 03 80 73-f1 82 1a 0b 8e 77 b2 3d : ...V...s.....w.=
ipxe: c5d6f990 : c1 bf b2 bd de fb 15 e1-84 3a df 17 ad 0f 44 4f : .........:....DO
ipxe: c5d6f9a0 : 57 4e 47 52 44 01 20 41-a1 65 44 3e 4a 5c 53 b3 : WNGRD. A.eD>J\S.
ipxe: c5d6f9b0 : 6b c6 93 d1 61 fc c9 88-d0 ae e9 8a 11 80 cc 72 : k...a..........r
ipxe: c5d6f9c0 : 22 62 28 cb a6 19 5d 00-2f 00 00 0e ff 01 00 01 : "b(...]./.......
ipxe: c5d6f9d0 : 00 00 00 00 00 00 01 00-01 04                   : ..........      
ipxe: TLS 0xc5d71008 using protocol version 3.3
ipxe: TLS 0xc5d71008 selected pubkey-rsa-aes_cbc-128-sha1
ipxe: TLS 0xc5d71008 new session ID:
ipxe: 00000000 : 41 a1 65 44 3e 4a 5c 53-b3 6b c6 93 d1 61 fc c9 : A.eD>J\S.k...a..
ipxe: 00000010 : 88 d0 ae e9 8a 11 80 cc-72 22 62 28 cb a6 19 5d : ........r"b(...]
ipxe: Received plaintext data:
ipxe: c5d79300 : 0b 00 12 e7 00 12 e4 00-06 7d 30 82 06 79 30 82 : .........}0..y0.
ipxe: c5d79310 : 04 61 a0 03 02 01 02 02-11 00 a8 d6 39 3a 31 91 : .a..........9:1.
ipxe: X509 0xc5d73008 "pxe.factory.talos.dev" is valid (at time 1739619696)
ipxe: X509 0xc5d73008 "pxe.factory.talos.dev" is not a root certificate
ipxe: X509 0xc5d73008 "pxe.factory.talos.dev" has no trusted issuer
ipxe: X509 0xc5d76e48 "ZeroSSL RSA Domain Secure Site CA" is valid (at time 1739619696)
ipxe: X509 0xc5d76e48 "ZeroSSL RSA Domain Secure Site CA" is not a root certificate
ipxe: X509 0xc5d76e48 "ZeroSSL RSA Domain Secure Site CA" has no trusted issuer
ipxe: X509 0xc5d74008 "USERTrust RSA Certification Authority" is valid (at time 1739619696)
ipxe: X509 0xc5d74008 "USERTrust RSA Certification Authority" is not a root certificate
ipxe: X509 0xc5d74008 "USERTrust RSA Certification Authority" has no trusted issuer
ipxe: X509 chain 0xc5d6cc68 found no usable certificates
ipxe: X509 0xc5d73008 "pxe.factory.talos.dev" is valid (at time 1739619697)
ipxe: X509 0xc5d73008 "pxe.factory.talos.dev" is not a root certificate
ipxe: X509 0xc5d73008 "pxe.factory.talos.dev" has no trusted issuer
ipxe: X509 0xc5d76e48 "ZeroSSL RSA Domain Secure Site CA" is valid (at time 1739619697)
ipxe: X509 0xc5d76e48 "ZeroSSL RSA Domain Secure Site CA" is not a root certificate
ipxe: X509 0xc5d76e48 "ZeroSSL RSA Domain Secure Site CA" has no trusted issuer
ipxe: X509 0xc5d74008 "USERTrust RSA Certification Authority" is valid (at time 1739619697)
ipxe: X509 0xc5d74008 "USERTrust RSA Certification Authority" is not a root certificate
ipxe: X509 0xc5d74008 "USERTrust RSA Certification Authority" has no trusted issuer
ipxe: X509 chain 0xc5d6cc68 found no usable certificates
ipxe: X509 0xc5d73008 "pxe.factory.talos.dev" is valid (at time 1739619698)
ipxe: X509 0xc5d73008 "pxe.factory.talos.dev" is not a root certificate
ipxe: X509 0xc5d73008 "pxe.factory.talos.dev" has no trusted issuer
ipxe: X509 0xc5d76e48 "ZeroSSL RSA Domain Secure Site CA" is valid (at time 1739619698)
ipxe: X509 0xc5d76e48 "ZeroSSL RSA Domain Secure Site CA" is not a root certificate
ipxe: X509 0xc5d76e48 "ZeroSSL RSA Domain Secure Site CA" has no trusted issuer
ipxe: X509 0xc5d74008 "USERTrust RSA Certification Authority" is valid (at time 1739619698)
ipxe: X509 0xc5d74008 "USERTrust RSA Certification Authority" is not a root certificate
ipxe: X509 0xc5d74008 "USERTrust RSA Certification Authority" has no trusted issuer
ipxe: X509 chain 0xc5d6cc68 found no usable certificates
ipxe: . [XCRT pxe.factory.talos.dev]X509 0xc5d73008 "pxe.factory.talos.dev" is valid (at time 1739619699)
ipxe: X509 0xc5d73008 "pxe.factory.talos.dev" is not a root certificate
ipxe: X509 0xc5d73008 "pxe.factory.talos.dev" has no trusted issuer
ipxe: X509 0xc5d76e48 "ZeroSSL RSA Domain Secure Site CA" is valid (at time 1739619699)
ipxe: X509 0xc5d76e48 "ZeroSSL RSA Domain Secure Site CA" is not a root certificate
ipxe: X509 0xc5d76e48 "ZeroSSL RSA Domain Secure Site CA" has no trusted issuer
[... 1831 lines cut ...]
ipxe: 0003aef0 : 73 3d 22 68 74 74 70 3a-2f 2f 77 77 77 2e 77 33 : s="http://www.w3
ipxe: 0003af00 : 2e 6f 72 67 2f 32 30 30-30 2f 73 76 67 22 20 66 : .org/2000/svg" f
ipxe: 0003af10 : 69 6c 6c 3d 22 6e 6f 6e-65 22 20 76 69 65 77 42 : ill="none" viewB
[... cut ...]
ipxe: 0003afe0 : 3e 0a 0a 0a 20 20 20 20-20 20 20 20 20 20 20 20 : >...            
ipxe: 0003aff0 : 3c 2f 66 6f 72 6d 3e 0a-20 20 20 20 20 20 20 20 : </form>.        
ipxe: 0003b000 : 3c 2f 64 69 76 3e 0a 20-20 20 20 3c 2f 6d 61 69 : </div>.    </mai
ipxe: 0003b010 : 6e 3e 0a 3c 2f 62 6f 64-79 3e 0a 0a 3c 2f 68 74 : n>.</body>..</ht
ipxe: 0003b020 : 6d 6c 3e 0a 0d 0a                               : ml>...          
ipxe: Received plaintext data:
ipxe: 00036c50 : 30 0d 0a 0d 0a                                  : 0....           
ipxe: Downloaded "<NULL>"
ipxe:  ok
```

Poulpatine, 2025-02-18:
I've been able to figure out what the problem was.
IPv6 was not correctly configured in my network (2 routers were sending advertisements but the global setup was wrong).

iPXE resolved an AAAA record for ca.ipxe.org but was not able to reach the ip afterward.
By disabling router advertisements in my network everything has been fixed.

I've been able to understand that by using the `certstore` command with `certstore http://ca.ipxe.org/ca.crt` and various debug options.

Many thanks for your help.
From my point of view the problem is fixed (but maybe a debug message could be added to help to pinpoint the issue ?).

closed 2025-02-18 by Poulpatine

mcb30, 2025-02-18:
Glad to know you got it fixed! I think the debug output from `DEBUG=validator` would probably have been sufficient.

Poulpatine, 2025-02-18:
I can make a test in the coming days if you want it ?

mcb30, 2025-02-18:
> I can make a test in the coming days if you want it ?

Entirely up to you. If the debug output from `DEBUG=validator` does not give enough information to indicate why the download is failing then yes, it would be useful to add.