ipxe/ipxe #1408: Autoexec.ipxe download fails with https URL on UEFI hardware

mtniehaus, 2025-02-17:
When configuring an HTTPS URL for downloading snponly_x64.efi, the autoexec.ipxe file should be fetched from the same URL.  But this fails with error code 1c25e082:

[image]

In this case, I was using a Dell Optiplex Micro 7010 PC.  When configuring the URL in the UEFI firmware, you have to specify a root cert to be used.  Since this is a LetsEncrypt cert, I specified the https://letsencrypt.org/certs/isrgrootx1.pem cert.  This was sufficient to get the firmware to download the .EFI file successfully.  Perhaps the autoexec.ipxe file isn't using the same cert (or isn't supporting TLS at all)?

Changing the URL to http: works fine (.efi file and autoexec.ipxe file are downloaded fine from the same server, just without using TLS).

NiKiZe, 2025-02-17:
Following the link https://ipxe.org/1c25e082 essentially tells us that the random generator has not been initialized yet.

mtniehaus, 2025-02-17:
Since it wasn't an exact match on the error code, it wasn't obvious if that was applicable, but it's not obvious how those codes work exactly.  Regardless of the root cause though, it doesn't work :-)

NiKiZe, 2025-02-17:
Isn't https://ipxe.org/1c25e082 exactly the url in your message?
There is then some redirects to collect several minor codes under one major.

Yes, I do find it possible that this is a scenario that could be improved.
Is there any chance that this could be reproduced in qemu?

mtniehaus, 2025-02-17:
I have reproduced it in Proxmox, which uses QEMU with a relatively recent EDK2 UEFI firmware.  However, it's not particularly simple to set up because the firmware doesn't let you create the boot entry; you would have to boot into Windows or Linux in order to create a new HTTPS boot entry (I used my UEFIv2 PowerShell module), and then boot back into the firmware to add a root CA to the firmware (using the isrgrootx1.pem file mentioned previously).

In the EDK2/QEMU/Proxmox case, the root cert isn't associated with the boot entry, it's added as a separate TLS entry.

The boot entry I created:

[image]

The cert enrollment:

[image]

The boot menu:

[image]

The error:

[image]

mcb30, 2025-02-18:
This is an awkward artefact of the way that the ANS X9.82 abstraction defines interaction with the RBG.  In this case, the attempt to draw random bits from the RBG is happening before `RBG_Startup` has been called, and so fails because the DRBG is not yet valid.

iPXE attempts to follow the text from ANS X9.82 / NIST SP 800-90 extremely literally, to make it easy to review for compliance. See, for example, the implementation of `Generate_function` in `drbg_generate()`, where the specification's text "Go step 7." is implemented in iPXE with a literal `goto step_7`:
https://github.com/ipxe/ipxe/blob/b35300fc67cc050ce3f709a88a4437153d85e0ee/src/crypto/drbg.c#L399-L400

The specification implicitly assumes that the `RBG_Startup` function will be called before it is needed, and includes checks to make sure that `Generate_function` fails if this has not happened: this is why you are seeing the error.  However, there is no well-defined point at which the `RBG_Startup` function is to be called: it's just assumed that this happens as part of system startup.

We currently call `RBG_Startup` to instantiate the DRBG as an iPXE startup function, with the corresponding shutdown function uninstantiating the DRBG.  This works for most use cases, and avoids an otherwise unexpected user-visible delay when a caller first attempts to use the DRBG (e.g. by attempting a first HTTPS download).  But it clearly doesn't correctly cover this use case.

The download of `autoexec.ipxe` for UEFI is triggered by the EFI root bus probe in `efi_probe()`, since this is the last opportunity for us to create a temporary MNP device for the download, before we potentially unload or disconnect the drivers required to provide that device.  Both the root bus probe and the RBG startup run at `STARTUP_NORMAL`, so there is no defined ordering between them.

One solution would be to add another startup ordering, so that the RBG and driver probe startup functions have a defined relative order.  But I don't like this because it adds complexity.

I'm tempted to instead replace the logic in `rbg_generate()` to call `rbg_startup()` if needed, with `rbg_startup()` modified to allow for the fact that it may be called multiple times.  I can't see any real downside of allowing the DRBG to be instantiated early if needed.

mcb30, 2025-02-18:
@mtniehaus Should now be fixed via #1411.  I didn't test your precise scenario due to the tediousness of configuring UEFI HTTPS boot, but I tested with a hardcoded call to `rbg_generate()` from within `efi_autoexec_network()` and it did successfully fix the return code from this call.

mtniehaus, 2025-02-18:
Working great, thanks.

PR #1411 (2025-02-18): [crypto] Start up RBG on demand if needed
The ANS X9.82 specification implicitly assumes that the RBG_Startup function will be called before it is needed, and includes checks to make sure that Generate_function fails if this has not happened. However, there is no well-defined point at which the RBG_Startup function is to be called: it's just assumed that this happens as part of system startup.

We currently call RBG_Startup to instantiate the DRBG as an iPXE startup function, with the corresponding shutdown function uninstantiating the DRBG.  This works for most use cases, and avoids an otherw [... cut]