ipxe/ipxe #1412: strict choose timeout : choose regardless user input at the end of timeout

eoli3n, 2025-02-19:
I manage 900 dualboot hosts, which boot default pxe, i replaced grub with ipxe to be able to trigger os deployments from the server.
Here the default menu

```
#!ipxe

set menu-timeout 60000
set submenu-timeout ${menu-timeout}
isset ${menu-default} || set menu-default debian
cpair --foreground 0 --background 7 2
#console --x 640 --y 480
console --picture http://fsi-core.univ-tlse3.fr/ipxe/ipxe.png

:start 
menu iPXE Boot menu
item debian            	Debian
item windows          	Windows

choose --timeout ${menu-timeout} --default ${menu-default} selected || goto cancel
set menu-timeout 0
goto ${selected}

:debian
echo Booting Debian
console && sanboot --drive 0 --filename \EFI\debian\grubx64.efi
:windows
echo Booting Windows
console && sanboot --drive 0 --filename \EFI\Microsoft\Boot\bootmgfw.efi
```

Each morning i have a script that scan 900 hosts states, it tests if host pings, and if it does, then it test if port 22 or 3389 is open to get os running in a quick way.
If host ping but has no 22/3389 ports open, it fallbacks in "unknown state".
All hosts stops each night at 22pm and automatically boot at 6am, scan script runs at 7am.
Each morning i get ~5 hosts stucked in ipxe menu.
In that state, the host needs a physical fix.

Strict choose timeout would be a way to boot selected entry regardless a key has been pressed, timeout doesn't stop.

NiKiZe, 2025-02-19:
Let's say that something has fallen on keyboard.
In that case maybe a timeout counter pause could be a better option, but harder to explain to users how that works.

mcb30, 2025-02-19:
`--retimeout` is the best word I have been able to come up with.  Implemented in #1413 

Any opinions before I merge?

mcb30, 2025-02-19:
I've updated https://ipxe.org/cmd/choose to include the `--retimeout` option - the updated wording is reasonably clean, which is usually a good sign.

eoli3n, 2025-02-24:
This is buggy, with ``--retimeout 10``, pressing any key instant boot the selected entry.

mcb30, 2025-02-24:
> This is buggy, with `--retimeout 10`, pressing any key instant boot the selected entry.

Timeouts are expressed in milliseconds as per the documentation: a 10ms timeout is basically instantaneous.

PR #1413 (2025-02-19): [menu] Allow a post-activity timeout to be defined
Allow the "--retimeout" option to be used to specify a timeout value that will be (re)applied after each keypress activity.  This allows script authors to ensure that a single (potentially accidental) keypress will not pause the boot process indefinitely.

Fixes: #1412