ipxe/ipxe #1422: Cannot Boot, Bad CPIO Magic Error

camhendricks, 2025-03-07:
Using wimboot v2.8.0 and the newest version of IPXE I cannot boot. The error is as follows

> Bad CPIO Magic
FATAL: could not extract initrd files

I have been using the same ipxe boot script for a year now which has always worked. I build IPXE from source in my CI/CD pipeline. This just started happening recently and I have removed all variables that I can but still am having this issue. Only on Legacy Boot systems, UEFI Systems are booting properly.

liuzhaoyzz, 2025-03-07:
Oh,I have the same problem.
BIOS+IPXE1.21.1+ipxe.pxe+wimboot2.8.0
WiMboot v2.8.0 -- WindoWs IMaging ForMat bootloader -- https://ipxe.org/Wimboot
Found initrd at [Oxb41d6000,Oxbfd636c3)
Placing_initrd at [0x74472000,0x7ffff6c3)
Bad CPIO Magic
FATAL: could not extract initrd files
Press a key to reboot...
(the screenpicture upload failed)

iPXE -  [err：0f0a60] --- iPXE - open source boot firmware [err:0f0a60]
https://ipxe.org/err/0f0a60

http://boot.ipxe.org/
ipxe.pxe | 2025-03-07 01:00 | 337K |  

autoexec.ipxe
:[BIOS]/imgs/210219WinPE19043-844boot.wim
kernel http://${next-server}/pxe/wimboot
initrd http://${next-server}/pxe/bootmgr bootmgr
initrd http://${next-server}/pxe/BCD bcd
initrd http://${next-server}/boot.sdi boot.sdi
initrd http://${next-server}/imgs/210219WinPE19043-844boot.wim boot.wim
boot

BIOS+IPXE1.0.0+ipxe.pxe+wimboot2.8.0,boot successfully!

NiKiZe, 2025-03-07:
Show output of `imgstat` before `boot` in your scripts.

And a suggestion, newer wimboot will extract needed files from the .wim so usually you can drop boot.sdi, BCD and the others.

omnix32, 2025-03-07:
Same issue here, tested with wimboot 2.6.0 and 2.8.0 same error. It's working with 1.21.1+ (g90625), but getting this error under (gbe3a). imgstat and prompt don't appear to add any info. Droppping boot.sdi, BCD doesn't appear to help either.

NiKiZe, 2025-03-07:
If you have a known bad commit, and a known good, do follow https://ipxe.org/howto/bisect

`imgstat` will output all loaded files in iPXE it should be before the `boot` line that starts wimboot, if it's there but not showing then I would expect that your script isn't executing.

omnix32, 2025-03-07:
Yes, sorry,I had put the imgstat on the wrong line, it now shows the files loaded: BCD, boot.sdi, boot.wim (or just boot.wim with the other two disabled), so the error appears to happen after that? Botting with the known working one shows the same thing.

I'll see if I can find some time to bisect (I'm not great with git).

camhendricks, 2025-03-07:
Without doing a git bisect I would guess the problem is in 12ea8c4, since heavy change were made to CPIO headers which is the same error I am getting. Later today I can test out the commit before that and let you know what I find.

camhendricks, 2025-03-07:
Can now confirm that e7595fe is working, but not 12ea8c4. Thanks to all those that support this tool its a huge help to me and those who work with me!

illsk1lls, 2025-03-07:
Here is a screenshot after adding imgstat to boot.ipxe

[image]

liuzhaoyzz, 2025-03-07:
> If you have a known bad commit, and a known good, do follow https://ipxe.org/howto/bisect如果你有一个已知的坏提交和一个已知的好提交，请遵循 https://ipxe.org/howto/bisect
> 
> `imgstat` will output all loaded files in iPXE it should be before the `boot` line that starts wimboot, if it's there but not showing then I would expect that your script isn't executing.`imgstat` 将在 iPXE 中输出所有加载的文件，它应该在启动 wimboot 的`引导`行之前，如果它在那里但没有显示，那么我希望您的脚本没有执行。

......
imgstat
sleep 8
boot
Here is the screenshot:

[image]

omnix32, 2025-03-11:
> Can now confirm that [e7595fe](https://github.com/ipxe/ipxe/commit/e7595fe88d8c32354cd61055cbc9f2e7b3f91ff8) is working, but not [...]

Thank you! worked out how to run a git checkout and confirmed [e7595fe](https://github.com/ipxe/ipxe/commit/e7595fe88d8c32354cd61055cbc9f2e7b3f91ff8) works for me too.

mcb30, 2025-03-12:
Should now be fixed in https://github.com/ipxe/ipxe/commit/da3024d25

liuzhaoyzz, 2025-03-13:
http://boot.ipxe.org/
The last release is 1.21.1+ (g5c49e),still BIOS+IPXE 1.21.1+ (g5c49e)+ipxe.pxe+wimboot2.8.0 boot failed!
Where can I download the release you compiled?

NiKiZe, 2025-03-13:
> http://boot.ipxe.org/ The last release is 1.21.1+ (g5c49e),still BIOS+IPXE 1.21.1+ (g5c49e)+ipxe.pxe+wimboot2.8.0 boot failed! Where can I download the release you compiled?

Latest master is at https://github.com/ipxe/ipxe/commit/da3024d257d73d2c40e65c6fb9db5a2a6bc86dbb

Pre compiled binaries are available at https://boot.ipxe.org but these does indeed look to be out of date @mcb30