ipxe/ipxe #1423: Using Intel nic cards on AMD platform, there will have legacy interrupt issue.

wolfqiancheng, 2025-03-07:
We found that in many cases AMD platforms have legacy interrupt issue when using Intel Nic cards including (i350, E810)， and it always causes problems with receiving/sending packets. 
We tried adding these card's PCI vender:device to the table of known-broken devices in undinet_irq_broken_list[] in undinet.c, if that this issue can be solved.

Can we merge these lines of code into the master branch of the ipxe source? Or do you have any other suggestions?

static const struct undinet_irq_broken undinet_irq_broken_list[] = {
	/* HP XX70x laptops */
	{ 0x8086, 0x1502, PCI_ANY_ID, PCI_ANY_ID },
	{ 0x8086, 0x1503, PCI_ANY_ID, PCI_ANY_ID },
	/* HP 745 G3 laptop */
	{ 0x14e4, 0x1687, PCI_ANY_ID, PCI_ANY_ID },
	/* Lenovo 810 laptop */
        { 0x8086, 0x159B, PCI_ANY_ID, PCI_ANY_ID }, /* <- add this line */
	{ 0x8086, 0x1592, PCI_ANY_ID, PCI_ANY_ID }, /* <- add this line */
	{ 0x8086, 0x1521, PCI_ANY_ID, PCI_ANY_ID }, /* <- add this line */
	{ 0x8086, 0x1522, PCI_ANY_ID, PCI_ANY_ID }, /* <- add this line */
};

NiKiZe, 2025-03-07:
What are you booting? snponly.efi?

wolfqiancheng, 2025-03-07:
I build a chainloadable iPXE image using:
  make bin/undionly.kpxe

wolfqiancheng, 2025-03-07:
In fact, these cards listed below have encountered the same issue. It doesn't seem that a specific device ID will appear. maybe we haven't seen it yet with other device .
E810-XXV:     vender id 0x8086  ; device id  0x159b;  sub vender id   0x8086;   sub device id 0x0003
E810-XXV:     vender id 0x8086  ; device id  0x159b;  sub vender id   0x8086;   sub device id 0x4002
E810-C:         vender id 0x8086  ; device id  0x1592;  sub vender id   0x8086;   sub device id 0x0002
i350:              vender id 0x8086  ; device id  0x1521;  sub vender id   0x1dcf;    sub device id 0x0308

NiKiZe, 2025-03-07:
All these are Lenovo machines? I do like to use their 4 letter type numbers to identify devices without confusion.

wolfqiancheng, 2025-03-07:
Not all Lenovo machines, but also IEISystem machines.

It seems that you are worried about inspecting other devices.  Of course your consideration is correct， i think.   But could you just use vender/device id instead of 4part id?  At least they can boot successfully in both modes. Because from what I see, it's not a issue found by a specific user. If other devices have problems next time and have to be added to the table, there may be a lot of troubles later. 

I'm also looking for the intel team to fix the nic issue, but it seems to be a system compatibility issue, and the pxe code hasn't been updated for decades. I'm worried about the difficulty. But I'm trying hard.

thx~

NiKiZe, 2025-03-07:
The pciid identifies a PCI device, in this case a NIC.
That could perform much worse on currently perfectly working machines.

I'm trying to understand exactly which machines are having issues, any change would have to point to exact models that has issues, just as the existing code does. And the initial message has "Lenovo 810 laptop" is that the actual laptop product name? Which model number does it have?

The update does seem to indicate NIC models, rather than actual computers that is having issues.

You might also want to use ipxe.pxe instead if undionly.kpxe is causing you pain.

mcb30, 2025-03-07:
> We found that in many cases AMD platforms have legacy interrupt issue when using Intel Nic cards including (i350, E810)， and it always causes problems with receiving/sending packets.

Thank you.  This is an interesting report.  We've previously experienced issues with legacy interrupts on AMD systems, but it has only so far been reproducible in Azure VMs (see commit https://github.com/ipxe/ipxe/commit/f3ba0fb5f).

Are you saying that you are experiencing an issue with legacy interrupts when running on a bare metal AMD system (no hypervisor)?

wolfqiancheng, 2025-03-07:
> > We found that in many cases AMD platforms have legacy interrupt issue when using Intel Nic cards including (i350, E810)， and it always causes problems with receiving/sending packets.
> 
> Thank you. This is an interesting report. We've previously experienced issues with legacy interrupts on AMD systems, but it has only so far been reproducible in Azure VMs (see commit [f3ba0fb5f](https://github.com/ipxe/ipxe/commit/f3ba0fb5f)).
> 
> Are you saying that you are experiencing an issue with legacy interrupts when running on a bare metal AMD system (no hypervisor)?

wolfqiancheng, 2025-03-11:
> The pciid identifies a PCI device, in this case a NIC. That could perform much worse on currently perfectly working machines.
>
> I'm trying to understand exactly which machines are having issues, any [...]

OK, I understand your concern.

SR665 Milan server, this should be the exact machine.  Regarding the message, “Lenovo 810 laptop”, i means this machine produced by Lenovo. And we reproduced this issue in the lab at lenovo.

What I said before is that in the list of these nics + AMD Milan, there will be an interruption issue.  it seems that the "intel undi driver" or ipxe driver has a compatibility issue with the AMD platforms. 

I have tried ipxe.pxe and undionly.kpx, and both of them have this problem.

mcb30, 2025-03-11:
> SR665 Milan server, this should be the exact machine. Regarding the message, “Lenovo 810 laptop”, i means this machine produced by Lenovo. And we reproduced this issue at the lenovo lab.

Thank you.  Does this mean that you work for Lenovo, or have a commercial relationship with them?

wolfqiancheng, 2025-03-13:
thanks Michael.   I work for intel, and of course have a commercial relationship with Lenovo. 

However, I still wonder if we have a chance to put these cards in undinet_irq_broken_list[]？  Or do you have any better suggestions for this issue?

mcb30, 2025-03-14:
> thanks Michael. I work for intel, and of course have a commercial relationship with Lenovo.
> 
> However, I still wonder if we have a chance to put these cards in undinet_irq_broken_list[]？ Or do you have any better suggestions for this issue?

Putting those cards in `undinet_irq_broken_list` is not a viable solution, if this is a generic problem that is affecting large numbers of AMD systems and only AMD systems.  In general, any PCI device may be placed into either an Intel or AMD system, and so using the PCI IDs is not a sensible approach.

While I appreciate that some PCI IDs are likely to appear only in AMD-based OEM machines (particularly with laptop models, where hardware is integrated), this is still not a sustainable approach, since the list could grow without limit.

Please contact me via the `vendor-support@ipxe.org` email to arrange shipment of a suitable laptop on which I can reproduce the issue, and we'll take it from there.

wolfqiancheng, 2025-03-18:
Thank you for your reply. It is obviously not easy for me to shipment a server to you at present. But I am still working hard to find our backend team to solve this issue.  If there is any progress, I will sync it to you. thanks again!

closed 2025-03-18 by wolfqiancheng

mcb30, 2025-03-18:
> Thank you for your reply. It is obviously not easy for me to shipment a server to you at present. But I am still working hard to find our backend team to solve this issue. If there is any progress, I will sync it to you. thanks again!

Thank you.  Your initial patch suggests that the affected machines are laptops:
> / Lenovo 810 laptop /
> { 0x8086, 0x159B, PCI_ANY_ID, PCI_ANY_ID }, / <- add this line /
> { 0x8086, 0x1592, PCI_ANY_ID, PCI_ANY_ID }, / <- add this line /
> { 0x8086, 0x1521, PCI_ANY_ID, PCI_ANY_ID }, / <- add this line /
> { 0x8086, 0x1522, PCI_ANY_ID, PCI_ANY_ID }, / <- add this line */

in which case they should be fairly easy to transport.

wolfqiancheng, 2025-03-21:
thanks! 
Since our Lab currently does not have this models and has no reproduced now, all reproduction are based on Lenovo customer's machines.  It is for this reason, it seems hard for me to get the machine to ship to you.