ipxe/ipxe #1424: dhcpv6 problem?

sprootik, 2025-03-07:
Hi!
I have a strange problem with ipv6, but some commits ago everything worked correctly. Unfortunately, I can't say the exact moment when the problem occurred.

I use latest version IPXE in efi mode and slaac + dhcp statles server.
On a virtual machine (qemu) ipv6 the address is is issued correctly. But on any hardware there is no attempt to even obtain an ipv6 address. 

For example:
I use server with network card Intel E810-XXV. The built-in pxe bootloader receives ipv6 address as well as the dhcp option to download (ipxe.efi) the be correctly. Next, download iPXE and try to get an ipv6 address and it doesn't work. I don't see even a hint of requests (solicit) from iPXE.
In this example the bootloader is compiled with dhcpv6 debugs.
[image]
it looks like something is going wrong at the socket opening stage.
I will also clarify that I checked the work on various hardware, this problem is everywhere. There is no problem only with virtual machines (qemu).

sprootik, 2025-03-07:
The same in qemu. It's work correctly.

[image]

NiKiZe, 2025-03-07:
If you can find a working commit and a non working commit then use git bisect.

https://ipxe.org/howto/bisect

Do you see the same issue if using snponly.efi?

sprootik, 2025-03-07:
> If you can find a working commit and a non working commit then use git bisect.
> 
> https://ipxe.org/howto/bisect
> 
> Do you see the same issue if using snponly.efi?

In legacy mode the same. I'll try to find bad commit and return late.

NiKiZe, 2025-03-07:
You have to be more specific.

In legacy pcbios mode you could use ipxe.pxe, or undionly.kpxe and result could change.
In efi you have ipxe.efi, which you seem to be using, but also snponly.efi

Please test snponly.efi this would help narrow down if the native iPXE Intel driver is related to the issue.

sprootik, 2025-03-07:
> You have to be more specific.
>
> In legacy pcbios mode you could use ipxe.pxe, or undionly.kpxe and result could change. In efi you have ipxe.efi, which you seem to be using, but also snponly.efi [...]

I used undionly.kpxe in legacy. Tried using snponly.efi in uefi, everything works fine!!! Could you tell me or give a link to what the differences are between snponly.efi and ipxe.efi? 
P.S. About ipxe.efi unfortunately, I couldn't find a working commit, although I looked at commits even from 2023. Probably the working version was built before 2023. But alas, I can't build versions with support i386.

NiKiZe, 2025-03-07:
Ok, so undionly, and snponly that doesn't have native iPXE driver for the NIC works, but the native iPXE driver for E810-XXV does seem to have issues with getting ipv6, but maybe also more. This might be similar to other reports.

Have you tried to check for firmware updates for the nic or bios, just in case that does make a difference?

NiKiZe, 2026-01-11:
Closing as stale, happy to reopen if any new info comes in.
This seems to be an issue with E810-XXV and ice driver only, but would be nice to also have pciid

closed 2026-01-11 by NiKiZe