ipxe/ipxe #1428: timeout option ignored

h908714124, 2025-03-13:
I have a network/DHCP problem, but that is not the issue here.

The following command runs for approximately 15 seconds, then prints an error message:

```
iPXE> ifconf --configurator dhcp
Configuring [dhcp] (net0 46:4e:a4:9b:4d:51).................. Connection timed out (https://ipxe.org/4c106092)
```

I tried increasing the timeout to two minutes, but **the** `--timeout`  **option seems to be ignored**. The command still prints an error after only 15 seconds, appoximately (note there should be more dots to indicate longer wait):

```
iPXE> ifconf --configurator dhcp --timeout 120000
Configuring [dhcp] (net0 46:4e:a4:9b:4d:51).................. Connection timed out (https://ipxe.org/4c106092)
```

mcb30, 2025-03-13:
The explicit timeout that you can specify with the `--timeout` option is applied to the operation as a whole, and is intended to allow scripts to prevent a rogue configurator from getting stuck. The net effect is that you can shorten the overall timeout, but you can't change the timing or retry behaviour of the underlying configurator.

In this case, the DHCP configurator is timing out after its usual discovery period. At that point, since there are no other configurators running, the whole operation ends (with the timeout error propagated from DHCP).

NiKiZe, 2025-03-13:
By following the url in the message https://ipxe.org/4c106092 you get quite detailed information on what to check and investigate.

closed 2025-06-11 by zube[bot]