ipxe/ipxe #1430: sanboot is trying to access a bootloader that does not exist

AlexanderZhirov, 2025-03-16:
I installed Debian 12 over the network onto an iSCSI device. After the reinstallation, I am trying to boot from it, but iPXE in debug mode shows that it is attempting to boot from `<DEVICE>/EFI/BOOT/BOOTX64.EFI`, but this bootloader is missing.

[image]

I mounted the device with the bootloader and got the following data:

```
/mnt/
└── EFI
    └── debian
        ├── BOOTX64.CSV
        ├── fbx64.efi
        ├── grub.cfg
        ├── grubx64.efi
        ├── mmx64.efi
        └── shimx64.efi
```

After that, I created a `BOOT` directory inside `EFI` and copied `grubx64.efi` there under the name `BOOTX64.EFI`.

Following this, iPXE detected the bootloader and successfully booted from it.

[image]

Is there a way to manage these settings depending on where the bootloader is located, so that I don’t have to manually make changes to the boot partition?  
Or is this an issue at the iPXE level? How can this be configured?

mcb30, 2025-03-16:
As per the documentation at https://ipxe.org/cmd/sanboot you can use the `--filename` parameter to specify a non-standard boot filename such as `\EFI\debian\grubx64.efi`.

AlexanderZhirov, 2025-03-16:
@mcb30 Thank you! 🙂

NiKiZe, 2025-03-16:
You can tell sanboot which filename to give to the firmware to boot.
But do report this as a bug of the installer, which should always install the fallback loader since they can't rely on EFI firmware keeping the boot settings, this is also true for virtual machines, or USB disks, where you take the disk and move to a different machine, and that is what iSCSI boot is intended to be bale of.

More detailed explanation at https://github.com/ipxe/ipxe/discussions/1325#discussioncomment-11042035

closed 2025-06-15 by zube[bot]