ipxe/ipxe #1431: [tls] Improper handling of a Certificate Request when no Client Certificate is available

alexandrevoilab, 2025-03-18:
Hello,

I did not find any issue relating to this.

## Summary

When trying to `chain` to an https hosted image the request fail if the server send a `Certificate Request` and no `Client Certificate` is available.
The reported error is:
```
TLS 0x7d49f008 could not find certificate corresponding to private key
Operation not permitted (https://ipxe.org/410de38f)
```

## Bases for this claim

According to the [RFC 5246 (TLS1.2), section 7.4.2](https://datatracker.ietf.org/doc/html/rfc5246#section-7.4.2):
> 7.4.2.  Server Certificate
> [...]
> The same message type and structure will be used for the client's
> response to a certificate request message.  **Note that a client MAY
> send no certificates if it does not have an appropriate certificate**
> to send in response to the server's authentication request.

## Test setup

Setup nginx with `ssl_verify_client: optional` so that it will request a client certificate, but not require it.
See [ssl_verify_client's documentation](https://nginx.org/en/docs/http/ngx_http_ssl_module.html#ssl_verify_client).

Build ipxe **without** any client certificate.

Ensure that the connection is sucessfull with openssl:
```
openssl s_client -tls1_2 -msg -connect srv.local:443
```

## Patch

Here is the patch I used to get ipxe to handle this usecase.
```tls.patch
--- net/tls.c.orig      2025-03-12 15:03:43.574277250 +0100
+++ net/tls.c   2025-03-12 21:30:17.993533967 +0100
@@ -2468,25 +2468,24 @@
        x509_chain_put ( tls->client.chain );
        tls->client.chain = NULL;

+       /* Create client certificate chain */
+       tls->client.chain = x509_alloc_chain();
+       if ( ! tls->client.chain ) {
+               rc = -ENOMEM;
+               goto err_alloc;
+       }
+
        /* Determine client certificate to be sent */
        cert = x509_find_key ( NULL, tls->client.key );
        if ( ! cert ) {
                DBGC ( tls, "TLS %p could not find certificate corresponding "
-                      "to private key\n", tls );
-               rc = -EPERM_CLIENT_CERT;
-               goto err_find;
+                      "to private key, sending empty client chain\n", tls );
+               return 0;
        }
        x509_get ( cert );
        DBGC ( tls, "TLS %p selected client certificate %s\n",
               tls, x509_name ( cert ) );

-       /* Create client certificate chain */
-       tls->client.chain = x509_alloc_chain();
-       if ( ! tls->client.chain ) {
-               rc = -ENOMEM;
[... 254 lines cut ...]
<<< TLS 1.2, Handshake [length 0004], ServerHelloDone
    0e 00 00 00
>>> TLS 1.2, RecordHeader [length 0005]
    16 03 03 00 07
>>> TLS 1.2, Handshake [length 0007], Certificate
    0b 00 00 03 00 00 00
>>> TLS 1.2, RecordHeader [length 0005]
[snip]
>>> TLS 1.2, Handshake [length 0025], ClientKeyExchange
[snip]
>>> TLS 1.2, RecordHeader [length 0005]
[snip]
>>> TLS 1.2, ChangeCipherSpec [length 0001]
[snip]
>>> TLS 1.2, RecordHeader [length 0005]
[snip]
>>> TLS 1.2, Handshake [length 0010], Finished
[snip]
<<< TLS 1.2, RecordHeader [length 0005]
[snip]
<<< TLS 1.2, RecordHeader [length 0005]
[snip]
<<< TLS 1.2, Handshake [length 0010], Finished
[snip]
```

mcb30, 2025-03-31:
Good catch, and thank you for the detailed explanation!  Your patch looks generally good - I will commit a lightly modified version.

closed 2025-03-31 by mcb30

stappersg, 2025-04-16:
#1206 could have avoid this issue ...

PR #1437 (2025-03-31): [tls] Send an empty client certificate chain if we have no certificate
RFC5246 states that "a client MAY send no certificates if it does not have an appropriate certificate to send in response to the server's authentication request".  This use case may arise when the server is using optional client certificate verification and iPXE has not been provided with a client certificate to use.

Treat the absence of a suitable client certificate as a non-fatal condition and send a Certificate message containing no certificates as permitted by RFC5246.

Fixes: #1431 

Reported-by: Alexandre Ravey <alex [... cut]