ipxe/ipxe #1433: "No filesystem could mount root" on arm64 u-boot EFI due to incompatible initrd handling

kohlschuetter, 2025-03-24:
With commit 6a004be0cceab5d669eedb5a2e6ee2feab31d5bd, specific handling was added for `EFI_LOAD_FILE2_PROTOCOL` (Linux 5.7 and newer).

While a purely additive change, this specific change broke net-booting for my kernel setup on U-Boot EFI-boot on ARM64 only (my setup is pretty much identical for x86_64 where it works).

My iPXE boots a Linux 6 kernel with three initrd files: the DTB, some initramfs as well as a cpio overlay for the init script itself:
```
initrd --name dtb ${dtb-url}
initrd --name initramfs ${initramfs-url}
initrd --name init ${initcpio-url}
kernel ${kernel-url} ${kernel-params} ... initrd=initramfs initrd=init dtb=dtb
boot
```

From the (failing) boot log (with DEBUG=efi_init,efi_file):
```
http://SOME-SERVER/kernels/aarch64/6.12.18/boot/dtbs/6.12.18/amlogic/meson-g12b-odroid-n2-plus.dtb... ok

Not a PE-COFF file
http://SOME-SERVER/kernels/aarch64/6.12.18/boot/initramfs... 93%....    .... ok

Not a PE-COFF file
http://boot.snoboot.net/ipxe/init.cpio.gz... ok

Not a PE-COFF file
http://SOME-SERVER/kernels/aarch64/6.12.18/boot/vmlinuz... ok

Not a PE-COFF file
EFIFILE initrd.magic installed as /VenMedia(5568e427-68fc-4f3d-ac74-ca555231cc68)

EFI stub: Booting Linux Kernel...
EFIFILE initrd.magic load at 0x0+0x0
EFIFILE initrd.magic [0x00000000,0x0000d59f) dtb
EFIFILE initrd.magic [0x0000d59f,0x0000e000) pad
EFIFILE initrd.magic [0x0000e000,0x067a68f5) initramfs
EFIFILE initrd.magic [0x067a68f5,0x067a7000) pad
EFIFILE initrd.magic [0x067a7000,0x067e572b) init
EFIFILE initrd.magic load at 0xdb610000+0x67e572b
EFIFILE initrd.magic [0x00000000,0x0000d59f) dtb
EFIFILE initrd.magic [0x0000d59f,0x0000e000) pad
EFIFILE initrd.magic [0x0000e000,0x067a68f5) initramfs
EFIFILE initrd.magic [0x067a68f5,0x067a7000) pad
EFIFILE initrd.magic [0x067a7000,0x067e572b) init
EFIFILE initrd.magic read [0x00000000,0x067e572b)
EFIFILE initrd.magic [0x00000000,0x0000d59f) dtb
EFIFILE initrd.magic [0x0000d59f,0x0000e000) pad
EFIFILE initrd.magic [0x0000e000,0x067a68f5) initramfs
EFIFILE initrd.magic [0x067a68f5,0x067a7000) pad
EFIFILE initrd.magic [0x067a7000,0x067e572b) init
EFI stub: Loaded initrd from LINUX_EFI_INITRD_MEDIA_GUID device path
EFIFILE open volume
EFIFILE <root> opened via <volume>
EFIFILE dtb opened via dtb
EFIFILE dtb get file information
EFIFILE dtb read [0x00000000,0x0000d59f)
EFIFILE dtb closed
EFIFILE <root> closed
EFI stub: Using DTB from command line
EFI stub: Exiting boot services...

[    1.418718] printk: legacy console [ttyAML0] enabled
[    1.421759] Freeing initrd memory: 106388K
[    1.441744] brd: module loaded
[    1.447756] usbcore: registered new interface driver cdc_acm
[    1.452678] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[    1.460728] usbcore: registered new interface driver usbserial_generic
[    1.467142] usbserial: USB Serial support registered for generic
[    1.476349] meson-sm: secure-monitor enabled
[    1.484095] hid: raw HID events driver (C) Jiri Kosina
[    1.489095] usbcore: registered new interface driver usbhid
[    1.494106] usbhid: USB HID core driver
[    1.501758] gre: GRE over IPv4 demultiplexor driver
[    1.506747] IPv6: Loaded, but administratively disabled, reboot required to enable
[    1.513580] NET: Registered PF_PACKET protocol family
[    1.518724] Key type dns_resolver registered
[    1.531387] registered taskstats version 1
[    1.536730] Loading compiled-in X.509 certificates
[    1.544962] Loaded X.509 cert 'Build time autogenerated kernel key: e2e59444020b9636952d59779a70d979bb7b9685'
[    1.566279] zswap: loaded using pool lzo/zsmalloc
[    1.571970] Demotion targets for Node 0: null
[    1.577484] Key type .fscrypt registered
[    1.582342] Key type fscrypt-provisioning registered
[    1.608037] printk: legacy console [netcon0] enabled
[    1.612985] netconsole: network logging started
[    1.618069] clk: Not disabling unused clocks
[    1.622985] PM: genpd: Disabling unused power domains
[    1.628064] md: Skipping autodetection of RAID arrays. (raid=autodetect will force)
[    1.635096] RAMDISK: Couldn't find valid RAM disk image starting at 0.
[    1.641503] List of all partitions:
[    1.646351] 0100            4096 ram0 
[    1.646358]  (driver?)
[    1.656029] 0101            4096 ram1 
[    1.656035]  (driver?)
[    1.665691] 0102            4096 ram2 
[    1.665697]  (driver?)
[    1.675360] 0103            4096 ram3 
[    1.675365]  (driver?)
[    1.685013] 0104            4096 ram4 
[    1.685018]  (driver?)
[    1.694676] 0105            4096 ram5 
[    1.694682]  (driver?)
[    1.704332] 0106            4096 ram6 
[    1.704337]  (driver?)
[    1.714008] 0107            4096 ram7 
[    1.714013]  (driver?)
[    1.723668] 0108            4096 ram8 
[    1.723674]  (driver?)
[    1.733338] 0109            4096 ram9 
[    1.733343]  (driver?)
[    1.743000] 010a            4096 ram10 
[    1.743006]  (driver?)
[    1.752680] 010b            4096 ram11 
[    1.752685]  (driver?)
[    1.762345] 010c            4096 ram12 
[    1.762350]  (driver?)
[    1.771600] 010d            4096 ram13 
[    1.771605]  (driver?)
[    1.780860] 010e            4096 ram14 
[    1.780865]  (driver?)
[    1.790102] 010f            4096 ram15 
[    1.790107]  (driver?)
[    1.799370] No filesystem could mount root, tried: 
[    1.799374] 
[    1.808627] Kernel panic - not syncing: VFS: Unable to mount root fs on "" or unknown-block(0,0)
[    1.821437] CPU: 3 UID: 0 PID: 1 Comm: swapper/0 Not tainted 6.12.18 #6
[    1.827265] Hardware name: hardkernel Hardkernel ODROID-N2/Hardkernel ODROID-N2, BIOS 2025.04-rc4-00030-g69fb899133b6 04/01/2025
[    1.843637] Call trace:
[    1.848455]  dump_backtrace+0xfc/0x160
[    1.853312]  show_stack+0x30/0x60
[    1.858152]  dump_stack_lvl+0x60/0x98
[    1.862991]  dump_stack+0x18/0x30
[    1.867825]  panic+0x170/0x378
[    1.872661]  mount_root_generic+0x3cc/0x418
[    1.877521]  mount_root+0x3d8/0x438
[    1.882360]  prepare_namespace+0x4d0/0x588
[    1.887208]  kernel_init_freeable+0x43c/0x4d0
[    1.892075]  kernel_init+0x2c/0x150
[    1.896917]  ret_from_fork+0x10/0x20
[    1.901761] SMP: stopping secondary CPUs
[    1.906606] Kernel Offset: disabled
[    1.911437] CPU features: 0x00,00000183,40200000,0200421b
[    1.916298] Memory Limit: none
[    1.921118] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on "" or unknown-block(0,0) ]---
```

I've tried my setup with the latest stock U-Boot as well as an older one that was working before that commit.

With the most recent iPXE source, I'm able to boot once I disable parts of `efi_file_install` (in src/interface/efi/efi_file.c), by simply returning `0` before `/* Claim Linux initrd fixed device path */ `:

```
http://SOME-SERVER/kernels/aarch64/6.12.18/boot/dtbs/6.12.18/amlogic/meson-g12b-odroid-n2.dtb... ok

Not a PE-COFF file
http://SOME-SERVER/kernels/aarch64/6.12.18/boot/initramfs... 95%....    .... ok

Not a PE-COFF file
http://SOME-SERVER/init.cpio.gz... ok

Not a PE-COFF file
http://SOME-SERVER/kernels/aarch64/6.12.18/boot/vmlinuz... ok

Not a PE-COFF file
EFIFILE DAWG

EFI stub: Booting Linux Kernel...
EFIFILE open volume
EFIFILE <root> opened via <volume>
EFIFILE initramfs opened via initramfs
EFIFILE initramfs get file information
EFIFILE initramfs read [0x00000000,0x067988f5)
EFIFILE initramfs closed
EFIFILE <root> closed
EFIFILE open volume
EFIFILE <root> opened via <volume>
EFIFILE init opened via init
EFIFILE init get file information
EFIFILE init read [0x00000000,0x0003e72b)
EFIFILE init closed
EFIFILE <root> closed

EFI stub: Loaded initrd from command line option
EFIFILE open volume
EFIFILE <root> opened via <volume>
EFIFILE dtb opened via dtb
EFIFILE dtb get file information
EFIFILE dtb read [0x00000000,0x0000d51b)
EFIFILE dtb closed
EFIFILE <root> closed
EFI stub: Using DTB from command line
EFI stub: Exiting boot services...

[    1.382425] printk: legacy console [ttyAML0] enabled
[    1.396803] brd: module loaded
[    1.402924] usbcore: registered new interface driver cdc_acm
[    1.407888] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[    1.415997] ...
```

As you can see, there's something weird going on with "initrd.magic" in the failing case (too much "magic"?)

mcb30, 2025-03-24:
As it happens, there's an `fdt` command due to be added this week, to cover this specific use case of providing a downloaded Devicetree blob to the loaded kernel (and hiding it from the magic initrd).

kohlschuetter, 2025-03-24:
Output from imgstat (sizes vary depending on kernel, of course).

```
dtb : 54683 bytes

initramfs : 107053740 bytes [gzip]

init : 255851 bytes [gzip]

vmlinuz : 10548948 bytes [gzip] [SELECTED] "earlyprintk no_console_suspend clk_ignore_unused fsck.repair=yes net.ifnames=0 nomodeset noquiet zswap.enabled=1 earlycon=efifb keep_bootcon console=ttyS0,115200n8 console=aml-uart,0xff803000 console=ttyAML0,115200n8 initrd=initramfs initrd=init dtb=dtb "
```

kohlschuetter, 2025-03-24:
@NiKiZe nice idea!!! :-D  This really works haha

```
initramfs : 107053740 bytes [gzip]

init : 255851 bytes [gzip]

dtb : 54683 bytes

vmlinuz : 10548948 bytes [gzip] [SELECTED] 
```

kohlschuetter, 2025-03-24:
@NiKiZe @mcb30 So it appears that for the "magic" initrd, the order really matters.

In other words, the "fdt" command won't make much of a difference if more than one "non-DTB" initrd command is called.
It's actually a very subtle bug that can sit undetected:

If I reorder the initrd calls such that "init" comes before "initramfs" and "dtb", my custom init script is actually just overridden by the original initramfs code, which is of course not intended.

With my setup, the system will then hang in the init phase due to some missing configuration. In some other case, it may not even be noticed until something else goes wrong (e.g., the wrong modules being loaded, some service not called, etc.)

I don't think just documenting" it somewhere will be sufficient. Since this codepath is Linux-specific anyways, we should probably parse the kernel command line for "init=" and "dtb=" entries and at least determine a partial order for the images that go into "initrd.magic"...

NiKiZe, 2025-03-24:
The initramfs is a cpio archive that can be appended, however I do always consider it as "it must start with a valid archive".
It is, and always have been up to "you" to make sure this order is correct. It is true that the behaviour is "new" for EFI boot. But this is how it has always been for pcbios. And most users, before knowing anything EFI specifics (adding multiple initrd= incarnations), does expect the pcbios equivalent of concating initrd parts.

The `fdt` command would help here in hiding that file from magic creation.

There is one more thing tho, I would not have expected the magic initrd to get files unless permission or cpio name is given.
Also I would not expect the kernel to read `EFI_LOAD_FILE2_PROTOCOL` when initrd= was given, but is this one of those kernels with a bug for that case?

Regardless, parsing the command line for any of these is IMO a horrible idea.

kohlschuetter, 2025-03-24:
I don't know why think parsing is a horrible idea: the command line arguments specify the order in which initrds are loaded. I think adding a Linux-specific extension at all is already opening that can of worms.

Anyhow. I'd be happy if this `EFI_LOAD_FILE2_PROTOCOL` special-case could be removed by a macro. The way it is right now is too brittle.

kohlschuetter, 2025-03-27:
OK, I've found one argument against my proposed reordering by parsing the command line, adding here as a cautionary tale, and real, as it happened today...

For some reason, either U-Boot or iPXE corrupts the first 1M or so of my initrd img (really not sure what's going there, but I need a fix).

The bug is quite subtle, as only a few bytes (at random places, different ones each time) are modified (many times the boot just succeeds, but sometimes it crashes with "Illegal instruction" in the middle of `udhcpc` etc.).

When I add an empty "padding" image as the first `initrd` call, a cpio.gz (I named it `aaaaargh.cpio` to represent the wasted day) with a zero-byte-filled file of 1MB, the problem goes away, because the modifications hit that padding area, not the actual payload. The padding initrd does not need to be specified as `initrd=` for the Linux kernel.

In other words:
```
initrd --name aaaaargh ${baseurl}/aaaaargh.cpio
initrd --name initramfs ${initramfs-url}
initrd --name dtb ${dtb-url}
kernel ${kernel-url} ${kernel-params} init=initramfs dtb=dtb
boot
```

mcb30, 2025-03-28:
> As it happens, there's an `fdt` command due to be added this week, to cover this specific use case of providing a downloaded Devicetree blob to the loaded kernel (and hiding it from the magic initrd).

@kohlschuetter Try using the new `fdt` command to download the devicetree blob.  It will automatically hide the downloaded file from the magic initrd, so your problem will go away.

With the current iPXE master (commit https://github.com/ipxe/ipxe/commit/b20f506a7) you should find that the kernel ignores the `dtb=...` command-line parameter and automatically picks up whatever file you specified with the iPXE `fdt` command.

mcb30, 2025-03-28:
> I therefore don't think just documenting this behavior somewhere will be sufficient. Since this codepath is Linux-specific anyways, we should probably parse the kernel command line for "init=" and "dtb=" entries and at least determine a partial order for the images that go into "initrd.magic"...

No.  The order of `initrd` commands in iPXE scripts has always mattered for BIOS, and (since it came into existence) for the magic initrd.  This ordering has been well-defined for almost two decades.

It's a Linux kernel choice to ignore the `initrd=` and `dtb=` command-line parameters in favour of "magic" initrd or device tree blob provision, and the kernel will already print messages such as the "Loaded initrd from LINUX_EFI_INITRD_MEDIA_GUID device path" to let you know that it's done so.  (It might be nice if the kernel explicitly warned you that it had ignored some existent `initrd=...` arguments, but that's not our decision to make.)

closed 2025-03-28 by kohlschuetter

kohlschuetter, 2025-03-31:
@mcb30 I think I've found the culprit https://lists.denx.de/pipermail/u-boot/2025-March/585238.html

mcb30, 2025-03-31:
> [@mcb30](https://github.com/mcb30) I think I've found the culprit https://lists.denx.de/pipermail/u-boot/2025-March/585238.html

Great work: thank you for tracking that down and for copying me in on the email!