ipxe/ipxe #1438: GPF on Dell R7625/BIOS 1.11.2 attempting to exit or boot anything from ipxe

optiz0r, 2025-04-07:
We're seeing a general protection fault while trying to boot anything from an ipxe built from recent git master (commit 7e64e9b6703e6dd363c063d545a5fe63bbc70011) (and also a build from Aug 2024, so not a recent behaviour change).

[image]

```
# objdump -S bin-x86_64-efi/ipxe.efi.tmp
000000000005f474 <unregister_settings>:
   5f474:       55                      push   %rbp
   5f475:       48 8d 6f 28             lea    0x28(%rdi),%rbp
   5f479:       53                      push   %rbx
   5f47a:       48 89 fb                mov    %rdi,%rbx
   5f47d:       52                      push   %rdx
   5f47e:       48 8b 7b 28             mov    0x28(%rbx),%rdi  # crash with RIP here
   5f482:       48 39 ef                cmp    %rbp,%rdi
   5f485:       74 0b                   je     5f492 <unregister_settings+0x1e>
   5f487:       48 83 ef 18             sub    $0x18,%rdi
   5f48b:       e8 e4 ff ff ff          callq  5f474 <unregister_settings>
   5f490:       eb ec                   jmp    5f47e <unregister_settings+0xa>
   5f492:       48 8b 43 10             mov    0x10(%rbx),%rax
   5f496:       48 8b 38                mov    (%rax),%rdi
   5f499:       e8 14 a0 ff ff          callq  594b2 <ref_decrement>
   5f49e:       48 8b 43 20             mov    0x20(%rbx),%rax
   5f4a2:       48 8b 53 18             mov    0x18(%rbx),%rdx
   5f4a6:       48 c7 43 10 00 00 00    movq   $0x0,0x10(%rbx)
   5f4ad:       00 
   5f4ae:       48 8b 3b                mov    (%rbx),%rdi
   5f4b1:       48 89 42 08             mov    %rax,0x8(%rdx)
   5f4b5:       48 89 10                mov    %rdx,(%rax)
   5f4b8:       e8 f5 9f ff ff          callq  594b2 <ref_decrement>
   5f4bd:       58                      pop    %rax
   5f4be:       5b                      pop    %rbx
   5f4bf:       5d                      pop    %rbp
   5f4c0:       e9 58 f8 ff ff          jmpq   5ed1d <apply_settings>
```

This includes both attempting to boot from next local boot device (`exit` in the ipxe script), or attempting to chain load an OS with something like the following:

```
initrd http://yum.example.com/pxeboot/initrd.img
chain http://yum.example.com/pxeboot/vmlinuz inst.ks=https://build.example.com/kickstart ramdisk_size=8216 pcie_aspm=off inst.sshd inst.noverifyssl crashkernel=auto ipv6.disable=1 initrd=initrd.img 
```

The ipxe binary being booted is undionly.kpxe, booted via pxe, built at the same time as the EFI binary (bin-x86_64-efi/ipxe.efi.tmp). I'm not familiar enough to know if the object dump of the efi binary and the undionly.kpxe offsets are directly comparable. Any tips on how to collect more diagnostically useful data here appreciated.

Will see about enabling Serial-over-lan and enabling debug output to try and capture more info in the meantime.

NiKiZe, 2025-04-07:
You mention undionly.kpxe, while the screenshot shows  ipxe.efi, which one is it?

* Have you tried snponly.efi just to narrow down what it is that could go wrong?
* Do you have any known working build of iPXE? If so could you bisect?

Or is this perhaps a case of iDRAC using the NIC which iPXE will take over?

optiz0r, 2025-04-07:
> You mention undionly.kpxe, while the screenshot shows ipxe.efi, which one is it?
Apologies, I was confused about which one was being used. I have confirmed this is booting the ipxe.efi binary, which was built this morning from git master.

> Have you tried snponly.efi just to narrow down what it is that could go wrong?
Not yet, will try this and report back.

> Do you have any known working build of iPXE? If so could you bisect?

The 2024-08-06 and 2025-04-07 binaries both work fine on other hardware platforms (VMs, other physical servers, including other Dell Poweredge R-series). The affected machines (a batch of four newly delivered, with same spec and fully updated firmwares) are failing on both the older and newer ipxe.efi binaries, so nothing to bissect. The ipxe binary was upgraded just to check it wasn't an issue with the older one.

A previous batch of R7625s with same CPU/memory configurations, but older firmwares have previously been successfully built with the 2024-08-06 ipxe.efi binary. Have not tried downgrading firmwares on the newer batch of servers to match the old, or upgrade firmwares on the older batch to match the new yet.

> Or is this perhaps a case of iDRAC using the NIC which iPXE will take over?

iDRAC is using dedicated on-board RJ45 connector, and we are pxe booting from a 25GB PCI network card, so there should be no conflict between iDRAC and ipxe here.

NiKiZe, 2025-04-07:
Probable duplicate of #1391
Since you are running ipxe.efi, it will try to "claim" all nics it can reach. (Load it's native driver if there is one)
snponly will only use the nic it was booted over, snp.efi will use all nics, but thru the firmware provided drivers.
It's interesting tho that the RSOD only happens when iPXE exits, probably during release of handles.

You will probably want to report this to Dell as well if you haven't already, since as you say, this is only on this specific model.

optiz0r, 2025-04-08:
For the record, booting with:
* snponly.efi - works
* snp.efi - works
* PCI slot containing BCM57508 disabled in BIOS - does not work.

Suspect this might be because the boot interface is a broadcom 25G NIC using the same firmware package as the BCM57508 reported in #1391, so it's actually the same problem even though the disabling workaround does not work here. We'll try physically removing the 100G card later today, but I don't expect that'll change anything.

I'll close this off as a duplicate. Thanks again for the quick responses.

NiKiZe, 2025-04-08:
Thanks for confirming.

snp usage is a workaround at best, but more just a debug step, the real fix(es) will probably involve firmware updates by Dell/Broadcom, and maybe some handling in the native iPXE driver, but that also usually comes in from Broadcom, so important that this issue is raised at their end. (Only something that usually real customers can do)

mcb30, 2025-04-11:
@optiz0r Thank you for the detailed debug information.  You

optiz0r, 2025-04-11:
@mcb30 Forgive me, I'm not familiar with obj dump. I presume you're referring to these lines here:

```
Stack Dump:
  56D117D0  00000000526762E0 .bgR....
  56D117D8  FFFFFFFFFFFFFFE8 ........
  56D117E0  0000000000000010 ........
  56D117E8  00000000525CC490 ..\R....
  56D117F0  00000000526762E0 .bgR....
  56D117F8  0000000000000360 `.......
  56D11800  0000000000000388 ........
...
```

The offsets 56D11xxx don't appear in the object dump output. These numbers look broadly similar to the second column in the screenshot, whereas the objdump offset is in the last column with the plus symbol. How do I go about doing the equivalent mapping for the Stack Dump above? I presume the stack dump absolute and the objdump output is relative to some base address?

FTR we pulled out the 100G PCIe card earlier this week, and the problem persisted. Likely because the 25G NICs we're actually booting from use the same firmware. Therefore we're sure the problem relates to the 23.xx firmware versions. Downgrading to 22.xx also fixed the problem as in #1391.

I do now have a support case open with Dell for this with a request they escalate the problem up to Broadcom for investigation/fix.

mcb30, 2025-04-11:
> [@mcb30](https://github.com/mcb30) Forgive me, I'm not familiar with obj dump. I presume you're referring to these lines here:
> 
> ```
> Stack Dump:
>   56D117D0  00000000526762E0 .bgR....
>   56D117D8  FFFFFFFFFFFFFFE8 ........
>   56D117E0  0000000000000010 ........
>   56D117E8  00000000525CC490 ..\R....
>   56D117F0  00000000526762E0 .bgR....
>   56D117F8  0000000000000360 `.......
>   56D11800  0000000000000388 ........
> ...
> ```
> 
> The offsets 56D11xxx don't appear in the object dump output. These numbers look broadly similar to the second column in the screenshot, whereas the objdump offset is in the last column with the plus symbol. How do I go about doing the equivalent mapping for the Stack Dump above? I presume the stack dump absolute and the objdump output is relative to some base address?
> 
> 525c c48b - 05 d48b = 5256 d000 (base offset from screenshot) 56D1 17D0 - 5256 d000 = 047a 47d0 (calculated offset from stack dump) object dump taps out at 0x000b 527e, doesn't go as far as 0x047x xxxx

Sorry: it's the stack **values** (rather than the stack **addresses**) that are of interest.  For example, the value 0x526762E0 is at offset +0x5f490, which should be within your objdump output.

optiz0r, 2025-04-11:
> 0x526762E0 is at offset +0x5f49

```
000000000005f474 <unregister_settings>:
   5f474:       55                      push   %rbp
   5f475:       48 8d 6f 28             lea    0x28(%rdi),%rbp
   5f479:       53                      push   %rbx
   5f47a:       48 89 fb                mov    %rdi,%rbx
   5f47d:       52                      push   %rdx
   5f47e:       48 8b 7b 28             mov    0x28(%rbx),%rdi
   5f482:       48 39 ef                cmp    %rbp,%rdi
   5f485:       74 0b                   je     5f492 <unregister_settings+0x1e>
   5f487:       48 83 ef 18             sub    $0x18,%rdi
   5f48b:       e8 e4 ff ff ff          callq  5f474 <unregister_settings>
   5f490:       eb ec                   jmp    5f47e <unregister_settings+0xa>
   5f492:       48 8b 43 10             mov    0x10(%rbx),%rax
   5f496:       48 8b 38                mov    (%rax),%rdi
   5f499:       e8 14 a0 ff ff          callq  594b2 <ref_decrement>
   5f49e:       48 8b 43 20             mov    0x20(%rbx),%rax
   5f4a2:       48 8b 53 18             mov    0x18(%rbx),%rdx
   5f4a6:       48 c7 43 10 00 00 00    movq   $0x0,0x10(%rbx)
   5f4ad:       00 
   5f4ae:       48 8b 3b                mov    (%rbx),%rdi
   5f4b1:       48 89 42 08             mov    %rax,0x8(%rdx)
   5f4b5:       48 89 10                mov    %rdx,(%rax)
   5f4b8:       e8 f5 9f ff ff          callq  594b2 <ref_decrement>
   5f4bd:       58                      pop    %rax
   5f4be:       5b                      pop    %rbx
   5f4bf:       5d                      pop    %rbp
   5f4c0:       e9 58 f8 ff ff          jmpq   5ed1d <apply_settings>
```

> 56D117F0  00000000526762E0 .bgR....

`00000000526762E0 -5256d000=1092e0`, this offset doesn't exist, but within objdump I see multiple references to `efi_path_net_drivers`

```
   4da7a:	48 8d 05 5f b8 0b 00 	lea    0xbb85f(%rip),%rax        # 1092e0 <efi_path_net_driver>
   4daaf:	48 8d 15 2a b8 0b 00 	lea    0xbb82a(%rip),%rdx        # 1092e0 <efi_path_net_driver>
   4dad4:	48 8d 1d 05 b8 0b 00 	lea    0xbb805(%rip),%rbx        # 1092e0 <efi_path_net_driver>
   4df61:	48 8d 35 78 b3 0b 00 	lea    0xbb378(%rip),%rsi        # 1092e0 <efi_path_net_driver>
   4e0c2:	48 8d 05 17 b2 0b 00 	lea    0xbb217(%rip),%rax        # 1092e0 <efi_path_net_driver>
   4e154:	4c 8d 25 85 b1 0b 00 	lea    0xbb185(%rip),%r12        # 1092e0 <efi_path_net_driver>
   4e369:	4c 8d 3d 70 af 0b 00 	lea    0xbaf70(%rip),%r15        # 1092e0 <efi_path_net_driver>
   4e3e6:	4c 8d 3d f3 ae 0b 00 	lea    0xbaef3(%rip),%r15        # 1092e0 <efi_path_net_driver>
   4e5d0:	48 8d 2d 09 ad 0b 00 	lea    0xbad09(%rip),%rbp        # 1092e0 <efi_path_net_driver>
   4e688:	4c 8d 25 51 ac 0b 00 	lea    0xbac51(%rip),%r12        # 1092e0 <efi_path_net_driver>
   4e6d7:	48 8d 05 02 ac 0b 00 	lea    0xbac02(%rip),%rax        # 1092e0 <efi_path_net_driver>
   4e715:	48 8d 05 c4 ab 0b 00 	lea    0xbabc4(%rip),%rax        # 1092e0 <efi_path_net_driver>
```

> 56D11828  000000005260B440 @.`R....

`5260B440-5256d000=9e440`

```
000000000009e434 <lldp_remove>:
   9e434:       53                      push   %rbx
   9e435:       48 89 f7                mov    %rsi,%rdi
   9e438:       48 89 f3                mov    %rsi,%rbx
   9e43b:       e8 34 10 fc ff          callq  5f474 <unregister_settings>
   9e440:       48 8b 7b 58             mov    0x58(%rbx),%rdi
   9e444:       e8 ce 69 fa ff          callq  44e17 <free>
   9e449:       48 c7 43 58 00 00 00    movq   $0x0,0x58(%rbx)
   9e450:       00 
   9e451:       5b                      pop    %rbx
   9e452:       c3                      retq 
```

>  56D11830  0000000052676420  dgR....
[... 106 lines cut ...]
   230c4:       48 89 10                mov    %rdx,(%rax)
   230c7:       e8 1c f4 ff ff          callq  224e8 <efi_close_by_driver>
   230cc:       5a                      pop    %rdx
   230cd:       48 89 ef                mov    %rbp,%rdi
   230d0:       5d                      pop    %rbp
   230d1:       41 5c                   pop    %r12
   230d3:       e9 3f 1d 02 00          jmpq   44e17 <free>
```

>  56D11880  0000000052675600 .VgR....

`52675600-5256d000=108600`

```
   219c4:	4c 8d 2d 35 6c 0e 00 	lea    0xe6c35(%rip),%r13        # 108600 <efipci_driver>
   96d2f:	48 8d 1d ca 18 07 00 	lea    0x718ca(%rip),%rbx        # 108600 <efipci_driver>
   96fb0:	48 8d 1d 49 16 07 00 	lea    0x71649(%rip),%rbx        # 108600 <efipci_driver>
   970b2:	4c 8d 35 47 15 07 00 	lea    0x71547(%rip),%r14        # 108600 <efipci_driver>
```

>  56D11888  00000000526893C8 ..hR....
>  56D11890  0000000056D118A8 ...V....
>  56D11898  0000000052603F2A *?`R....

Let me know if you need to go back further.

mcb30, 2025-04-11:
Thank you very much!  It's not yet clear where the garbage pointer value 0xF000FF54F000DDC0 is coming from, but at least we can see part of the stack backtrace:
* `unregister_settings(<garbage pointer>)`
* ...called from `lldp_remove()`
* ...called via the `driver->remove()` method from `unregister_netdev()`
* ...called from `bnxt_remove_one()`
* ...called from `pci_remove()`
* ...called from `efipci_stop()`

As a first step, it's therefore worth building iPXE with `DEBUG=efi_pci,pci,bnxt,netdevice,lldp,settings` since that will enable assertions in all of those objects and potentially reveal some corruption.

optiz0r, 2025-04-11:
From ipxe.efi with the requested DEBUG flags, triggered by selecting entry which runs `exit` from initial ipxe script menu

```
  NBP file downloaded successfully.
Settings 0x527d7de0 ("efi") registered
Settings 0x527dcce0 ("builtin") registered
Settings 0x527dd800 ("smbios") registered
Settings 0x527e1360 ("pci") registered
Settings 0x527e1080 ("netX") registered
iPXE initialising devices...
NETDEV net0 registered (phys MNP-0000:01:00.0 hwaddr 6c:92:cf:f6:d6:f0)
Settings 0x527f67f8 ("net0") registered
Settings 0x527f68d0 ("net0.dhcp") registered
Settings 0x527f6928 ("net0.link") registered
Settings 0x527f68d0 ("net0.dhcp") unregistered
Settings 0x527f6c20 ("net0.dhcp") registered
NETDEV net0 transmission 0x527f7590 failed: Network unreachable (https://ipxe.org/28086090)
Settings 0x527f69e8 ("net0.lldp") registered
LLDP net0 registered
NETDEV net0 link is up
NETDEV net0 opening
autoexec.ipxe... Not found (https://ipxe.org/2d12618e)
Settings 0x527f6c20 ("net0.dhcp") unregistered
NETDEV net0 closing
Settings 0x527f69e8 ("net0.lldp") unregistered
LLDP net0 unregistered
Settings 0x527f6928 ("net0.link") unregistered
Settings 0x527f67f8 ("net0") unregistered
NETDEV net0 unregistered
EFIPCI 0000:00:00.0 (1022:14a4 class 060000) has no driver
EFIPCI 0000:00:00.2 (1022:149e class 080600) has no driver
EFIPCI 0000:00:00.3 (1022:14a6 class 080700) has no driver
EFIPCI 0000:00:01.0 (1022:149f class 060000) has no driver
EFIPCI 0000:00:01.1 type 81 is not type 00
EFIPCI 0000:01:00.0 (14e4:1751 class 020000) has driver "14e4-1751"
EFIPCI 0000:01:00.0 (14e4:1751 class 020000) has driver "14e4-1751"
EFIPCI 0000:01:00.0 (14e4:1751 class 020000) has driver "14e4-1751"
EFIPCI 0000:01:00.0 (14e4:1751 class 020000) has driver "14e4-1751"
0000:01:00.0 (14e4:1751) has driver "14e4-1751"
0000:01:00.0 has mem ef030000 io 0 irq 255
0000:01:00.0 device not enabled by BIOS! Updating PCI command 0406->0407
0000:01:00.0 latency timer is unreasonably low at 0. Setting to 32.
NETDEV net0 registered (phys 0000:01:00.0 hwaddr 6c:92:cf:f6:d6:f0)
Settings 0x527f6f38 ("net0") registered
Settings 0x527f7010 ("net0.dhcp") registered
Settings 0x527f7068 ("net0.link") registered
Settings 0x527f7010 ("net0.dhcp") unregistered
Settings 0x527f6c20 ("net0.dhcp") registered
NETDEV net0 transmission 0x527f6990 failed: Network unreachable (https://ipxe.org/28086090)
Settings 0x527f7128 ("net0.lldp") registered
LLDP net0 registered
EFIPCI 0000:01:00.0 using driver "14e4-1751"
EFIPCI 0000:01:00.1 (14e4:1751 class 020000) has driver "14e4-1751"
EFIPCI 0000:01:00.1 (14e4:1751 class 020000) has driver "14e4-1751"
EFIPCI 0000:01:00.1 (14e4:1751 class 020000) has driver "14e4-1751"
EFIPCI 0000:01:00.1 (14e4:1751 class 020000) has driver "14e4-1751"
0000:01:00.1 (14e4:1751) has driver "14e4-1751"
0000:01:00.1 has mem ef020000 io 0 irq 255
0000:01:00.1 device not enabled by BIOS! Updating PCI command 0006->0007
0000:01:00.1 latency timer is unreasonably low at 0. Setting to 32.
NETDEV net1 registered (phys 0000:01:00.1 hwaddr 6c:92:cf:f6:d6:f1)
Settings 0x527f9278 ("net1") registered
Settings 0x527f93a8 ("net1.link") registered
Settings 0x527f9468 ("net1.lldp") registered
LLDP net1 registered
EFIPCI 0000:01:00.1 using driver "14e4-1751"
EFIPCI 0000:01:00.2 (14e4:1751 class 020000) has driver "14e4-1751"
EFIPCI 0000:01:00.2 (14e4:1751 class 020000) has driver "14e4-1751"
EFIPCI 0000:01:00.2 (14e4:1751 class 020000) has driver "14e4-1751"
EFIPCI 0000:01:00.2 (14e4:1751 class 020000) has driver "14e4-1751"
[... 468 lines cut ...]
   a5b03:       48 89 df                mov    %rbx,%rdi
   a5b06:       e8 65 3d ff ff          callq  99870 <dbg_autocolourise>
   a5b0b:       f6 05 fe 2b 10 00 01    testb  $0x1,0x102bfe(%rip)        # 1a8710 <__debug_disable_lldp>
   a5b12:       75 20                   jne    a5b34 <lldp_remove+0x46>
   a5b14:       48 8b 73 50             mov    0x50(%rbx),%rsi
   a5b18:       31 c0                   xor    %eax,%eax
   a5b1a:       48 8d 3d 24 7d 04 00    lea    0x47d24(%rip),%rdi        # ed845 <.LC0>
   a5b21:       e8 b7 3b ff ff          callq  996dd <dbg_printf>
   a5b26:       f6 05 e3 2b 10 00 01    testb  $0x1,0x102be3(%rip)        # 1a8710 <__debug_disable_lldp>
   a5b2d:       75 05                   jne    a5b34 <lldp_remove+0x46>
   a5b2f:       e8 c6 3d ff ff          callq  998fa <dbg_decolourise>
   a5b34:       48 8b 7b 58             mov    0x58(%rbx),%rdi
   a5b38:       e8 c9 03 fa ff          callq  45f06 <free>
   a5b3d:       48 c7 43 58 00 00 00    movq   $0x0,0x58(%rbx)
   a5b44:       00 
   a5b45:       5b                      pop    %rbx
   a5b46:       c3                      retq   

```

There are also a great many of these, which taking a look at the code might be happening on receipt of any packet:

```
assert(!iob) failed at drivers/net/bnxt/bnxt.c line 527
```

mcb30, 2025-04-11:
> There are also a great many of these, which taking a look at the code might be happening on receipt of any packet:
> 
> ```
> assert(!iob) failed at drivers/net/bnxt/bnxt.c line 527
> ```

Thank you.  Yes, those need to be addressed first, since they are most likely indicating memory corruption (which is why we later see invalid pointer dereferences).

You could try putting:
```
if ( ! iob ) {
    DBG ( "Invalid RX completion %d\n", desc_idx );
    return;
}
```
before the `assert ( !iob )`, just as an experiment to see what happens.

@jw14812 This looks to be pretty clearly a bnxt driver bug - any suggestions?

jw14812, 2025-04-11:
@mcb30 - Agreed, I was planning to address this in my next commit after the DMA pull request was finished

optiz0r, 2025-04-12:
> You could try putting: ... before the assert ( !iob ), just as an experiment to see what happens.

Tried, no change other than the line number of the printed asserts increasing by 4 to 537.

jw14812, 2025-04-13:
@optiz0r - please try this this patch

[pointer3.patch](https://github.com/user-attachments/files/19723339/pointer3.patch)

optiz0r, 2025-04-13:
@jw14812 That's done the trick!

jw14812, 2025-04-13:
@mcb30 - The changes are in pull request #1445

mcb30, 2025-04-14:
> [@mcb30](https://github.com/mcb30) - The changes are in pull request [#1445](https://github.com/ipxe/ipxe/pull/1445)

Thank you very much for looking at this so quickly.  I have requested some changes to the PR.

jw14812, 2025-04-24:
@optiz0r - If convenient, could you try the patch in https://github.com/ipxe/ipxe/pull/1445?  This has some of the suggested changes in place of pointer3.patch.

optiz0r, 2025-04-24:
@jw14812 The latest patches in 1445 still avoid the GPF issue.

I saw a stalled HTTP download again on a third attempt without debug info built in (after two previously successful boots).

With debug mode, I am seeing a great many logs of `asset(!iob)` failures on the console while the download percentage is still increasing, interspersed with periodic 440e60 and 380a60 errors which seem non-fatal. The rate of these messages and the slow speed of the serial console printing them meant it took an hour to (successfully) download the systemrescuecd12 boot files, which takes mere seconds without debug. So there definitely still seems something wrong, but probably a separate issue from the GPF crash.

NiKiZe, 2025-05-14:
Is the main issue here resolved with #1445 merged? (Test current git master)

I think there is a separate issue open for some stalling on high speed links over multiple hops, maybe relevant? (Will try to locate and update this comment)

NiKiZe, 2025-05-21:
I'm going to assume that this is resolved, if any evidence to the opposite shows up we'll reopen.

closed 2025-05-21 by NiKiZe