ipxe/ipxe #1454: Initramfs unpacking failed: Invalid magic at start of Compressed archive

hjensas, 2025-04-25:
With iPXE binary `bin/ipxe.usb` compiled from master (2025-04-24) booting linux kernel+initramfs fails.

The kernel and initrd is downloaded, and the kernel start booting. 
However it fails with Kernel panic and I see the following error message, indicating an issue with the initramfs.
```
Initramfs unpacking failed: Invalid magic at start of Compressed archive
```

iPXE script used:
```
#!ipxe
initrd --name initrd http://192.168.32.254:8081/boot-artifacts/agent.x86_64-initrd.img
kernel http://192.168.32.254:8081/boot-artifacts/agent.x86_64-vmlinuz initrd=initrd coreos.live.rootfs_url=http://192.168.32.254:8081/boot-artifacts/agent.x86_64-rootfs.img rw ignition.firstboot ignition.platform.id=metal console=ttyS0
boot
```

Switching to an iPXE image that I built from master around 2025-03-13 solves my issue.

The full boot linux boot log:
```
    [    0.000000] Linux version 5.14.0-427.50.1.el9_4.x86_64 (mockbuild@x86-64-02.build.eng.rdu2.redhat.com) (gcc (GCC) 11.4.1 20231218 (Red Hat 11.4.1-3), GNU ld version 2.35.2-43.el9) #1 SMP PREEMPT_DYNAMIC Wed Dec 18 13:06:23 EST 2024
    [    0.000000] The list of certified hardware and cloud instances for Red Hat Enterprise Linux 9 can be viewed at the Red Hat Ecosystem Catalog, https://catalog.redhat.com.
    [    0.000000] Command line: initrd=initrd coreos.live.rootfs_url=http://192.168.32.254:8081/boot-artifacts/agent.x86_64-rootfs.img rw ignition.firstboot ignition.platform.id=metal console=ttyS0
    [    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
    [    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
    [    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
    [    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
    [    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'compacted' format.
    [    0.000000] signal: max sigframe size: 1776
    [    0.000000] BIOS-provided physical RAM map:
    [    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
    [    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
    [    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
    [    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007ffdafff] usable
    [    0.000000] BIOS-e820: [mem 0x000000007ffdb000-0x000000007fffffff] reserved
    [    0.000000] BIOS-e820: [mem 0x00000000b0000000-0x00000000bfffffff] reserved
    [    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
    [    0.000000] BIOS-e820: [mem 0x00000000feffc000-0x00000000feffffff] reserved
    [    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
    [    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000087fffffff] usable
    [    0.000000] NX (Execute Disable) protection: active
    [    0.000000] SMBIOS 2.8 present.
    [    0.000000] DMI: Red Hat OpenStack Compute/RHEL, BIOS 1.16.1-1.el9 04/01/2014
    [    0.000000] Hypervisor detected: KVM
    [    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
    [    0.000001] kvm-clock: using sched offset of 464526636159 cycles
    [    0.000002] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
    [    0.000005] tsc: Detected 2595.122 MHz processor
    [    0.000935] last_pfn = 0x880000 max_arch_pfn = 0x400000000
    [    0.000993] MTRR map: 4 entries (3 fixed + 1 variable; max 19), built from 8 variable MTRRs
    [    0.000996] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
    [    0.001070] last_pfn = 0x7ffdb max_arch_pfn = 0x400000000
    [    0.003399] found SMP MP-table at [mem 0x000f5a70-0x000f5a7f]
    [    0.003419] Using GB pages for direct mapping
    [    0.003533] RAMDISK: [mem 0x74135000-0x7fe5cfff]
    [    0.003538] ACPI: Early table checksum verification disabled
    [    0.003541] ACPI: RSDP 0x00000000000F5A30 000014 (v00 BOCHS )
    [    0.003544] ACPI: RSDT 0x000000007FFE35D4 000034 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
    [    0.003549] ACPI: FACP 0x000000007FFE338C 0000F4 (v03 BOCHS  BXPC     00000001 BXPC 00000001)
    [    0.003553] ACPI: DSDT 0x000000007FFDF840 003B4C (v01 BOCHS  BXPC     00000001 BXPC 00000001)
    [    0.003556] ACPI: FACS 0x000000007FFDF800 000040
    [    0.003558] ACPI: APIC 0x000000007FFE3480 0000F0 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
    [    0.003560] ACPI: MCFG 0x000000007FFE3570 00003C (v01 BOCHS  BXPC     00000001 BXPC 00000001)
    [    0.003563] ACPI: WAET 0x000000007FFE35AC 000028 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
    [    0.003564] ACPI: Reserving FACP table memory at [mem 0x7ffe338c-0x7ffe347f]
    [    0.003566] ACPI: Reserving DSDT table memory at [mem 0x7ffdf840-0x7ffe338b]
    [    0.003566] ACPI: Reserving FACS table memory at [mem 0x7ffdf800-0x7ffdf83f]
    [    0.003567] ACPI: Reserving APIC table memory at [mem 0x7ffe3480-0x7ffe356f]
    [    0.003568] ACPI: Reserving MCFG table memory at [mem 0x7ffe3570-0x7ffe35ab]
    [    0.003568] ACPI: Reserving WAET table memory at [mem 0x7ffe35ac-0x7ffe35d3]
[... 739 lines cut ...]
    [    1.922258] md: autorun ...
    [    1.922741] md: ... autorun DONE.
    [    1.923393] List of all partitions:
    [    1.924035] No filesystem could mount root, tried: 
    [    1.924036] 
    [    1.925236] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
    [    1.926746] CPU: 6 PID: 1 Comm: swapper/0 Not tainted 5.14.0-427.50.1.el9_4.x86_64 #1
    [    1.928139] Hardware name: Red Hat OpenStack Compute/RHEL, BIOS 1.16.1-1.el9 04/01/2014
    [    1.929529] Call Trace:
    [    1.929996]  <TASK>
    [    1.930372]  dump_stack_lvl+0x34/0x48
    [    1.931064]  panic+0x107/0x2f7
    [    1.931601]  mount_root_generic+0x1bc/0x1cf
    [    1.932361]  ? __mptcp_subflow_push_pending+0x1f0/0x370
    [    1.933308]  ? mount_root+0x134/0x176
    [    1.933956]  prepare_namespace+0x1ae/0x1f5
    [    1.934695]  kernel_init_freeable+0x186/0x1ab
    [    1.935495]  ? __pfx_kernel_init+0x10/0x10
    [    1.936222]  kernel_init+0x16/0x130
    [    1.936872]  ret_from_fork+0x2c/0x50
    [    1.937511]  </TASK>
    [    1.938346] Kernel Offset: 0x34400000 from 0xffffffff81000000 (relocation range: 0xffffffff80000000-0xffffffffbfffffff)
    [    1.940237] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0) ]---

```

mcb30, 2025-04-25:
Could you please add an `imgstat` in your script immediately before the `boot` command and report the output that you get from that?

hjensas, 2025-04-26:
I recompiled with `#define CONSOLE_SERIAL` - from master (commit 4c8bf666f4afb578645269a5a81431d784fad771) so that I can capture the requested data.

Added `imgstat` command to the script:
```
$ cat /var/www/html/boot-artifacts/agent.x86_64.ipxe 
#!ipxe
initrd --name initrd http://192.168.32.254:8081/boot-artifacts/agent.x86_64-initrd.img
kernel http://192.168.32.254:8081/boot-artifacts/agent.x86_64-vmlinuz initrd=initrd coreos.live.rootfs_url=http://192.168.32.254:8081/boot-artifacts/agent.x86_64-rootfs.img rw ignition.firstboot ignition.platform.id=metal console=ttyS0
imgstat
boot
```

This is the outpot on the console:
```
net0: fa:16:3e:47:7f:32 using virtio-net on 0000:03:00.0 (Ethernet) [open]
  [Link:up, TX:133 TXE:0 RX:115 RXE:21]
  [RXE: 15 x "Operation not supported (https://ipxe.org/3c046003)"]
  [RXE: 6 x "The socket is not connected (https://ipxe.org/380f6001)"]
Configuring (net0 fa:16:3e:47:7f:32)...... ok
net0: 192.168.32.10/255.255.255.0 gw 192.168.32.1
net0: fe80::f816:3eff:fe47:7f32/64
net1: fe80::f816:3eff:fef7:9a5e/64 (inaccessible)
Filename: http://192.168.32.254:8081/boot-artifacts/agent.x86_64.ipxe
http://192.168.32.254:8081/boot-artifacts/agent.x86_64.ipxe... ok
agent.x86_64.ipxe : 343 bytes [script]
http://192.168.32.254:8081/boot-artifacts/agent.x86_64-initrd.img... 27%     36%     42%     68%     75%     83%     91%     99%     ok
http://192.168.32.254:8081/boot-artifacts/agent.x86_64-vmlinuz... ok
initrd : 99252842 bytes
agent.x86_64-vmlinuz : 13617272 bytes [bzImage] [SELECTED] "initrd=initrd coreos.live.rootfs_url=http://192.168.32.254:8081/boot-artifacts/agent.x86_64-rootfs.img rw ignition.firstboot ignition.platform.id=metal console=ttyS0"
[    0.000000] Linux version 5.14.0-427.50.1.el9_4.x86_64 (mockbuild@x86-64-02.build.eng.rdu2.redhat.com) (gcc (GCC) 11.4.1 20231218 (Red Hat 11.4.1-3), GNU ld version 2.35.2-43.el9) #1 SMP PREEMPT_DYNAMIC Wed Dec 18 13:06:23 EST 2024
[    0.000000] The list of certified hardware and cloud instances for Red Hat Enterprise Linux 9 can be viewed at the Red Hat Ecosystem Catalog, https://catalog.redhat.com.
[    0.000000] Command line: initrd=initrd coreos.live.rootfs_url=http://192.168.32.254:8081/boot-artifacts/agent.x86_64-rootfs.img rw ignition.firstboot ignition.platform.id=metal console=ttyS0
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'

<snip>

[    1.056629] Trying to unpack rootfs image as initramfs...
[    1.057503] software IO TLB: mapped [mem 0x0000000075fb5000-0x0000000079fb5000] (64MB)
[    1.057525] ACPI: bus type thunderbolt registered
[    1.058438] Initramfs unpacking failed: invalid magic at start of compressed archive

<snip>

[    1.412507] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[    1.413167] CPU: 2 PID: 1 Comm: swapper/0 Not tainted 5.14.0-427.50.1.el9_4.x86_64 #1
[    1.413800] Hardware name: RDO OpenStack Compute/RHEL, BIOS 1.16.3-4.el9 04/01/2014
[    1.414426] Call Trace:
[    1.414631]  <TASK>
[    1.414815]  dump_stack_lvl+0x34/0x48
[    1.415120]  panic+0x107/0x2f7
[    1.415382]  mount_root_generic+0x1bc/0x1cf
[    1.415722]  ? mount_root+0x134/0x176
[    1.416026]  prepare_namespace+0x1ae/0x1f5
[    1.416366]  kernel_init_freeable+0x186/0x1ab
[    1.416722]  ? __pfx_kernel_init+0x10/0x10
[    1.417059]  kernel_init+0x16/0x130
[    1.417354]  ret_from_fork+0x29/0x50
[    1.417649]  </TASK>
[    1.417993] Kernel Offset: 0x1000000 from 0xffffffff81000000 (relocation range: 0xffffffff80000000-0xffffffffbfffffff)
[    1.418849] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0) ]---
```

A listing of the file on the web server hosting the boot artifacts:
```
$ ls -l /var/www/html/boot-artifacts
total 1248932
-rw-r--r--. 1 root root   99252842 Apr 26 08:03 agent.x86_64-initrd.img
-rw-r--r--. 1 root root        343 Apr 26 08:03 agent.x86_64.ipxe
-rw-r--r--. 1 root root 1166027776 Apr 26 08:03 agent.x86_64-rootfs.img
-rw-r--r--. 1 root root   13617272 Apr 26 08:03 agent.x86_64-vmlinuz
``

hjensas, 2025-04-26:
I spent some time bisecting, In my tests it looks like this issue was introduced in commit: ef038491858cb51f8aa17b1f6e50444d2e627413 on  April 20.

commit: ef038491858cb51f8aa17b1f6e50444d2e627413 (April 20 #4)  :: NOK
commit: b65f67d443d213bbaf9d7859649f25920e6d6785 (April 20 #3) :: OK

NiKiZe, 2025-04-26:
Reading your `imgstat` output, the script is in the list of loaded files.

Before your `initrd` line, add `imgfree` and see if anything changes.

You might also check `imgstat` output between the working and non working case.

mcb30, 2025-04-28:
@ondrejj Thank you for bisecting and for suggesting using Fedora.  I have now been able to reproduce the problem, which was not occurring with my standard TinyCore test distro.

closed 2025-04-28 by mcb30

mcb30, 2025-04-28:
@hjensas @ondrejj Could you please retest with current master (commit https://github.com/ipxe/ipxe/commit/412ad5601)?

hjensas, 2025-04-28:
> [@hjensas](https://github.com/hjensas) [@ondrejj](https://github.com/ondrejj) Could you please retest with current master (commit [412ad5601](https://github.com/ipxe/ipxe/commit/412ad5601))?

I tested with master, at (commit 6ccb6bcfc8f76f69d7bc01498b762bfe79f1cb6c) - the fix (commit https://github.com/ipxe/ipxe/commit/412ad5601) was the one prior to that.

Issue has been resolved. Thanks @mcb30!

ondrejj, 2025-04-29:
Works well, thank you.