ipxe/ipxe #1459: clonezilla ipxe boot stuck at end Kernel panic -not syncing: No working init found.Try passing init= options to kernel

amardeepl11, 2025-04-28:
Hello ,

Getting error while booting virtualbox vm using ipxe and chain load.

**error**

[image]

My ipxe config.

#!ipxe

set server_ip  <serverip>
set iso_path   bootfiles/os

set menu_color_normal=white/black
set menu_color_highlight=black/light-gray

# Figure out if client is 64-bit capable
cpuid --ext 29 && set arch x64 || set arch x86
cpuid --ext 29 && set archl amd64 || set archl i386

:MENU
menu
item exit                 Exit iPXE and continue BIOS boot
item --gap -- --------------- iPXE Install Menu ---------------
item ubuntu-22.04-desktop-amd64         Install Ubuntu Desktop 22.04 LTS
item Clonzilla-Basic                    Clonzilla-Basic
choose --default exit --timeout 10000 option && goto ${option}
item --gap -- -------------------------------------------------

choose target && goto ${target}

:ubuntu-22.04-desktop-amd64

kernel http://${server_ip}/${iso_path}/ubuntu-22.04/casper/vmlinuz  initrd=initrd ip=dhcp url=http://${server_ip}/${iso_path}/ubuntu-22.04/ubuntu-22.04.5-desktop-amd64.iso
initrd http://${server_ip}/${iso_path}/ubuntu-22.04/casper/initrd

boot

:Clonzilla-Basic
echo Starting CloneZilla with default options
kernel http://${server_ip}/bootfiles/os/pxeboot/live/vmlinuz
initrd http://${server_ip}/bootfiles/os/pxeboot/live/initrd.img
imgargs vmlinuz boot=live config noswap nolocales edd=on nomodeset ocs_live_run="ocs-live-general" ocs_live_extra_param="" ocs_live_keymap="" ocs_live_batch="no" ocs_daemonon="ssh" ocs_lang="" vga=788 nosplash noprompt fetch=http://${server_ip}/bootfiles/os/pxeboot/live/filesystem.squashfs
boot

:exit
exit

Please suggest.

Thank you
Amar

mcb30, 2025-04-28:
Most likely the issue that was fixed just now in #1458 - please retry with current master (commit https://github.com/ipxe/ipxe/commit/412ad5601)

closed 2025-04-28 by mcb30

amardeepl11, 2025-04-28:
big thank you @mcb30 

its works.

[image]